raspberrypi/linux #5563: does not recognize external microsd larger than 32GB

Ninoh-FOX, 2023-08-04:
### Describe the bug

When I try to insert a microsd from an external drive (aside from the external drive I use is a transcend usb 2.0 and it recognizes it as 05e3:0745 Genesys Logic, Inc. Logilink CR0012) it only shows me to open microsds smaller than 32GB.

I have the same problem since kernel version 5.10.*

### Steps to reproduce the behaviour

for a 16gb microsd I have this message and it works correctly:
```
[  195.316812] usb 1-1.2.2: new high-speed USB device number 7 using xhci_hcd
[  195.470285] usb 1-1.2.2: New USB device found, idVendor=05e3, idProduct=0745, bcdDevice= 9.03
[  195.470313] usb 1-1.2.2: New USB device strings: Mfr=0, Product=1, SerialNumber=2
[  195.470329] usb 1-1.2.2: Product: USB Storage
[  195.470342] usb 1-1.2.2: SerialNumber: 000000000903
[  195.473535] usb-storage 1-1.2.2:1.0: USB Mass Storage device detected
[  195.475274] scsi host1: usb-storage 1-1.2.2:1.0
[  196.792297] scsi 1:0:0:0: Direct-Access     Generic  STORAGE DEVICE   0903 PQ: 0 ANSI: 6
[  196.793660] sd 1:0:0:0: Attached scsi generic sg1 type 0
[  196.795505] sd 1:0:0:0: [sdb] 31116288 512-byte logical blocks: (15.9 GB/14.8 GiB)
[  196.797153] sd 1:0:0:0: [sdb] Write Protect is off
[  196.797168] sd 1:0:0:0: [sdb] Mode Sense: 21 00 00 00
[  196.798448] sd 1:0:0:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[  196.812871]  sdb: sdb1
[  196.813282] sd 1:0:0:0: [sdb] Attached SCSI removable disk
```

for one greater than 32gb instead I get this:

```
[  481.833863] usb 1-1.2.2: new high-speed USB device number 8 using xhci_hcd
[  481.997336] usb 1-1.2.2: New USB device found, idVendor=05e3, idProduct=0745, bcdDevice= 9.03
[  481.997360] usb 1-1.2.2: New USB device strings: Mfr=0, Product=1, SerialNumber=2
[  481.997373] usb 1-1.2.2: Product: USB Storage
[  481.997385] usb 1-1.2.2: SerialNumber: 000000000903
[  482.001562] usb-storage 1-1.2.2:1.0: USB Mass Storage device detected
[  482.005182] scsi host1: usb-storage 1-1.2.2:1.0
[  483.518883] scsi 1:0:0:0: Direct-Access     Generic  STORAGE DEVICE   0903 PQ: 0 ANSI: 6
[  483.519771] sd 1:0:0:0: Attached scsi generic sg1 type 0
[  483.797025] sd 1:0:0:0: [sdb] 124735488 512-byte logical blocks: (63.9 GB/59.5 GiB)
[  483.798171] sd 1:0:0:0: [sdb] Write Protect is off
[  483.798177] sd 1:0:0:0: [sdb] Mode Sense: 21 00 00 00
[  483.799317] sd 1:0:0:0: [sdb] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[  483.806543]  sdb: sdb1
[  483.806774] sd 1:0:0:0: [sdb] Attached SCSI removable disk
```

but it is not shown on the desktop, I can only access it from gparter and analyze it, not much else.

### Device (s)

Raspberry Pi 400

### System

- Which OS and version: Raspbian OS 32bit bulleyes

Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, a17501d7c91a584085cd794ab9c007c9d1b9b435, stage2

- Which firmware version:

Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

- Kernel version:

Linux raspberrypi 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux

pelwell, 2023-08-04:
How have you formatted this card, in terms of partitions and filing systems?

Ninoh-FOX, 2023-08-06:
In fat32 and only primary partition, in debian 11 amd 64 this automount well, but in raspbian I need mount this manually and only works how root.

pelwell, 2023-08-07:
> In fat32 and only primary partition

FAT32 can't handle partitions larger than 32GB, so I don't understand how this can work. Are you sure it's FAT32 and not exFAT?

I have a feeling that exFAT support is provided by a user-space driver, which may explain the problems you are having.

Ninoh-FOX, 2023-08-07:
You can format a sdcard of 64GB or more tall since gparted to fat32 without problems.

Ninoh-FOX, 2023-08-11:
fixed, for one thing that can happen in a trillion, two of my sd had the same partition id as the boot partition, so I changed the fstab to the uuid of the boot partition and the system and they work again, sorry for the inconvenient.

closed 2023-08-11 by Ninoh-FOX