raspberrypi/linux #5564: VC4-KMS-V3D Screen tearing artefacts in video player and OpenGL apps

qrp73, 2023-08-04:
### Describe the bug

First I notice that there is ugly screen tearing during video playback or running OpenGL application. I found this discussion: https://forums.raspberrypi.com/viewtopic.php?t=246291 they suggested to disable xcompmgr composition manager in raspi-config. I disabled it and at first look screen tearing disappears. But later I found that it is still here, just less noticeable.

So I wrote a small test app for testing vblank wait behavior of the video driver. With this app I found that screen tearing is still here, but its position depends on rendering window size and position. In a fullscreen mode screen tearing disappears, but it may be just due to specific conditions, needs to test it on different display resolution.

At a glance it looks that the video driver performs buffers swap with some unacceptable delay relative to hardware vblank event, so it happens in the middle of video buffer scan and as result, the display frame is teared and contains mixed content combined from several video buffers.

### Steps to reproduce the behaviour

1) create file test-vblank.c and fill it with following program:
```
#include <stdio.h>
#include <time.h>
#include <GL/freeglut.h>

// gcc test-vblank.c -o test-vblank -Wall -lGL -lglut

void fillRect(float x, float y, float w, float h)
{
    GLfloat rect[] = { x, y+h, x+w, y+h, x+w, y, x, y };
    glEnableClientState(GL_VERTEX_ARRAY);
    glVertexPointer(2, GL_FLOAT, 0, rect);
    glDrawArrays(GL_TRIANGLE_FAN, 0, 4);    
}

char _buf[128];
int _counter = 0;
int _cnt1 = 0;
struct timespec _ts1, _ts2;
float _fps = 0;
#define TS2NS(ts) (((uint64_t)ts.tv_sec)*1e9 + (uint64_t)ts.tv_nsec)

void displayFunc(void)
{
    // calculate fps
    timespec_get(&_ts2, TIME_UTC);
    _counter++;
    uint64_t dt_ns = TS2NS(_ts2) - TS2NS(_ts1);
    if (dt_ns > 1e9)
    {
        // update fps every second
        int frameCount = _counter - _cnt1;
        _ts1 = _ts2;
        _cnt1 = _counter;
        _fps = (frameCount*1e9) / (float)dt_ns;
    }
    
    glLoadIdentity();
    glClearColor(0.0f,0.0f,0.0f,1.0f);
    glClear(GL_COLOR_BUFFER_BIT | GL_DEPTH_BUFFER_BIT);

    // fill rectangle with different color on every next frame (multicolor effect)
    // it helps to detect inconsistency or instability of vblank wait and buffer update 
    switch (_counter & 3)
    {
        case 0: glColor3f(0.1f, 0.1f, 0.1f); break;
        case 1: glColor3f(1.0f, 1.0f, 1.0f); break;
        case 2: glColor3f(0.1f, 0.1f, 1.0f); break;
        case 3: glColor3f(0.0f, 1.0f, 0.0f); break;
    }
    fillRect(-1.0f, -1.0f, 2, 2);
    
    // print resolution, bpp and fps
    sprintf(_buf, "%dx%d:%d\n%.1f fps\n", 
        glutGet(GLUT_WINDOW_WIDTH), 
        glutGet(GLUT_WINDOW_HEIGHT), 
        glutGet(GLUT_WINDOW_BUFFER_SIZE),
        _fps);
    glColor4f(1.0f, 0.0f, 0.0f, 1.0f);
    glPushMatrix();
    glTranslatef(-0.95f, 0.85f, 0.0f);
    glScalef(0.001,0.001,1);
    glutStrokeString(GLUT_STROKE_ROMAN, (const unsigned char*)_buf);
    glPopMatrix();

    // swap buffers and request glut for next display() call for continuous update
    glutSwapBuffers();
    glutPostRedisplay();
}

int _isFullscreen;
int _wndWidth, _wndHeight;
void keyboardFunc(unsigned char key, int x, int y) 
{
    int mod = glutGetModifiers(); //1=SHIFT, 2=CTRL, 4=ALT
    if (key == ' ' && mod == 0)
    {
        // Use space key to switch fullscreen/window mode
        _isFullscreen ^= 1;
        if (_isFullscreen)
        {
            _wndWidth = glutGet(GLUT_WINDOW_WIDTH); 
            _wndHeight = glutGet(GLUT_WINDOW_HEIGHT); 
            glutFullScreen();
        }
        else
        {
            glutReshapeWindow(_wndWidth, _wndHeight);
        }
    }
}

int main(int argc, char** argv)
{
    glutInit(&argc, argv);
    glutInitDisplayMode(GLUT_DOUBLE | GLUT_RGBA);

    glutInitWindowSize(800, 600);
    glutInitWindowPosition(0, 0);
    glutCreateWindow("test-vblank");

    glutDisplayFunc( displayFunc );
    glutKeyboardFunc( keyboardFunc );

    // explicit matrices init
    glMatrixMode(GL_PROJECTION);
    glLoadIdentity();
    glMatrixMode(GL_MODELVIEW);
    glLoadIdentity();
    
    glLineWidth(2);         // use bold lines for glutStrokeString
    
    glutMainLoop();
    return 0;
}
```
2) compile it with a command line gcc test-vblank.c -o test-vblank -Wall -lGL -lglut
3) run it with ./test-vblank
4) try to move window position/size and notice that at some position/size screen tearing appears and you can see video artifacts on the display. Use button SPACE to switch between windowed and fullscreen mode for testing different modes.

The expected result: the window is flickering at constant and stable speed and doesn't contains video artifacts, it should change color for entire window surface simultaneously on every video frame. The color of windows surface should be solid with no artifacts.

The actual result: the window contains video artifacts, the display shows mixed frame which contains parts from several video buffers.

Note: you cannot take screenshot of these video artifacts, because it appears on hardware display, but not in the video buffer. So all screenshot will be good, but the displayed picture will contains video artifacts.

### Device (s)

Raspberry Pi 4 Mod. B

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage4

$ vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

$ uname -a
Linux raspi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

qrp73, 2023-08-05:
just tried it on 32-bit raspi os - the same bug, also catch tearing in fullscreen mode, so on 32-bit OS this bug is even worse...

qrp73, 2023-08-06:
just tried it on ubuntu arm64 23.04, and screen tearing issue is missing from ubuntu, it shows all frames with no tearing. But unfortunately ubuntu has much slower peroformance than Raspi OS, so I returned back. Hope screen tearing will be fixed, because it is very noticeable when playing video, opengl app or even when scrolling text in the browser...

popcornmix, 2023-08-07:
This is unlikely to be a kernel issue, so this isn't the right place for the report.

Are you using mutter? My understanding is that is required to avoid tearing, but due to increased memory use, is only enabled on Pi4 with 2GB or more. What is yours?

You can check with:
```
pi@pi4:~ $ ps aux | grep mutter | grep -v grep
pi           874  2.1  1.8 664172 71440 ?        Sl   11:33   0:01 mutter
```

popcornmix, 2023-08-07:
>  but due to increased memory use, is only enabled on Pi4 with 2GB or more. What is yours?

qrp73, 2023-08-07:
```
pi@raspi:~ $ vcgencmd get_mem arm
arm=896M
```
I'm using RPi4 with 4GB (revision 1.5, reading from MIDR_EL1 returns 0x410fd083)

popcornmix, 2023-08-07:
Actually `vcgencmd get_config total_mem` would be a better command.

Have you enabled VNC in raspi-config? That also disables mutter.

`systemctl status vncserver-x11-serviced` and see if it is active.

qrp73, 2023-08-07:
I tried to disable VNC and screen tearing really disappears. 

But now screen tearing appears in fullscreen mode. :( Just run test program, press space and wait for 1-2 sec. At beginning, when you switch to a fullscreen mode, screen tearing is missing, but after second or two it appears at the top of screen...

Also I catch strange bug. When I switch to fullscreen and back several times in test app (see source code in the issue description) it suddenly freeze the system... The window is no more updated and UI system doesn't respond to any user actions anymore. The mouse pointer is still moves and I can switch to a text tty with Ctrl+Shift+F1 and kill process, after killing process the system starts to respond to a user actions.

I can reproduce this bug several times, even after reboot it leads to a system freeze...

qrp73, 2023-08-07:
Also I found that hdmi display detect on the fly doesn't works when vncserver is active. If you connect display to another hdmi port it don't recognize it until reboot. With disabled vncserver it started works on the fly, I can switch hdmi display and it is detected and started to use immediately.

qrp73, 2023-08-07:
Also with disabled vncserver chromium-browser starts to works more smoothly, when vncserver was active, it has some lags (short freezes), I thoughts this is because RPI4 has not enough performance. But now, when I disabled vncserver, it works smoothly... The only issue is these strange freze when exit from fullscreen mode and that fullscreen now has screen tearing, with active vncserver there was no screen tearing in fullscreen mode.

vanfanel, 2023-08-08:
@qrp73 Save your time for coding: you will never get proper tearing-less movement on X11 using the Pi4 video stack.
Use Wayland or KMS/DRM instead. No tearing there and way less latency.

vanfanel, 2023-08-08:
> > Why hasn't the foundation moved Pi OS to Wayland, it's a unknown to me. Performance is way better, no tearing in any situation, better input lag using Vulkan + 2 video buffers, etc.
> 
> Bookworm (currently in beta testing) uses wayland by default.

Hell yeah, that's what I call good news! About time.. :)

popcornmix, 2023-08-08:
wayfire/wlroots.

qrp73, 2023-08-16:
Where I can get image with Bookworm and opengl support for testing?

vanfanel, 2023-08-17:
Or you can build you own: burn latest DietPi image, then build latest stable libDRM, latest stable MESA, latest stable libwayland+wlroots+labwc... here you go!

qrp73, 2023-08-18:
Just found that if I switch to ```dtoverlay=vc4-fkms-v3d``` in config.txt, the opengl performance remains exactly the same (for both high load scene and lightweight scene), but it leads to much-much worse screen tearing... 

I think there is some issue in vc4-kms-v3d and vc4-fkms-v3d drivers which cause screen tearing issue... vc4-kms-v3d works better, but still has some screen tearing issues...

I just wonder - is it open source driver?

popcornmix, 2023-08-18:
fkms is closed source and deprectated. It has known issues with tearing (that won't be fixed).

kms is open source and the default.  It has no known issues with tearing. But it can be driven in a single buffered way, which will have tearing. That is a decision of the windowing framework/compositor.

qrp73, 2023-08-18:
> But it can be driven in a single buffered way, which will have tearing.

What this means? It uses single buffer for update and the same time this buffer is scanned by video circuit in parallel?

popcornmix, 2023-08-18:
max_framebuffers determines whether a framebuffer is allocated for each of the hdmi displays.
Nothing to do with double/tripe buffering.  And it's not used by kms anyway.

qrp73, 2023-08-18:
Hm, just tried to set ```max_framebuffers=2```, it leads to increase opengl app performance for +19% and there is no tearing in window mode. Fullscreen mode still has a fixed offset screen tearing at the bottom of screen. 

There is also some bug when I switch between fullscreen mode and window mode, it can lead to broken window appearance and video artefacts. The same bug happens in VLC player. It looks like compositor bug.

popcornmix, 2023-08-24:
> There is also some bug when I switch between fullscreen mode and window mode, it can lead to broken window appearance and video artefacts. The same bug happens in VLC player. It looks like compositor bug. This is strange, but this bug is missing when VNC is enabled.

mutter is the default window manager/compositor on Pi4 with 2G or more RAM.
It doesn't play nicely with VNC, so if you enable VNC mutter is disabled.

You can check if mutter is running with:
```
pi@raspberrypi:~ $ ps aux | grep mutter | grep -v grep
pi           781  0.6  2.2 533788 87380 ?        Sl   13:59   0:07 mutter
```

qrp73, 2023-08-25:
> I've tested openarea with a timedemo on Pi4 and get: max_framebuffers=2 fullscreen 42.9 windowed 38.6
> max_framebuffers=0 fullscreen 42.8 windowed 38.7
> so I'm not seeing the performance change you see.

just tried it again, but cannot reproduce this behavior again, now it gives almost the same fps for max_framebuffers=0/1/2. I remember that I catch it with lightweight scenes, but may be it was effect of some background process.

qrp73, 2023-08-25:
> mutter is the default window manager/compositor on Pi4 with 2G or more RAM. It doesn't play nicely with VNC, so if you enable VNC mutter is disabled.

it was happened with mutter enabled (VNC disabled). But now I cannot reproduce this bug. Previously it was easy to catch this bug with several window/fullscreen switch, but now it works ok. I'm not sure if it was fixed with some recent updates or due to some changes in my OS configuration, but at the moment it seems that this bug disappears.

qrp73, 2023-08-25:
No, I was wrong, it still here. There is need to wait several seconds after switch into fullscreen mode until screen tearing appears at the top of screen, then switch into window mode and window layout become broken. But it needs to do it several times to catch it.

qrp73, 2023-11-15:
not relevant anymore

closed 2023-11-15 by qrp73