raspberrypi/linux #5566: On CM4  `ttyAMA1` (v6.1.21-v8+) changes to `ttyS0` (v6.1.42-v8+) with latest kernel/firmware from `rpi-update`

JinShil, 2023-08-07:
### Describe the bug

We have our Raspberry Pi 64-bit Bullseye OS configured below:

```
dtoverlay=disable-bt
dtparam=uart0=on
dtoverlay=uart2
dtoverlay=uart3
dtoverlay=uart4
```

We also have the following udev rule:

```
KERNEL=="ttyAMA0", SYMLINK:="serial0"
KERNEL=="ttyAMA1", SYMLINK:="serial1"
KERNEL=="ttyAMA2", SYMLINK:="serial2"
KERNEL=="ttyAMA3", SYMLINK:="serial3"
KERNEL=="ttyS0", PROGRAM="/bin/sh -c '\
        if [ -e /dev/ttyAMA1 ]; then \
                echo ; \
        else \
                exit 1; \
        fi \
'", SYMLINK:="%c"
```

In kernel 6.1.21-v8+, this results in the creation of the following serial port devices:
```
$ ls -al /dev/serial*
lrwxrwxrwx 1 root root 7 Aug  7 00:33 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 7 Aug  7 00:33 /dev/serial1 -> ttyAMA1
lrwxrwxrwx 1 root root 7 Aug  7 00:33 /dev/serial2 -> ttyAMA2
lrwxrwxrwx 1 root root 7 Aug  7 00:33 /dev/serial3 -> ttyAMA3
```

However, with the latest kernel obtained through `rpi-update`, 6.1.42-v8+, The serial devices are being enumerated as...
```
$ ls -al /dev/serial*
lrwxrwxrwx 1 root root 7 Aug  7 00:35 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Aug  7 00:35 /dev/serial1 -> ttyS0
lrwxrwxrwx 1 root root 7 Aug  7 00:35 /dev/serial2 -> ttyAMA2
lrwxrwxrwx 1 root root 7 Aug  7 00:35 /dev/serial3 -> ttyAMA3
```
 `/dev/ttyAMA1` no longer exists

### Steps to reproduce the behaviour

1.  Configured the serial ports as described earlier in /boot/config.txt using kernel 6.1.21-v8+
2.  After booting the OS confirm that `/dev/ttyAMA0~3` are created
3. Run `rpi-update` and reboot
4. After rebooting the OS confirm that `/dev/ttyAMA1` no longer exists and is replaced by `/dev/ttyS0`

### Device (s)

Raspberry Pi CM4

### System

Prior to running `rpi-update`:
```
March 17, 2023 10:50:39 
82f3750a65fadae9a38077e3c2e217ad158c8d54
6.1.21-v8+ #1642 SMP PREEMPT
```

After running `rpi-update`:
```
Apr 25 2023 18:26:03
version d7f9c2b4ef7e4a8c0b04374a879ce89d7a948453
6.1.42-v8+ #1668 SMP PREEMPT
```

pelwell, 2023-08-07:
Since the 6.1.21 kernel I have removed a downstream patch that disabled the setting of UART device numbers based on Device Tree entries, which has led to the change in behaviour that you have seen. The problem with the old behaviour is that the UART device names depended on which other UARTs were enabled, and there may even have been a dependence on the precise order in which the devices are initialised. With the newer kernels, which match the upstream kernel, UARAT device numbers are controlled by the "serial<n>" aliases in Device Tree, and we set those aliases to match UART numbers in the datasheet:

| alias | hardware name | device name |
|--------|--------|--------|
| serial0 | UART0 | ttyAMA0 |
| serial1 | UART1 | ttyS0 |
| serial2 | UART2 | ttyAMA2 |
| serial3 | UART3 | ttyAMA3 |
| serial4 | UART4 | ttyAMA4 |
| serial5 | UART5 | ttyAMA4 |

You've probably seen the anomaly in the list - `ttyS0`. This is the mini-UART, a bare-bones implementation of a classic 8250 UART, the driver for which doesn't make use of the DT aliases. Given that there is only one such UART, and that the driver uses a different stem for the device names, I don't think `ttyS0` is any worse than `ttyS1`.

I don't understand your ttyS0 udev rule - it looks as though it would never return a number for ttyS0, so no `serial<n>` symbolic link, so why bother including it?

If you want to continue to use the same symbolic link names then I think your udev rules should be:
```
KERNEL=="ttyAMA0", SYMLINK:="serial0"
KERNEL=="ttyAMA2", SYMLINK:="serial1"
KERNEL=="ttyAMA3", SYMLINK:="serial2"
KERNEL=="ttyAMA4", SYMLINK:="serial3"
```

JinShil, 2023-09-01:
I have the following devices:
```
crw-rw---- 1 root dialout 204, 64  9월  1 15:06 /dev/ttyAMA0
crw-rw---- 1 root dialout 204, 65  9월  1 15:06 /dev/ttyAMA1
crw-rw---- 1 root dialout 204, 66  9월  1 15:06 /dev/ttyAMA2
crw-rw---- 1 root dialout 204, 67  9월  1 15:06 /dev/ttyAMA3
crw-rw---- 1 root dialout   4, 64  9월  1 15:06 /dev/ttyS0
```
... when configured with ...
```
dtoverlay=disable-bt
dtparam=uart0=on
dtoverlay=uart2
dtoverlay=uart3
dtoverlay=uart4
```

What is `ttyAMA1`?  It isn't listed in the table in your previous comment.  

Contrary to my original comment, this test is with `Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux` firmware `version 82f3750a65fadae9a38077e3c2e217ad158c8d54 Mar 17 2023 10:50:39`, created today with a pi-gen-like script.  I did not run `rpi-update`.

Are the changes you mentioned released in the apt repositories yet?

JinShil, 2023-09-01:
> I don't understand your ttyS0 udev rule - it looks as though it would never return a number for ttyS0, so no serial<n> symbolic link, so why bother including it?

I need to support both CM4 and CM4S devices with the same OS image, but different config.txt.

JinShil, 2023-09-01:
> Since the 6.1.21 kernel I have removed a downstream patch that disabled the setting of UART device numbers based on Device Tree entries, which has led to the change in behaviour that you have seen.

Is that a typo?  The latest kernel from the apt repository is 6.1.21, but these udev rules work fine:

```
KERNEL=="ttyAMA0", SYMLINK:="serial0"
KERNEL=="ttyAMA1", SYMLINK:="serial1"
KERNEL=="ttyAMA2", SYMLINK:="serial2"
KERNEL=="ttyAMA3", SYMLINK:="serial3"
```

After running `rpi-update` then those udev rules no longer work and these udev rules do
```
KERNEL=="ttyAMA0", SYMLINK:="serial0"
KERNEL=="ttyAMA2", SYMLINK:="serial1"
KERNEL=="ttyAMA3", SYMLINK:="serial2"
KERNEL=="ttyAMA4", SYMLINK:="serial3"
```

But, I need to guard against things breaking when eventually running an apt upgrade.  How can I modify the udev rules to detect the change, when its released, and switch to the new naming convention.

pelwell, 2023-09-01:
> > Since the 6.1.21 kernel I have removed a downstream patch that disabled the setting of UART device numbers based on Device Tree entries, which has led to the change in behaviour that you have seen.

> Is that a typo?  The latest kernel from the apt repository is 6.1.21, but these udev rules work fine:

Not a typo, but possibly a difference in understanding of the word "Since". I use it to mean "in the time after", whereas you might be interpreting it to mean "starting with". The change actually went into 6.1.26.

> I need to guard against things breaking when eventually running an apt upgrade. How can I modify the udev rules to detect the change, when its released, and dynamically switch to the new naming convention?

It's not quite as user-friendly, but the IO base address is not going to change:
```
pi@raspberrypi:~$ udevadm info -a --name=/dev/ttyAMA0 | grep iomem_base
    ATTR{iomem_base}=="0xFE201000"
pi@raspberrypi:~$ udevadm info -a --name=/dev/ttyAMA2 | grep iomem_base
    ATTR{iomem_base}=="0xFE201400"
pi@raspberrypi:~$ udevadm info -a --name=/dev/ttyAMA3 | grep iomem_base
    ATTR{iomem_base}=="0xFE201600"
pi@raspberrypi:~$ udevadm info -a --name=/dev/ttyAMA4 | grep iomem_base
    ATTR{iomem_base}=="0xFE201800"
```

So you should be able to make your rules something like this:
```
ATTR{iomem_base}=="0xFE201000", SYMLINK:="serial0"
ATTR{iomem_base}=="0xFE201400", SYMLINK:="serial1"
ATTR{iomem_base}=="0xFE201600", SYMLINK:="serial2"
ATTR{iomem_base}=="0xFE201800", SYMLINK:="serial3"
```

JinShil, 2023-09-04:
I believe that gives us what we need.  Thank you.

closed 2023-09-04 by JinShil