raspberrypi/linux #5568: kunit logic broken

yu-re-ka, 2023-08-08:
### Describe the bug

Raspberry Pi kernel built with CONFIG_KUNIT=m is

### Steps to reproduce the behaviour

Build a raspberry pi kernel with arm64 bcm2711_defconfig and CONFIG_KUNIT=m

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry pi kernel 6.1.21 [1.20230405](https://github.com/raspberrypi/linux/releases/tag/1.20230405) compiled from source

```
[yuka@yuyu:~]$ zcat /proc/config.gz  | grep CONFIG_KUNIT
CONFIG_KUNIT=m
# CONFIG_KUNIT_DEBUGFS is not set
CONFIG_KUNIT_TEST=m
CONFIG_KUNIT_EXAMPLE_TEST=m
CONFIG_KUNIT_ALL_TESTS=m
CONFIG_KUNIT_DEFAULT_ENABLED=y

[yuka@yuyu:~]$ 
```

### Logs

```
[   45.346364]     # Subtest: vc4-pv-muxing-combinations
[   45.346380]     1..2
[   45.346387]         # Subtest: drm_vc4_test_pv_muxing
[   45.353717] [drm] Initialized vc4 0.0.0 20140616 for drm-kunit-mock-device on minor 0
[   45.353842] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000058
[   45.353905] Mem abort info:
[   45.353972]   ESR = 0x0000000096000005
[   45.354000]   EC = 0x25: DABT (current EL), IL = 32 bits
[   45.354042]   SET = 0, FnV = 0
[   45.354060]   EA = 0, S1PTW = 0
[   45.354078]   FSC = 0x05: level 1 translation fault
[   45.354105] Data abort info:
[   45.354120]   ISV = 0, ISS = 0x00000005
[   45.354137]   CM = 0, WnR = 0
[   45.354153] user pgtable: 4k pages, 39-bit VAs, pgdp=000000006221b000
[   45.354179] [0000000000000058] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[   45.354226] Internal error: Oops: 0000000096000005 [#1] SMP
[   45.354251] Modules linked in: vc4 snd_soc_hdmi_codec drm_kunit_helpers cec drm_dma_helper drm_display_helper drm_kms_helper snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine kunit wireguard libchacha20poly1305 chacha_neon poly1305_neon libcurve25519_generic libchacha ip6_udp_tunnel udp_tunnel nfnetlink 8021q garp btsdio bluetooth ecdh_generic ecc libaes brcmfmac joydev bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) rpivid_hevc(C) brcmutil v4l2_mem2mem bcm2835_mmal_vchiq(C) snd_bcm2835(C) videobuf2_vmalloc cfg80211 videobuf2_dma_contig nls_cp437 videobuf2_memops videobuf2_v4l2 snd_pcm videobuf2_common videodev snd_timer rfkill snd raspberrypi_hwmon crct10dif_ce mc vc_sm_cma(C) nvmem_rmem uio_pdrv_genirq uio sch_fq_codel atkbd libps2 serio vivaldi_fmap tun tap macvlan bridge drm stp llc fuse backlight efi_pstore ip_tables x_tables xhci_pci xhci_pci_renesas dm_mod dax
[   45.354670] CPU: 1 PID: 1873 Comm: kunit_try_catch Tainted: G         C       N 6.1.21 #1-NixOS
[   45.354767] Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
[   45.354819] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   45.354853] pc : vc4_hvs_get_fifo_frame_count+0xa4/0xf4 [vc4]
[   45.354998] lr : vc4_hvs_get_fifo_frame_count+0x48/0xf4 [vc4]
[   45.355068] sp : ffffffc009253ce0
[   45.355083] x29: ffffffc009253ce0 x28: 0000000000000000 x27: 0000000000000000
[   45.355124] x26: 0000000000000000 x25: ffffff8047f5a168 x24: 00000000ffffffff
[   45.355161] x23: ffffff804ac53e98 x22: 0000000000000038 x21: ffffff804ac53c80
[   45.355197] x20: 0000000000000000 x19: 0000000000000058 x18: ffffffffffffffff
[   45.355227] x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
[   45.355256] x14: 0000000000000040 x13: 0000000000000000 x12: 0000000000000000
[   45.355286] x11: 0000000000000000 x10: ffffff804bfb6800 x9 : ffffffd84f392f44
[   45.355315] x8 : ffffffd84f3ec000 x7 : 0000000000000000 x6 : 0000000000000000
[   45.355345] x5 : 0000000000000001 x4 : 0000000000000000 x3 : 0000000000000001
[   45.355377] x2 : ffffffa8a7a20000 x1 : 0000000000000000 x0 : 0000000000000001
[   45.355409] Call trace:
[   45.355429]  vc4_hvs_get_fifo_frame_count+0xa4/0xf4 [vc4]
[   45.355517]  vc4_hvs_mark_dlist_entry_stale+0x50/0x14c [vc4]
[   45.355638]  vc4_crtc_destroy_state+0x30/0x50 [vc4]
[   45.355708]  drm_atomic_state_default_clear+0x11c/0x344 [drm]
[   45.355978]  __drm_atomic_state_free+0x80/0xbc [drm]
[   45.356112]  vc4_pv_muxing_test_exit+0xac/0xc0 [vc4]
[   45.356205]  kunit_try_run_case+0x60/0xb0 [kunit]
[   45.356246]  kunit_generic_run_threadfn_adapter+0x28/0x34 [kunit]
[   45.356282]  kthread+0xe0/0xe4
[   45.356314]  ret_from_fork+0x10/0x20
[   45.356344] Code: d50323bf d65f03c0 f9400ab3 91016273 (b9400273)
[   45.356381] ---[ end trace 0000000000000000 ]---
```

### Additional context

In NixOS we provide a kernel for the Raspberry Pi 4 that is built from downstream sources (release tags of this repo) and the appropriate defconfig, but with AUTO_MODULES=1.
Becuase we have AUTO_MODULES=1 and thus CONFIG_KUNIT=m we are running into this bug but Raspbian isn't.

These commits seem to be not working as intended:

https://github.com/raspberrypi/linux/commit/6395d01970dfee74e1c5255eb1394213d8a65666

https://github.com/raspberrypi/linux/pull/5336/commits/46234a943f07af7faa3db7267bd7bfcd0dec7159

yu-re-ka, 2023-08-08:
CC @sulix @mripard

pelwell, 2023-08-08:
What's AUTO_MODULES? It's not a kernel config setting.

yu-re-ka, 2023-08-08:
Ah, nevermind. I thought it was a kernel thing, but it's a NixOS thing.
Basically, going through an `make oldconfig` and answering 'm' where possible.

pelwell, 2023-08-08:
Is that a sensible thing to do? There are circumstances where a driver, although module-capable, is required before the filing system is available. Or do you use initramfs as a matter of course?

yu-re-ka, 2023-08-08:
`make oldconfig` only asks about the options that are not already specified in the bcm2711_defconfig, so everything that is specified as 'y' in bcm2711_defconfig will stay like that.

sulix, 2023-08-08:
I'll have a more thorough look in the morning, but I _think_ this might be due to a missing fix: db105c37a4d69d684c1edf2915557463d0ba172c

Normally modpost would pick up that issue, though, so it could be something else.

That being said, you probably don't want to be running the KUnit tests on a normal boot — they're intended for developer testing only, and do things like leak kernel addresses (potentially defeating KASLR) and can leave the system in an undefined state. On newer kernels, they'll trigger a kernel taint, too.

I'd probably keep CONFIG_KUNIT=n for release builds, but if you can't do that for some reason (e.g., you want to have a single compiled build for both testing and release), you can set kunit.enable=0 on the command line (and make this default with ``CONFIG_KUNIT_DEFAULT_ENABLED=n``).

yu-re-ka, 2023-08-09:
Thank you for the explanation. We will now change CONFIG_KUNIT to 'n' on all NixOS kernel builds, after finding the following documentation:

> CONFIG_KUNIT should not be enabled in a production environment. Enabling KUnit disables Kernel Address-Space Layout Randomization (KASLR), and tests may affect the state of the kernel in ways not suitable for production.

https://www.kernel.org/doc/html/latest/dev-tools/kunit/start.html

closed 2023-09-04 by yu-re-ka