raspberrypi/linux #5570: BMP180 not ready with 6.1.43+ on Zero

DougieLawson, 2023-08-09:
### Describe the bug

Unable to read BMP180 I2C device as it fails during boot.

dmesg | grep bmp gives
```
[   59.371515] bmp280 1-0077: supply vddd not found, using dummy regulator
[   59.435210] bmp280 1-0077: supply vdda not found, using dummy regulator
[   59.507681] bmp280 1-0077: non-rising trigger given for EOC interrupt, trying to enforce it
[   59.507892] bmp280 1-0077: unable to request DRDY IRQ
[   60.031037] SPI driver bmp280 has no spi_device_id for bosch,bmp085
```

### Steps to reproduce the behaviour

Boot from 6.1.43+

### Device (s)

Raspberry Pi Zero

### System

6.1.43+

### Logs

```
[   59.371515] bmp280 1-0077: supply vddd not found, using dummy regulator
[   59.435210] bmp280 1-0077: supply vdda not found, using dummy regulator
[   59.507681] bmp280 1-0077: non-rising trigger given for EOC interrupt, trying to enforce it
[   59.507892] bmp280 1-0077: unable to request DRDY IRQ
[   60.031037] SPI driver bmp280 has no spi_device_id for bosch,bmp085
```

IIO device doesn't exist.

DougieLawson, 2023-08-09:
I've reverted to 6.1.42+ (fc879d483956bb67ff5f4fc1a83b3df0f10222a2) and my I2C BMP180 is working again.

pelwell, 2023-08-10:
How are you loading the driver? 59s is a long way into a boot.

DougieLawson, 2023-08-10:
Hi Phil,

I'm loading it from config.txt

```
[0xebd5f1e8]
# ** Falcon     : ebd5f1e8 **
# ** Raspberry Pi Zero Rev 1.2
dtparam=audio=on
gpu_mem=64
dtparam=act_led_activelow=on
enable_uart=1
dtoverlay=i2c-sensor,bmp180
...
[all]
#dtparam=spi=on
dtoverlay=spi0-2cs
dtparam=i2c_arm=on
disable_splash=1
```

pelwell, 2023-08-10:
Nothing stands out as an obvious change that could affect the bmp180 apart from adding bmp380 to the i2c-sensor overlay, and that looks harmless.

Does reverting just /boot/overlay/i2c-sensor.dtbo to https://github.com/raspberrypi/rpi-firmware/blob/49ee148826117e12900ec544cbce2b6488271990/overlays/i2c-sensor.dtbo make a difference? Or the converse, if you prefer - staying pinned on 6.1.42+ but bumping the overlay to https://github.com/raspberrypi/rpi-firmware/blob/5eaa5839de98012c21fb75cd0a075ecfcb573bff/overlays/i2c-sensor.dtbo ?

And which error bmp280 messages do you get in the kernel log on 6.1.42+?

pelwell, 2023-08-10:
I found a bmp180 in a drawer, and although I'm seeing the same log messages as you it is working:
```
$ dmesg | grep bmp280
[   14.841331] bmp280 1-0077: supply vddd not found, using dummy regulator
[   14.850894] bmp280 1-0077: supply vdda not found, using dummy regulator
[   14.904869] bmp280 1-0077: non-rising trigger given for EOC interrupt, trying to enforce it
[   14.920450] bmp280 1-0077: unable to request DRDY IRQ
[   15.015812] SPI driver bmp280 has no spi_device_id for bosch,bmp085
$ cat /sys/bus/i2c/devices/1-0077/iio\:device0/in_pressure_input
```

DougieLawson, 2023-08-10:
Hi Phil,

I've upgraded everything to https://github.com/raspberrypi/rpi-firmware/6651502efc1c98189bf698583c6f15d2b4db7e1d (6.1.44) and it's all working. So I'll assume this was a transient error with my I2C device.

Thanks for looking at this.

closed 2023-08-10 by DougieLawson