raspberrypi/linux #5572: Raspberry Pi Zero W: kernel oops and panic under memory or storage load in any recent kernel

fossham, 2023-08-13:
### Describe the bug

Kernel oopses and panics at several stages of initial configuration and apt package installations immediately after booting. Several cold boots were necessary to complete initial configuration and updating to latest packages was almost impossible, requiring many restarts.

### Steps to reproduce the behaviour

1. Use "Raspberry Pi Imager v1.7.5" to create a MicroSD card with "Raspberry Pi OS Lite (32-bit)"
    1. I used two different brands of MicroSD cards; same results from both cards.
2. Insert card into Raspberry Pi Zero W connected to a monitor
    1. I used two different Raspberry Pi Zero Ws; same results on both devices.
3. Apply power
    1. Sometimes the oops or panic messages would occur during the boot process
4. If boot completes successfully, try to update packages: `sudo apt update` and `sudo apt dist-upgrade`
    1. I had to install each package individually, and sometimes had to restart multiple times during a single package update, to get all packages installed due to the frequent crashes.
5. Crashes can be isolated to RAM or storage access by installing and running `stress` command
    1. `sudo apt install stress`
    2. `sudo stress --verbose --timeout 30 --cpu 4`
        1. This never produced any oopses or panics.
    3. `sudo stress --verbose --timeout 30 --vm 1`
        1. This *usually* produced a panic or oops; increasing timeout to 60 seconds caused oopses 100% of the time.
    4. `sudo stress --verbose --timeout 30 --io 4`
        1. This never produced any oopses or panics.
    5. `sudo stress -verbose --timeout 30 --hdd 2`
        1. This always produced a panic or oops.
6. Same types of frequent oopses and panics were found when running Debian's latest 12.1 armel images with 6.1 kernel.
7. Isolation of this problem to later kernel versions can be accomplished by using "Raspberry Pi Imager v1.7.5" to create a MicroSD card with "Raspberry Pi OS Lite (Legacy)". Booting and updates do not cause any kernel panics or oopses, and the stress tests in step 5 do not cause any such crashes either. After updates, this is using kernel version 5.10.103+.

### Device (s)

Raspberry Pi Zero W / WH

### System

```
root@raspberrypi:~# cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage2

root@raspberrypi:~# vcgencmd version
Mar 17 2023 10:52:42 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

root@raspberrypi:~# uname -a
Linux raspberrypi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
```

### Logs

[image]
[image]

pelwell, 2023-08-13:
You make no mention of the power supply you are using. The Zero W requires less power than most Pis but its margins are small - the input voltage must be stable, with a consistent 2.5A available.

Each chip has variations in power requirements, and yours may be more demanding than most - try adding `over_voltage=2` to config.txt.

fossham, 2023-08-14:
I knew the power supply would make no difference due to replication step #7 above, but I tried a 3-amp power supply from a Raspberry Pi 3 kit and the indications were unchanged: oopses could still be triggered any time by heavy RAM and storage use, especially with the `stress` command.

However, your `over_voltage=2` workaround **did work!** and the device passed all stress tests successfully. I installed the Debian image, added that magical incantation to its boot partition, and am now happily running the latest Debian 12 "Bookworm" distro with all packages updated with no crashes. So that is a successful workaround, and I will be employing it on the second RPi Zero W as well.

I still consider this a regression bug and will be happy to test if updates are provided, but even if it's just left as a workaround, I suggest adding this use case for the `over_voltage` configuration to https://www.raspberrypi.com/documentation/computers/config_txt.html . Note that the documentation is currently ambiguous: it says,

> this sets the warranty bit if over_voltage_* > 0 is also set.

Technically `over_voltage` is not a member of the set `over_voltage_*` settings, but if setting it > 0 is a requirement even for the device to not crash, then it wouldn't be fair to the user to set the warranty bit. Does it? I suggest making this explicit in the documentation.

Thanks for the workaround suggestion! I'm a happy new Pi Zero user now.

pelwell, 2023-08-14:
It's a regression as far as you are concerned, but it is likely to be a benefit for others in that it probably represents better code optimisation, either in the kernel source or from compiler improvements.

closed 2023-08-14 by pelwell

popcornmix, 2023-08-14:
over_voltage only sets the warranty bit if used in conjunction with force_turbo.

(and as an aside the warranty is not checked in a warranty replacement - we haven't seen any correlation between that bit being set and early failures).

fossham, 2023-08-14:
@popcornmix thank you for the clear answer to my question.

I think it would be great if the documentation could be updated to say that the RPi Zero W is not supported by the default configuration, and oopses are accepted and expected under v6 kernels. Here are some good candidate places to put the magical incantation, `over_voltage=2`:

* https://www.raspberrypi.com/documentation/computers/config_txt.html#common-hardware-configuration-options
* https://www.raspberrypi.com/documentation/computers/config_txt.html#boot-options as a new over_voltage entry.

I will reach out to the Debian maintainer to update their documentation as well.

popcornmix, 2023-08-14:
> I think it would be great if the documentation could be updated to say that the RPi Zero W is not supported by the default configuration, and oopses are accepted and expected under v6 kernels. Here are some good candidate places to put the magical incantation, over_voltage=2:

I do not agree with that. oopses are no accepted nor expected with default over_voltage setting.
However we have found a very small percentage (<1%) of Pi boards that pass manufacture test, but may have reliability issues at the default voltage. Unfortunately yours is in this category.

A workaround is to increase the default voltage on this board. Alternatively contact your reseller for a replacement - it is very unlikely that a new board will require this workaround.