raspberrypi/linux #5573: Kernel 6.5 freezes on boot on Pi 3

ccdffddfddfdsfedeee, 2023-08-14:
### Describe the bug

When booting the 6.5 kernel on a Pi 3B or 3B+, the system freezes after the log message running init appears, then a message appears that the system has stalled. This does not happen on the Pi 4B. Kernel 6.5.0-rc5 and 6.5.0-rc6 were tested.

### Steps to reproduce the behaviour

- Install latest version of Pi OS 64-bit
- Compile and install kernel
- Boot up and screen freezes

### Device (s)

Raspberry Pi 3 Mod. B, Raspberry Pi 3 Mod. B+

### System

```
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage4

Mar 17 2023 10:52:42 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
```

### Logs

[image]

lategoodbye, 2023-08-20:
@ccdffddfddfdsfedeee I think, i can reproduce this issue with current mainline Linux 6.5-rc6 on Raspberry Pi 3B Plus (arm64/defconfig).

Are your USB devices also dead (no power)?

popcornmix, 2023-08-21:
I can reproduce. 64-bit (downstream) 6.5 kernel does hang on boot on Pi3. It boots on Pi4.
kernels 6.1, 6.2, 6.3, 6.4 all boot on Pi3 and Pi4.

ascendbeing, 2023-08-23:
I've had this issue since rc1. I've tried every rc of 6.5, rc1 thru rc7. I am using Manjaro's rpi4-linux-rc / rpi4-linux-rc-headers (despite the name, intended for 3 as well). it does not boot on 6.5 of the Manjaro. I've tried to  build my own kernel a few times, trying with both defconfigsz the rpi3 one and the other one. nothing works. all attempts were on arm64.

1. it falls around init (init is last or 2nd to last thing I see)
2. if I have a USB keyboard transceiver and a USB storage hooked up, it will fail with dwc_otg and complaining about 100mW ports and insufficient power.
3. if it gets any further last thing before or just after init will be about dwc_otg and then I'll get a blank screen and nothing ever happens after that, nothing on network, just blank video and nothing.
4. I've tried it both on mSD and USB neither works, with either hooked up or both hooked up
5. I have taken 1-2  videos and possibly camera photos of the boot failure issue which I may be able to share them to help this issue get fixed if it would. they're from rc5 or rc6 with loglevel=9
6. all of this is on my only rpi which is rp3b original not plus. running btrfs root with Manjaro arm64

ascendbeing, 2023-08-23:
[image]
this is the final part from my boot video, recorded Thursday. it went blank after displaying this. this would be rc 5 or 6 or 7.
it's possible this was 6.4.9 or .11. I can't tell because the camera was horizontal and out of focus for the beginning of recording.

popcornmix, 2023-08-23:
@ascendbeing is the kernel you build an unpatched upstream kernel (e.g. from upstream tag v6.5-rc7)?
i.e. with no downstream Pi patches?

lategoodbye, 2023-08-23:
Based on the photo this is a downstream kernel, since it contains dwc_otg.

popcornmix, 2023-08-23:
@lategoodbye and upstream kernel boots okay for you on Pi3?

lategoodbye, 2023-08-23:
At least with Linux 6.5-rc6 (arm64/defconfig) i wasn't able to see these rcu issues. The GUI comes without any issues.

ascendbeing, 2023-08-23:
> @ascendbeing is the kernel you build an unpatched upstream kernel (e.g. from upstream tag v6.5-rc7)? i.e. with no downstream Pi patches?
around rc4-rc6 I tried to build 2-3 different builds. they were all built based on this https://gitlab.manjaro.org/manjaro-arm/packages/core/linux-rpi4-rc which is the same as upstream except he patches in the Manjaro logo: https://gitlab.manjaro.org/manjaro-arm/packages/core/linux-rpi4-rc/-/blob/master/PKGBUILD#L49
they use a particular .config that looks to be based on bcm2711_defconfig. I tried either a build of his config, with modifications, or based on bcm2711_defconfig upstream. when I say modifications I just mean tuning some .config stuff, mostly to turn off stuff I don't use, cause I was compiling it on the rp3b.
the other build I tried was based on upstream bcmrpi3_defconfig. cause I thought maybe that was why I wasn't able to boot. maybe it's possible my .config modifications were over zealous, but I didn't manage to boot at any point with any of my own builds based on modified config of the upstream defconfigs or the downstream config ( https://gitlab.manjaro.org/manjaro-arm/packages/core/linux-rpi4-rc/-/blob/master/config )

ascendbeing, 2023-08-25:
[image]
this is from rc5. it hanged here forever. idk if I compiled it myself or if I used the Manjaro prebuilt but it's from 8/8

ascendbeing, 2023-08-25:
[image]
this is from 8/8. possibly from the downstream build possibly the build I was talking about doing based on their .config, judging by the "manufacturer" you see there. indefinite USB related hang just like the OP

lategoodbye, 2023-08-26:
@ascendbeing

Could you please try switching to the mainline USB driver by adding this line to config.txt

dtoverlay=dwc2,dr_mode=host

Does it improve the situation?

ccdffddfddfdsfedeee, 2023-08-26:
This line fixed the issue, the system boots now

ascendbeing, 2023-08-26:
I will try that and edit this post with whether that works
 thanks for the suggestion cause I never would've thought of trying that (new to ARM platforms)

ascendbeing, 2023-09-19:
just adding that the overlay enablement still lets me boot up thru 6.6rc1, haven't tried 6.6rc2 yet, and I haven't tried removing the line again, don't really plan on trying to remove it at least until it's easier when I get another proper machine up and running.

licaon-kter, 2023-12-04:
> `dtoverlay=dwc2,dr_mode=host`
> Does it improve the situation?

Fixed it here on 3+ too for 6.6.2, Debian Buster, local built [kernel](https://github.com/raspberrypi/linux/blob/rpi-6.6.y/Makefile) + [latest firmware](https://github.com/raspberrypi/firmware/tree/master/boot)

It creates the missing
`usb 1-1.1.1: new high-speed USB device number 5 using dwc2`
so the network card appears:
```
lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): No External EEPROM. Setting MAC Speed
lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): int urb period 64
```

but... it stops there... the interface is not brought up, the systemd networking service fails, as at the moment it runs `eth0` is not available

Tried to `up` it in `rc.local` but that too is run too fast/soon... again, no `eth0` available at that moment either.

Is there anything else to add to my kernel config? config.txt? Delay the systemd service or?