raspberrypi/linux #5575: Installation bug

sakittelsen, 2023-08-15:
### Describe the bug

[image]
This setup should give me a Norwegian keyboard layout, I would think? But it does not, it gives me a US layout. No matter if the US keyboard box is ticked or not, it will always give me a US keyboard. Even when the English language box is not ticked, it will give me a US layout.

Also, the Norwegian letter å is shown as a code, in Bokmål. If the Norwegian letter å cannot be shown, then it is standard to use a double aa, like Bokmaal. That would be more correct and look a lot better than what is shown on the screen

### Steps to reproduce the behaviour

Chose the country Norway and the language Bookmål during installation, and you get the US keyboard layout.

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi 400

### System

Any version of Raspberry Pi OS.

XECDesign, 2023-08-15:
Again, this doesn't look like a linux kernel issue.

It should be reported here - https://github.com/RPi-Distro/repo/issues/new

@spl237

sakittelsen, 2023-08-15:
No wonder that the bugs don't get reported, it's almost impossible :-)
I have tried the Raspberry Pi forum, Raspberry Pi contact, Github and now
another Github page.
But thanks for replying, anyway :-)

spl237, 2023-08-15:
The bug (if it is indeed a bug...) should be reported here - https://github.com/raspberrypi-ui/piwiz

sakittelsen, 2023-08-15:
I'll do :)

closed 2023-08-15 by popcornmix