raspberrypi/linux #5576: Waveshare 7inch DSI LCD (C) - doesn't work

miklosakos, 2023-08-15:
### Describe the bug

The open source reverse engineered driver doesn't drive the Waveshare 7inch DSI LCD display, the official ones aren't a suitable option.
@6by9 worked on the driver, according to them it should work but the backlight never turns and the touch doesn't work either.

### Steps to reproduce the behaviour

Update to latest raspberry pi firmware, enable the required dtoverlays.

### Device (s)

Raspberry Pi 4 Mod. B

### System

`cat /etc/rpi-issue`

```
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage2
```

`vcgencmd version`

```
Aug 10 2023 15:33:38
Copyright (c) 2012 Broadcom
version 03dc77429335caee083e22ddc8eec09c07f12a7a (clean) (release) (start)
```

`uname -a`

```
Linux raspberrypi 6.1.45-v8+ #1671 SMP PREEMPT Fri Aug 11 14:49:25 BST 2023 aarch64 GNU/Linux
```

### Logs

[dmesg.txt](https://github.com/raspberrypi/linux/files/12353775/dmesg.txt)
[i2cdetect.txt](https://github.com/raspberrypi/linux/files/12353779/i2cdetect.txt)

6by9, 2023-08-16:
Duplicate of https://forums.raspberrypi.com/viewtopic.php?t=355034 by the sounds of it.
The 7inchC display connects to i2c-1 by default, so use ```dtoverlay=vc4-kms-dsi-waveshare-panel,7_0_inchC,i2c1``

Your dmesg log shows
```
[    9.328105] ws_touchscreen 10-0045: I2C write failed: -5
[    9.328303] ws_touchscreen 10-0045: I2C write failed: -5
[    9.328477] ws_touchscreen 10-0045: I2C write failed: -5
```
i2c-10 is the I2C bus up the FFC.

miklosakos, 2023-08-16:
Thank you this fixed it, I have another issue I would like to mention: when the system gets restarted / shut down, the backlight never gets turned off so the display will have this full white fill out that only goes away if the pi is power cycled fully.
[image]
[image]
[image]

bou6, 2023-08-18:
@6by9 You need to put the power pin to low, and then the display will switch off.

6by9, 2023-08-22:
> I have another issue I would like to mention: when the system gets restarted / shut down, the backlight never gets turned off so the display will have this full white fill out that only goes away if the pi is power cycled fully.

Is that any different from the Waveshare drivers? I suspect not, and there is very little that can be done about it.
Linux doesn't tell the graphics stack that it's shutting down, therefore the display drive is just removed. For HDMI that generally results in the display sleeping. For DSI it is down to the panel implementation, and many will end up just fading to white.

miklosakos, 2023-08-25:
> Is that any different from the Waveshare drivers?

Yes.
> I suspect not, and there is very little that can be done about it.

The waveshare driver turns the display off properly when the system gets shut down / rebooted.

6by9, 2023-08-30:
I don't have a 7" Waveshare panel. The 2.8", 10.1", and 7.9" all appear to shut off cleanly, so I can't debug with those.
If anyone else with a 7" panel wishes to analyse it to see how it achieves powering the backlight off, then PRs are welcome.

The open source drivers exposes the backlight control through the standard Linux backlight API. Systemd should be able to do the right thing via a suitable file in ```/etc/systemd/system/poweroff.target.wants```. There is already one in there for the (old) Pi 7" display as ```rpi-display-backlight.service```. 

For the Waveshare panel with open drivers, the contents wants to be along the lines of:
```
[Unit]
Description=Turns off Waveshare display backlight on shutdown/reboot
ConditionPathIsDirectory=/proc/device-tree/soc/i2c0mux/i2c@1/panel_disp1@45
DefaultDependencies=no
Before=umount.target

[Service]
Type=oneshot
ExecStart=/bin/sh -c '/bin/echo 1 > /sys/class/backlight/10-0045/bl_power'

[Install]
WantedBy=reboot.target halt.target poweroff.target

```
Same thing in ```/etc/systemd/system/reboot.target.wants```, and ```/etc/systemd/system/halt.target.wants```.

Pretty much the same thing wants to be added for the Pi 7" panel when using DRM drivers, so it is likely that they will get shipped with Raspberry Pi OS at some point. As Waveshare used the same I2C address as the Pi 7" panel, it may just work for that too.

Then again with the Waveshare 7" panel using i2c-1 rather than i2c-10, that scuppers things. For that one that paths are going to be
```
[Unit]
Description=Turns off Waveshare display backlight on shutdown/reboot
ConditionPathIsDirectory=/proc/device-tree/soc/i2c@7e804000/panel_disp1@45
DefaultDependencies=no
Before=umount.target

[Service]
Type=oneshot
ExecStart=/bin/sh -c '/bin/echo 1 > /sys/class/backlight/1-0045/bl_power'

[Install]
WantedBy=reboot.target halt.target poweroff.target
```

Do note that this will only turn the backlight off. The LCD element will still be going white, you just won't have the illumination behind it.

6by9, 2023-08-30:
Closing as the original issue is resolved (use i2c-1 rather than i2c-10).
Backlight needs more information from others, and should be a separate issue rather than being combined with this one.

closed 2023-08-30 by 6by9