raspberrypi/linux #5579: Raspberry Pi OS (64-bit kernel) uses kodi to play video kodi will freeze

Oliv1a, 2023-08-22:
### Describe the bug

My environment is Raspberry Pi 4 v1.2 hardware version, using the latest Raspberry Pi OS, the latest version of kodi installed through apt, the specific performance is that when using kodi to watch local video or streaming media for a long time, it will randomly freeze and can’t do any operation on kodi, and the kill -9 command can’t end the process of kodi stuck. I can only restart the system. 

### Steps to reproduce the behaviour

1. Install Raspberry pi OS (64-bit kernel) on Raspberry Pi 4B
2. Install kodi from raspberry pi official repository using apt install kodi kodi-pvr-iptvsimple
3. Connect the display device and use kodi to play local video or streaming media
4. Continue to play for a period of time, usually Kodi will be stuck for several hours at most one day

### Device (s)

Raspberry Pi 4 Mod. B

### System

cat /etc/rpi-issue
Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage4

vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

uname -a
Linux pi.local 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr 3 17:24:16 BST 2023 aarch64 GNU/Linux

### Additional context

The problem can be reproduced whether using the 64-bit kernel of the 32-bit system or the 64-bit system and whether using kms or fkms.

popcornmix, 2023-08-22:
Output of `sudo vclog -m`, `dmesg` and `~/.kodi/temp/kodi.log` after a freeze may have some clues.

What is the format of the videos being played when it freezes? (i.e. codec and resolution/framerate).
It may be useful to disable hw acceleration in kodi, and confirm if you still have the issue (but this will limit the resolution of videos that play well).

qrp73, 2023-08-23:
When freeze happens try to switch into tty1 with Ctrl+Alt+F1 and then back into desktop with Ctrl+Alt+F7. Does it fix freeze?

Oliv1a, 2023-09-29:
> Output of `sudo vclog -m`, `dmesg` and `~/.kodi/temp/kodi.log` after a freeze may have some clues.
>
> What is the format of the videos being played when it freezes? (i.e. codec and [...]

I recorded two freezes. The video format played was H.264/25FPS/1080p. The logs are as follows:

1.

vclog -m

004195.568: brfs: File read: /mfs/sd/config.txt
004196.631: brfs: File read: 2308 bytes
004217.351: brfs: File read: /mfs/sd/edid.dat
004230.925: HDMI1:EDID error reading EDID block 0 attempt 0
004231.941: HDMI1:EDID giving up on reading EDID block 0
004232.021: brfs: File read: 256 bytes
004238.558: brfs: File read: /mfs/sd/config.txt
004999.874: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
005002.397: *** Restart logging
005002.418: brfs: File read: 2308 bytes
005019.388: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
005020.406: hdmi: HDMI1:EDID giving up on reading EDID block 0
005025.445: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
005026.460: hdmi: HDMI1:EDID giving up on reading EDID block 0
005026.479: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
005026.494: HDMI0: hdmi_pixel_encoding: 300000000
005026.506: HDMI1: hdmi_pixel_encoding: 300000000
005027.020: kernel=
005048.848: dtb_file 'bcm2711-rpi-4-b.dtb'
005053.396: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
005053.413: Loaded 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xcd71
005066.512: brfs: File read: 52593 bytes
005075.641: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
005156.438: brfs: File read: 2347 bytes
005160.046: brfs: File read: /mfs/sd/config.txt
005160.592: dtparam: audio=on
005165.270: brfs: File read: 2308 bytes
005178.031: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
005240.006: Loaded overlay 'vc4-kms-v3d'
005436.816: brfs: File read: 3913 bytes
005441.216: brfs: File read: /mfs/sd/overlays/disable-wifi.dtbo
005462.802: Loaded overlay 'disable-wifi'
005497.413: brfs: File read: 387 bytes
005501.345: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
005538.580: Loaded overlay 'disable-bt'
005637.064: brfs: File read: 1073 bytes
005638.905: brfs: File read: /mfs/sd/cmdline.txt
005639.018: Read command line from file 'cmdline.txt':
005639.055: 'console=serial0,115200 console=tty1 root=PARTUUID=62367d8a-02 rootfstype=ext4 fsck.repair=yes rootwait usbhid.mousepoll=0 quiet splash plymouth.ignore-serial-consoles vc4.force_hotplug=1 video=HDMI-A-1:1360x768M@60D'
005961.289: brfs: File read: 215 bytes
006361.313: brfs: File read: /mfs/sd/kernel8.img
006361.348: Loaded 'kernel8.img' to 0x80000 size 0x7d6bd0
009127.555: Kernel relocated to 0x200000
009127.583: Device tree loaded to 0x2eff2a00 (size 0xd53a)
009134.696: bfs_xhci_stop
009134.711: XHCI-STOP
009135.136: xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
009135.197: USBSTS 18
011048.210: vchiq_core: vchiq_init_state: slot_zero = 0xcf000000, is_master = 1
011058.296: TV service:host side not connected, dropping notification 0x00000002, 0x00000002, 0x00000027
209156.619: video_decode:8:RIL: resolution changing
240543.593: video_decode:10:RIL: resolution changing
-1525995.-301: video_decode:13:RIL: resolution changing
974451.376: video_decode:16:RIL: resolution changing
1077202.361: video_decode:17:RIL: resolution changing
1634162.961: video_decode:18:RIL: resolution changing
-1713572.-223: video_decode:19:RIL: resolution changing
-1671497.-588: video_decode:20:RIL: resolution changing
-162943.-939: video_decode:21:RIL: resolution changing
-129077.-894: video_decode:22:RIL: resolution changing
[... 238 lines cut ...]
[ 4821.505877] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -62
[ 4824.577873] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 4824.577894] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -62
[ 4827.653882] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 4827.653904] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -62
[ 4830.721921] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 4830.721942] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -62
[ 4833.793940] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 4833.793961] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -62
[ 4836.865966] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 4836.866009] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -62
[ 4839.937990] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 4839.938018] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -62
[ 4843.010007] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 4843.010028] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -62
[ 4846.082048] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 4846.082069] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -62
[ 4849.154063] bcm2835_mmal_vchiq: timed out waiting for sync completion
[ 4849.154084] bcm2835-codec bcm2835-codec: bcm2835_codec_start_streaming: Failed enabling i/p port, ret -62

~/.kodi/temp/kodi.log

[https://paste.kodi.tv/afazogisot.kodi](https://paste.kodi.tv/afazogisot.kodi)

I then tried to disable hardware acceleration, but the video playback was severely stuck and could not be tested.

Oliv1a, 2023-09-29:
> When freeze happens try to switch into tty1 with Ctrl+Alt+F1 and then back into desktop with Ctrl+Alt+F7. Does it fix freeze?

There seems to be no response

popcornmix, 2023-09-29:
Does this reliably fail with a certain video?
If so, can you post a sample video file (e.g. a 1 or 2 minute cut of video to google drive or dropbox) that exhibits the issue?

The messages suggest the video is unusual and contains resolution changes.

Also it could be worth adding `over_voltage=2` to `/boot/config.txt` and rebooting and trying again.
We have had occasional reports of h264 fails that get fixed that way.

Oliv1a, 2023-09-29:
> Does this reliably fail with a certain video? If so, can you post a sample video file (e.g. a 1 or 2 minute cut of video to google drive or dropbox) that exhibits the issue?
>
> The messages suggest [...]

What I have observed for a long time here is that whether it is local video or streaming media, regardless of the video format and encoding, as long as you switch to a 64-bit kernel, the problem can generally be reproduced within 24 hours of continuous video playback. This problem has persisted since I started using Raspberry Pi 4 (5.x kernel). I don’t know if it’s related to my 1.2 hardware version or because my monitor resolution is 1366*768? But with the same hardware and configuration, this problem disappears as long as you switch to the 32-bit kernel, so I guessed whether there is something wrong with the 64-bit kernel. I would try adding over_voltage=2 to /boot/config.txt to test and see if that resolves the issue.

Oliv1a, 2023-11-21:
> Does this reliably fail with a certain video? If so, can you post a sample video file (e.g. a 1 or 2 minute cut of video to google drive or dropbox) that exhibits the issue?
>
> The messages suggest [...]

After a long period of observation and testing, it is confirmed that the problem has been solved, thank you very much!

closed 2023-11-21 by Oliv1a