raspberrypi/linux #5581: Kernel Panic After Running sudo apt update and sudo apt upgrade on Raspberry Pi 4 B 4GB RAM

mhalizadeh90, 2023-08-24:
### Describe the bug

After performing a clean install of the latest Raspberry Pi OS on a Raspberry Pi 4 Model B with 4GB RAM and a 256GB SanDisk Extreme PRO micro SD card, the system fails to boot after running `sudo apt update` and `sudo apt upgrade`. It ends up in a kernel panic with the following error messages:

```
[2.041108] Starting init: <sbin init exists but couldn't execute it (error -8)
[2.042236] Starting init: retc/init exists but couldn't execute it (error -13)
‣bin/sh: 0: cannot open splash: No such file
[2.140652] Kernel panic - not syncing: Attempted to kill init exitcode=0x00000200
```

### Steps to reproduce the behaviour

1. Perform a clean install of the latest Raspberry Pi OS.
2. Run `sudo apt update`.
3. Run `sudo apt upgrade`.
4. Reboot the Raspberry Pi.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Which OS and version: The latest Raspberry Pi OS downloaded from the official Raspberry Pi website

### Logs

Error messages during boot:

```
[2.041108] Starting init: <sbin init exists but couldn't execute it (error -8)
[2.042236] Starting init: retc/init exists but couldn't execute it (error -13)
‣bin/sh: 0: cannot open splash: No such file
[2.140652] Kernel panic - not syncing: Attempted to kill init exitcode=0x00000200
```

### Additional context

The Raspberry Pi is using a 256GB SanDisk Extreme PRO. The issue seems to occur only after running `sudo apt update` and `sudo apt upgrade`.

qrp73, 2023-08-25:
Did you tried to reproduce it with a new clean install? 

It may be due to card type, I hear that A2 sdcards have some extensions which is not supported on RPI sdcard reader.  I also have kernel error for sdcard initialization on linux Arch boot with SanSisk Extreme 64GB A2 sdcard...

bou6, 2023-08-25:
The meaning of the errors you get is as follow, for `sbin init`, execution format error and for `retc/init`, permission denied.
```
#define	ENOEXEC      8	/* Exec format error */
#define	EACCES		13	/* Permission denied */
```

Can you check the the format and and the permission before and after `apt upgrade` ?
Can you post a log with higher debugging level ?:

```
sudo nano /boot/cmdline.txt
Add Kernel Parameter:
cmdline=... other_parameters_here loglevel=7
```

closed 2023-08-25 by mhalizadeh90