raspberrypi/linux #5582: kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 195840)

qrp73, 2023-08-25:
### Describe the bug

There is a lot of errors in the log from bcm2835-codec

### Steps to reproduce the behaviour

Install Raspberry Pi OS aarch64 on Raspberry Pi 4B 4G. Boot it, open youtube and play some music, then open log with journalctl and scroll it down. Make sure there are a lot of errors from bcm2835-codec.

### Device (s)

Raspberry Pi 4 Mod. B

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage4

$ vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

$ uname -a
Linux raspi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### Logs

```
Aug 25 08:44:58 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 622080)
Aug 25 08:44:58 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 622080)
Aug 25 08:44:58 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 622080)
Aug 25 08:44:58 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 622080)
Aug 25 08:44:58 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 622080)
Aug 25 08:44:58 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 622080)
Aug 25 08:44:58 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 622080)
Aug 25 08:45:02 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 622080)
Aug 25 08:45:02 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 622080)
Aug 25 08:45:02 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 622080)
Aug 25 08:45:02 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 622080)
Aug 25 08:45:55 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1413120)
Aug 25 08:45:55 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1413120)
Aug 25 08:45:55 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1413120)
Aug 25 08:45:55 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1413120)
Aug 25 08:45:55 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1413120)
Aug 25 08:45:55 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1413120)
Aug 25 08:45:55 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1413120)
Aug 25 08:45:55 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1413120)
Aug 25 08:47:52 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 794880)
Aug 25 08:47:52 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 794880)
Aug 25 08:49:30 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1413120)
Aug 25 08:49:30 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1413120)
Aug 25 08:51:22 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1017600)
Aug 25 08:51:22 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1017600)
Aug 25 08:51:22 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1017600)
Aug 25 08:51:22 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1017600)
Aug 25 08:51:22 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1017600)
Aug 25 08:51:22 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 1017600)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
Aug 25 08:52:12 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 554496)
[... 122 lines cut ...]
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
[... cut ...]
Aug 25 21:50:38 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 161280)
...
Aug 25 22:22:21 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 195840)
Aug 25 22:22:21 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 195840)
Aug 25 22:22:21 raspi kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_buf_prepare data will not fit into plane (1536 < 195840)
```

6by9, 2023-08-30:
It's Chromium queuing buffers of an incorrect size.
Logging message priority dropped in #5588.

qrp73, 2023-11-15:
not relevant anymore

closed 2023-11-15 by qrp73