raspberrypi/linux #5586: Pi 4 Model B - USB, Split Transactions broken with kernel 6.1.21-v8+

raveslave, 2023-08-30:
### Describe the bug

When sending data through USB Hubs that doesn't have "buffers" (normally called non multi-tt) packages are incorrectly handled in the Raspberry Kernel.

It would be great to solve this issue as it is causing big performance issues when trying to utilize full bandwidth on RPI4 Hardware. Note that this happens already on 12Mbit (high-speed) usb-CDC's which will cause big penalties trying to achieve higher throughputs.

isse is that split transactions are not correctly handled.

referencing another issue from 2017 that describes the same issue in more detail:   
https://github.com/raspberrypi/linux/issues/2140#issuecomment-322785627

[image]

### Steps to reproduce the behaviour

Send continuous data through a Single TT Mode (TT/Hub) will drop packages as soon as a split transaction happen.

use a hub / hub-chip like Infineon CY7C65642   

[image]

### Device (s)

Raspberry Pi 4 Mod. B

### System

running official RPI-Distro/pi-gen
6.1.47-v8+ #1674 SMP PREEMPT Fri Aug 25 13:17:22 BST 2023

P33M, 2023-08-31:
>Steps to reproduce the behaviour
>Send continuous data through a Single TT Mode (TT/Hub) will drop packages as soon as a split transaction happen.

With what device attached?
RP2040 can act as a full-speed CDC device, can you create a testcase that uses e.g. picoprobe/debugprobe hardware to replicate what you see with your mystery widget?

raveslave, 2023-09-19:
will try and put together a repro!