raspberrypi/linux #5587: bcm2835-codec sound distortions issue

qrp73, 2023-08-30:
### Describe the bug

When sound volume is above 85% it leads to a sound distortion. Oscilloscope shows that there is amplitude clipping.

RPI4 output waveform at volume level 75%:
[image]

RPI4 output waveform at volume level 100%:
[image]

This clipping distortions is present for any load impedance (tested for open output, 57 kΩ and 32 Ω).

### Steps to reproduce the behaviour

1) Open audacity
2) Select Generate->Tone
3) Generate 200 Hz sine with amplitude 0.9999 amplitude
4) Play it
5) Use OS Volume Control to change volume from 50% to 100% and notice that there is distortion appears at about 86%

Expected result: clean sine sound at 100% volume level.

Actual result: sound distortions happens when volume level is more than 85%

### Device (s)

Raspberry Pi 4 Mod. B

### System

$ cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage4

$ vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

$ uname -a
Linux raspi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### Logs

there is no error in the log related to this bug

popcornmix, 2023-08-30:
The analogue audio on Pi boards is rather low level for headphones use which can result in too quiet audio for some headphones.

Because of this, we allow the audio to be overdriven by 4dB (see [here]( https://github.com/raspberrypi/linux/blob/655fc658a15ae7a6f37103754adb39ba52a9a14e/drivers/staging/vc04_services/bcm2835-audio/bcm2835-ctl.c#L12C9-L12C21)).

When going above 0, you will clip a full range signal. But if you know signal is not full range, you may choose to get the higher volume level.

When using alsamixer to set volume, it shows red when above 0dB and clipping can occur. The recommendation is to not go above that if you may be using full range signals.

qrp73, 2023-08-30:
how to disable this 4 dB overdrive? I don't need distorted gain. I want to set 100% volume and don't get clipping for max level signal.

qrp73, 2023-08-31:
well is it possible to disable dB gain above 0 dB in the driver? So that 100% volume bar in the desktop panel can be equals to 0 dB gain for bcm2835 and alsamixer will be unable to set gain above 0 dB for bcm2835 and alsamixer Master volume 100% can be equals to 0 dB gain for2835?

qrp73, 2023-11-15:
not relevant anymore

closed 2023-11-15 by qrp73