raspberrypi/linux #5590: Increase maximum number of serial 8250 UARTS

jpiwek, 2023-09-02:
### Describe the bug

Link to Rasp Forum: https://forums.raspberrypi.com/viewtopic.php?t=355689
additional 4 serial 8250 UART interfaces were registered automatically.
Simple fix to add to Kernel Configuration
Increase **CONFIG_SERIAL_8250_NR_UARTS** from 1 to at least 5

### Steps to reproduce the behaviour

Connect an PCIe to UART expansion card to RASPI IO-BOARD

### Device (s)

Raspberry Pi CM4, Raspberry Pi CM4 Lite

### System

**uname -a**
Kernel: Linux ipgw2 6.1.46-v8+ #5 SMP PREEMPT Sat Sep  2 08:38:08 CEST 2023 aarch64 GNU/Linux

**vcgencmd version**
Mar 17 2023 10:50:39
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

### Logs

dmesg:
[    1.489381] pci1xxxx serial 0000:06:00.0: Couldn't register serial port 0, irq 33, type 2, error -28
[    1.489427] pci1xxxx serial 0000:06:00.0: Couldn't register serial port 0, irq 34, type 2, error -28
[    1.489466] pci1xxxx serial 0000:06:00.0: Couldn't register serial port 0, irq 35, type 2, error -28

### Additional context

Simple fix:
I kindly ask to increase **CONFIG_SERIAL_8250_NR_UARTS** from 1 to at least 5

CONFIG_SERIAL_8250_NR_UARTS:                                                                                    
                                                                                                            
  x Set this to the number of serial ports you want the driver                                                      
  x to support.  This includes any ports discovered via ACPI or                                                     
  x PCI enumeration and any ports that may be added at run-time                                                  
  x via hot-plug, or any ISA multi-port serial cards.

pelwell, 2023-09-04:
The per-port static allocation appears to be 768 bytes (not massive, but not nothing) so I'm going to go with 5 as a reasonable number - 1 built-in and up to 4 extras. The higher value only makes sense on a PCIe equipped system, so I'll only change the bcm2711 defconfigs. See https://github.com/raspberrypi/linux/commit/16d5960fd08a6c35bfa370d8471e5f0b101d213b.

6by9, 2023-09-04:
Slightly odd that there is both CONFIG_SERIAL_8250_NR_UARTS to define the size of the array, and then CONFIG_SERIAL_8250_RUNTIME_UARTS (also exposed via the module parameter ```nr_uarts```) to define how many of those entries you can actually use. I don't quite understand why there is a need for having the two options.

Note that the default for both is 4 (and now I'm thinking of 0x3f8, 0x2f8, 0x3e8, and 0x2e8), and it would seem sensible for them to be set to the same value so there is no need to change the module parameter in order to make them work.

The help text for the module parameter says the range is "1-CONFIG_SERIAL_8250_NR_UARTS ", but 0 is then handled to disable the 8250 core totally, and the Kconfig text allows 0 - consistency is a lovely thing.

jpiwek, 2023-09-04:
It's odd... 
I set 
CONFIG_SERIAL_8250_NR_UARTS to 5
CONFIG_SERIAL_8250_RUNTIME_UARTS to 4 by adding 8250.nr_uarts=4 in cmdline.txt. This might give more flexibily to the user.