raspberrypi/linux #5591: Screen blanking turns off display

Quantamm, 2023-09-03:
### Describe the bug

On Stretch, I had my Pi configured to blank the display, but still send a display signal. I'm using a TV for a display and if it looses the display signal, it turns off completely and is a pain to get set back up again.

I just installed Bullseye and now, whenever the display is blanked, the display signal stops as well.

### Steps to reproduce the behaviour

```
xset s blank
xset s 5
xset -dpms
```

Then wait 5 seconds.

I also tried adding:

```
hdmi_blanking=0
disable_dpms=1
```

to /boot/config.txt, but doesn't help.

### Device (s)

Raspberry Pi 4 Mod. B

### System

pi@workout:~ $  cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage4
pi@workout:~ $ vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
pi@workout:~ $ uname -a
Linux workout 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

### Logs

No output is sent to `dmesg` when the screen blanks.

whitslack, 2023-09-09:
Maybe it's not X doing it? Since the VideoCore now has a proper KMS driver, it might be the kernel's virtual terminal driver that's powering off the display. Have you tried `setterm --powersave off >/dev/ttyX` (where `ttyX` is your active VT)?

(This is just a wild guess. Something to try.)

Quantamm, 2023-09-10:
At your suggestion, I tried that command, but I get:

> $ sudo setterm --powersave off 
> setterm: cannot (un)set powersave mode: Inappropriate ioctl for device

I get the same thing as a regular user. And redirecting the output to /dev/tty7 doesn't change anything either.