raspberrypi/linux #5592: Hardware playback devices randomly change positions with reboot, sometimes

JimZipCode, 2023-09-03:
### Describe the bug

I have a Hifiberry soundcard installed on a Raspberry Pi 4 Model B Rev 1.4 running Raspbian GNU/Linux 11 (bullseye).

Today, the output of aplay -l shows this:

```
root@raspberrypi4:/# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HiFiBerry Digi+ Pro HiFi wm8804-spdif-0 [HiFiBerry Digi+ Pro HiFi wm8804-spdif-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

So the Hifiberry is card 1, **_today_**.  But we had a power failure two days ago; before the power failure, the Hifiberry was card 2.  For some reason it was reassigned on reboot.

That "breaks" mpd when it happens.  My mpd.conf audio_output.device was set to "hw:2,0", and that worked.  Today no music was coming out.  I looked at the  aplay -l output and compared it to mpd.conf, and changed the setting from  "hw:2,0", to "hw:1,0", and that fixed the issue.

This is not the first time I've had to do this.  In June of 2022 the Hifiberry was card 0 (so my mpd.conf audio_output.device was set to "hw:2,0"), and then on one reboot it randomly moved to card 2.  That time it took me days (and some high-power help) to troubleshoot why I was getting no sound.  At least this time I knew what to look for.

It does **NOT** happen every reboot.  I've rebooted plenty of times from June 2022 til this week (including some power failures); it only just happened.  And then before that it was solid (at position 0) from Jan of 2021 until June of 2022.  It's like an every-sixteen-month timebomb.

I guess I want to know this:

- Why do my hardware playback devices sometimes get their positions reassigned?
- Could it be related to system updates?  I perform those infrequently.
- How can I stop it; make the position permanent so the Hifiberry is always card 2 or whatever?
- Or if I can't stop it, is there a way to designate the device in mpd.conf so even if the soundcard changes "position" between 0 and 1 and 2, mpd can still "find" it?  Maybe use the name instead or something?

uname -a output shows this:

```
root@raspberrypi4:/# uname -a
Linux raspberrypi4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

### Steps to reproduce the behaviour

Inconsistent.  Randomly after ~16 months the aplay -l output shows that my Hifiberry soundcard has "moved" from position 0 to position 2 to position 1.

### Device (s)

Raspberry Pi 4 Mod. B

### System

root@raspberrypi4:/# raspinfo
System Information
------------------

Raspberry Pi 4 Model B Rev 1.4
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 226b479f8d32919c9fe36dd5b4c20c02682f8180, stage5

Linux raspberrypi4 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Revision	: c03114
Serial		: 100000005bb5753d
Model		: Raspberry Pi 4 Model B Rev 1.4
Throttled flag  : throttled=0x0
[... 674 lines cut ...]
[    8.521153] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    8.554335] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[    8.664979] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    8.671430] bcmgenet fd580000.ethernet eth0: Link is Down
[   12.767174] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   12.767218] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   14.264696] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[   14.274464] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[   18.402844] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead
[ 7868.848009] bcm2835-i2s fe203000.i2s: I2S SYNC error!
[ 8266.433643] bcm2835-i2s fe203000.i2s: I2S SYNC error!

### Additional context

root@raspberrypi4:/# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: sndrpihifiberry [snd_rpi_hifiberry_digi], device 0: HiFiBerry Digi+ Pro HiFi wm8804-spdif-0 [HiFiBerry Digi+ Pro HiFi wm8804-spdif-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

HiassofT, 2023-09-03:
Reference cards by name, not by number (which is not guaranteed to be stable as you already know).

look at the output of `aplay -L` and use the PCM device of your card (eg `hdmi:CARD=vc4hdmi0,DEV=0` or likely `hw:CARD=sndrpihifiberry` in your case)

JimZipCode, 2023-09-03:
Thanks!
Changed mpd.conf 
**From**:
```
audio_output {
        type            "alsa"
        name            "My ALSA Device"
        device          "hw:1,0"        # optional
        mixer_type      "software"      # optional
}
```

**To**:
```
audio_output {
        type            "alsa"
        name            "My ALSA Device"
        device          "hw:CARD=sndrpihifiberry,DEV=0" # optional
        mixer_type      "software"      # optional
}
```
And it works!
Nice thing about this solution, is that I should never notice if it's working or not.  👍  
Appreciate the help!

closed 2023-09-03 by JimZipCode