raspberrypi/linux #5594: uart5 (randomly?) becomes ttyAMA1 on kernel 6.1.y

BubuOT, 2023-09-06:
### Describe the bug

After switching our linux kernel to the 6.1 branch (currently on 6.1.46) we noticed that the ttyAMA<n> numbering changed and now corresponds to the hardware uart number. I.e. uart3 -> ttyAMA3, uart5 -> ttyAMA5. (see https://forums.raspberrypi.com/viewtopic.php?t=347868 and https://github.com/raspberrypi/linux/pull/5436)

So far so good, but recently we noticed that uart5 gets enumerated as ttyAMA1 now, without any further software change.

Here's the dmesg excerpt:
```
dmesg | grep ttyAMA
[...]
[    0.181829] fe201000.serial: ttyAMA0 at MMIO 0xfe201000 (irq = 33, base_baud = 0) is a PL011 rev2
[    0.421900] printk: console [ttyAMA0] enabled
[    0.593863] fe201400.serial: ttyAMA2 at MMIO 0xfe201400 (irq = 33, base_baud = 0) is a PL011 rev2
[    0.602932] fe201600.serial: ttyAMA3 at MMIO 0xfe201600 (irq = 33, base_baud = 0) is a PL011 rev2
[    0.817135] fe201a00.serial: ttyAMA1 at MMIO 0xfe201a00 (irq = 33, base_baud = 0) is a PL011 rev2
```

0xfe201a00 is defnitely uart5 in the device tree:
```
$ dtc -I fs /sys/firmware/devicetree/base | grep serial@
		serial@7e201a00 {
		serial@7e201800 {
		serial@7e201400 {
		serial@7e215040 {
		serial@7e201000 {
		serial@7e201600 {
		serial5 = "/soc/serial@7e215040";
		uart4 = "/soc/serial@7e201800";
		serial3 = "/soc/serial@7e201600";
		uart2 = "/soc/serial@7e201400";
		serial1 = "/soc/serial@7e215040";
		uart0 = "/soc/serial@7e201000";
		bluetooth = "/soc/serial@7e201000/bluetooth";
		uart5 = "/soc/serial@7e201a00";
		serial4 = "/soc/serial@7e201800";
		uart3 = "/soc/serial@7e201600";
		serial2 = "/soc/serial@7e201400";
		uart1 = "/soc/serial@7e215040";
		serial0 = "/soc/serial@7e201000";
		bt = "/soc/serial@7e201000/bluetooth";
		uart4 = "/soc/serial@7e201800";
		minibt = "/soc/serial@7e215040/bluetooth";
		uart2 = "/soc/serial@7e201400";
		uart0 = "/soc/serial@7e201000";
		uart5 = "/soc/serial@7e201a00";
		uart3 = "/soc/serial@7e201600";
		uart1 = "/soc/serial@7e215040";
```

Any idea what could be going wrong here?

### Steps to reproduce the behaviour

Unclear so far.

### Device (s)

Raspberry Pi CM4

### System

Custom OS built with buildroot.
firmware: https://github.com/raspberrypi/firmware/commit/74ad049f4c1502595721e184f0318633aaa1552a
kernel: 9ed4f05ba2e2bcd9065831674e97b2b1283e866d

BubuOT, 2023-09-06:
Okay, so it's definitely not the dtb file we actually ship in our boot partition:
```
$ dtc -I dtb /tmp/boot/bcm2711-rpi-cm4.dtb | grep serial@
		serial0 = "/soc/serial@7e215040";
		serial1 = "/soc/serial@7e201000";
		bluetooth = "/soc/serial@7e201000/bluetooth";
		uart0 = "/soc/serial@7e201000";
		uart1 = "/soc/serial@7e215040";
		uart2 = "/soc/serial@7e201400";
		uart3 = "/soc/serial@7e201600";
		uart4 = "/soc/serial@7e201800";
		uart5 = "/soc/serial@7e201a00";
		serial2 = "/soc/serial@7e201400";
		serial3 = "/soc/serial@7e201600";
		serial4 = "/soc/serial@7e201800";
		serial5 = "/soc/serial@7e201a00";
		serial@7e201000 {
		serial@7e215040 {
		serial@7e201400 {
		serial@7e201600 {
		serial@7e201800 {
		serial@7e201a00 {
		uart0 = "/soc/serial@7e201000";
		bt = "/soc/serial@7e201000/bluetooth";
		uart1 = "/soc/serial@7e215040";
		minibt = "/soc/serial@7e215040/bluetooth";
		uart2 = "/soc/serial@7e201400";
		uart3 = "/soc/serial@7e201600";
		uart4 = "/soc/serial@7e201800";
		uart5 = "/soc/serial@7e201a00";
```
```
so here's our config.txt, in case that helps:
$ cat /tmp/boot/config.txt 
kernel=u-boot.bin
gpu_mem=16

# enable 64bits support
arm_64bit=1

[cm4]
otg_mode=1

[pi4]
arm_boost=1

[all]
dtparam=watchdog=on
dtparam=i2c_arm=on
dtparam=i2c1=on
dtoverlay=i2c-rtc,pcf85063a
dtoverlay=i2c-rtc,pcf8523
disable_splash=1
camera_auto_detect=0
display_auto_detect=0
boot_delay=0
dtoverlay=disable-bt
dtoverlay=i2c1,pins_44_45
dtparam=spi=on
dtoverlay=uart2
dtoverlay=uart3
dtoverlay=uart5

# hardware button
gpio=26=ip,pu

# 1wire over gpio
dtoverlay=w1-gpio,gpiopin=19

start_cd=1
initial_turbo=10
disable_poe_fan=1
force_eeprom_read=0
```

pelwell, 2023-09-07:
...unless it's a stupid firmware bug. Which it is.

Grab the trial firmware from this link: https://drive.google.com/file/d/1xDBZfgwveKbmunzUzcLeVCmGIyqUjj_O/view?usp=sharing
It just contains `start4cd.elf` and `fixup4cd.dat`. Copy them into /boot and reboot - the `serial5` alias should be correct.

BubuOT, 2023-09-07:
> Copy them into /boot and reboot - the serial5 alias should be correct.

This works correctly, yes.

pelwell, 2023-09-07:
Cool - the fix has been merged, and will be in future firmware releases. For now you can continue to use that trial version.

BubuOT, 2023-09-07:
Is there a changelog for the firmware releases that I can follow?

pelwell, 2023-09-07:
All releases appear here (https://github.com/raspberrypi/rpi-firmware/commits/master), with stable releases here (https://github.com/raspberrypi/rpi-firmware/commits/stable) - significant changes are listed in the commit messages (click on the ...s).

BubuOT, 2023-10-05:
@pelwell As far as I can see there's been a new firmware release here: https://github.com/raspberrypi/rpi-firmware/commit/6f1dcc758829cfd56d7c2f16513aaefa876902b2

Does this include the changes fixing this issue?

pelwell, 2023-10-05:
If it contains the hat_map feature (which it does) then yes, it will have the fix mentioned about.

BubuOT, 2023-10-05:
Thanks for confirming!

Closing this then.

closed 2023-10-05 by BubuOT