raspberrypi/linux #5598: [rpi-6.1.y] Please backport fix for F2FS NULL pointer dereference in `f2fs_issue_flush`

whitslack, 2023-09-09:
### Describe the bug

Linux 6.1.x (up through 6.1.52) suffers from a crash bug affecting F2FS. A [fix](https://lore.kernel.org/linux-f2fs-devel/20221230154332.5082-1-chao@kernel.org/) for this bug was [accepted](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=b3d83066cbebc76dbac8a5fca931f64b4c6fff34) into the stable kernel for the 6.2.y series, but it was never backported to the 6.1.y series.

### Steps to reproduce the behaviour

The issue occurs at an undetermined time while accessing a mounted F2FS volume. It does not cause an immediate kernel hang, but it does log an "oops" report to the kernel log when it occurs. Then, some time later, the system will hang, often while processing an `fstrim` command, but also sometimes apparently spontaneously without an obvious trigger. When it occurs, the filesystem is left in an inconsistent state, and recent changes are lost.

### Device (s)

Raspberry Pi 1 Mod. B

### System

Which OS and version (`cat /etc/rpi-issue`)?
`cat: /etc/rpi-issue: No such file or directory`
(Gentoo Linux ~arm)

Which firmware version (`vcgencmd version`)?
`Aug 10 2023 15:35:01`  
`Copyright (c) 2012 Broadcom`  
`version 03dc77429335caee083e22ddc8eec09c07f12a7a (clean) (release) (start)`

Which kernel version (`uname -a`)?
`Linux hifipi 6.1.51+ #1676 Tue Sep  5 14:21:23 BST 2023 armv6l ARMv6-compatible processor rev 7 (v6l) BCM2835 GNU/Linux`

### Logs

`Linux version 6.1.45+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1671 Fri Aug 11 14:41:33 BST 2023`
```
[Fri Sep  8 19:20:44 2023] 8<--- cut here ---
[Fri Sep  8 19:20:44 2023] Unable to handle kernel NULL pointer dereference at virtual address 00000010
[Fri Sep  8 19:20:44 2023] [00000010] *pgd=00000000
[Fri Sep  8 19:20:44 2023] Internal error: Oops: 5 [#1] ARM
[Fri Sep  8 19:20:44 2023] Modules linked in: mac80211 vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper drm_kms_helper libarc4 sha256_generic libsha256 drm cfg80211 drm_panel_orientation_quirks snd_soc_core bcm2835_codec(C) snd_compress snd_pcm_dmaengine syscopyarea rfkill bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem sysfillrect sysimgblt fb_sys_fops backlight bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon videobuf2_common snd_bcm2835(C) snd_pcm videodev binfmt_misc i2c_bcm2835 snd_timer snd mc vc_sm_cma(C) uio_pdrv_genirq fixed uio
[Fri Sep  8 19:20:44 2023] CPU: 0 PID: 1658 Comm: seedrng Tainted: G         C         6.1.45+ #1671
[Fri Sep  8 19:20:44 2023] Hardware name: BCM2835
[Fri Sep  8 19:20:44 2023] PC is at f2fs_issue_flush+0x150/0x1fc
[Fri Sep  8 19:20:44 2023] LR is at f2fs_do_sync_file+0x694/0x78c
[Fri Sep  8 19:20:44 2023] pc : [<c042c49c>]    lr : [<c03df894>]    psr: 40000013
[Fri Sep  8 19:20:44 2023] sp : d8d51e98  ip : c18d3700  fp : 00000000
[Fri Sep  8 19:20:44 2023] r10: c0c8702c  r9 : 0000b3e2  r8 : 00000000
[Fri Sep  8 19:20:44 2023] r7 : c0c8702c  r6 : c18f0000  r5 : 00000010  r4 : 00000000
[Fri Sep  8 19:20:44 2023] r3 : 00000000  r2 : 100023bc  r1 : 0000b3e2  r0 : c18f0000
[Fri Sep  8 19:20:44 2023] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[Fri Sep  8 19:20:44 2023] Control: 00c5387d  Table: 0342c008  DAC: 00000055
[Fri Sep  8 19:20:44 2023] Register r0 information: slab kmalloc-4k start c18f0000 pointer offset 0 size 4096
[Fri Sep  8 19:20:44 2023] Register r1 information: non-paged memory
[Fri Sep  8 19:20:44 2023] Register r2 information: non-paged memory
[Fri Sep  8 19:20:44 2023] Register r3 information: NULL pointer
[Fri Sep  8 19:20:44 2023] Register r4 information: NULL pointer
[Fri Sep  8 19:20:44 2023] Register r5 information: zero-size pointer
[Fri Sep  8 19:20:44 2023] Register r6 information: slab kmalloc-4k start c18f0000 pointer offset 0 size 4096
[Fri Sep  8 19:20:44 2023] Register r7 information: non-slab/vmalloc memory
[Fri Sep  8 19:20:44 2023] Register r8 information: NULL pointer
[Fri Sep  8 19:20:44 2023] Register r9 information: non-paged memory
[Fri Sep  8 19:20:44 2023] Register r10 information: non-slab/vmalloc memory
[Fri Sep  8 19:20:44 2023] Register r11 information: NULL pointer
[Fri Sep  8 19:20:44 2023] Register r12 information: slab kmalloc-128 start c18d3700 pointer offset 0 size 128
[Fri Sep  8 19:20:44 2023] Process seedrng (pid: 1658, stack limit = 0xea452f42)
[Fri Sep  8 19:20:44 2023] Stack: (0xd8d51e98 to 0xd8d52000)
[Fri Sep  8 19:20:44 2023] 1e80:                                                       00000000 c03eb500
[Fri Sep  8 19:20:44 2023] 1ea0: 00000000 00000014 c18f0134 c18f0000 00000000 cfc4b745 00000000 c1e94d70
[Fri Sep  8 19:20:44 2023] 1ec0: c18f0000 00000000 d8d51eec 00000000 c0c8702c c03df894 d8d51eec 7fffffff
[Fri Sep  8 19:20:44 2023] 1ee0: c0c8702c 0000b3e2 be857590 00000000 7fffffff 00000000 00000000 00000000
[Fri Sep  8 19:20:44 2023] 1f00: 00000000 00000000 00000001 00000000 00000000 00000000 00000000 00000000
[Fri Sep  8 19:20:44 2023] 1f20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 cfc4b745
[Fri Sep  8 19:20:44 2023] 1f40: 00000000 ffffffff 7fffffff 00000000 00000076 c0008408 c3001040 00000000
[Fri Sep  8 19:20:44 2023] 1f60: 00000000 c03df9d0 ffffffff 7fffffff 00000000 00000000 c0008408 c21ef000
[Fri Sep  8 19:20:44 2023] 1f80: c21ef000 c0254250 ffffffff 7fffffff 00000000 00000004 00000020 be8575b0
[Fri Sep  8 19:20:44 2023] 1fa0: 00000020 c0008260 00000020 be8575b0 00000004 be857590 00000020 00000001
[Fri Sep  8 19:20:44 2023] 1fc0: 00000020 be8575b0 00000020 00000076 00000003 00000001 00000004 00000000
[Fri Sep  8 19:20:44 2023] 1fe0: 0045200c be8574d0 0044e218 b6f37eac 20000010 00000004 00000000 00000000
[Fri Sep  8 19:20:44 2023]  f2fs_issue_flush from f2fs_do_sync_file+0x694/0x78c
[Fri Sep  8 19:20:44 2023]  f2fs_do_sync_file from f2fs_sync_file+0x44/0x58
[Fri Sep  8 19:20:44 2023]  f2fs_sync_file from do_fsync+0x44/0x78
[Fri Sep  8 19:20:44 2023]  do_fsync from ret_fast_syscall+0x0/0x1c
[Fri Sep  8 19:20:44 2023] Exception stack(0xd8d51fa8 to 0xd8d51ff0)
[Fri Sep  8 19:20:44 2023] 1fa0:                   00000020 be8575b0 00000004 be857590 00000020 00000001
[Fri Sep  8 19:20:44 2023] 1fc0: 00000020 be8575b0 00000020 00000076 00000003 00000001 00000004 00000000
[Fri Sep  8 19:20:44 2023] 1fe0: 0045200c be8574d0 0044e218 b6f37eac
[Fri Sep  8 19:20:44 2023] Code: 1afffffa e59d0018 eaffffd9 f5d5f000 (e1953f9f) 
[Fri Sep  8 19:20:44 2023] ---[ end trace 0000000000000000 ]---
```

pelwell, 2023-09-09:
See #5599 for my attempt - it needed some fettling to make it fit, so please test it if you can.

pelwell, 2023-09-09:
Note that you can install a trial build (with all the usual caveats) by waiting an hour or so and running `sudo rpi-update pulls/5599`.

whitslack, 2023-09-09:
Thanks. I'll test this tonight.

whitslack, 2023-09-10:
@pelwell: I have not been able to reproduce the crash while running `pulls/5599` and subjecting the system to the usage pattern that would always crash it before, so I think we can call this one solved.

closed 2023-09-10 by pelwell