raspberrypi/linux #5602: Kernel 6.6 with RT patch does not allow selection of RT config

by, 2023-09-13:
### Describe the bug

Thank you for the fast provisioning of kernel 6.6!

Having patched the current kernel 6.6 with the respective RT patch (i.e., https://mirrors.edge.kernel.org/pub/linux/kernel/projects/rt/6.6/), it is no longer possible to select +PREEMPT_RT y, it is simply not offered in make menufconfig.

Thanks for looking into this!

### Steps to reproduce the behaviour

See for example https://github.com/by/RT-Kernel

### Device (s)

Raspberry Pi 4 Mod. B

### System

Debian 12 (bookworm) 64 bit

popcornmix, 2023-09-13:
Yes, we've noticed a number of config options are now disabled (when using the defconfig from 6.5).
You may want to try adding `CONFIG_EXPERT=y` to the defconfig as that seemed to restore some options - but I don't have enough information to explain why yet.

popcornmix, 2023-09-13:
I think this is due to [this](https://lore.kernel.org/lkml/CAHmME9pxC_e6Ftj1MJ4H2skhhKfNubd=7OwcH9xepCxjkDxpkg@mail.gmail.com/T/)

I've switched from `CONFIG_EMBEDDED` to `CONFIG_EXPERT` (as seems to have occurred in other defconfig files) and I believe we avoid losing a number of  options. Does that help your build?

by, 2023-09-13:
Thanks fro your quick help! – But no, that didn't help, unfortunately.

by, 2023-09-14:
Since commit d923061 is does work again, thank you very much!

closed 2023-09-14 by by