raspberrypi/linux #5606: "make dtbs" with "ARCH=arm64" causes errors

tinyreminder, 2023-09-18:
### Describe the bug

Building device tree blobs with "ARCH=arm64" option causes errors.

### Steps to reproduce the behaviour

$ make ARCH=arm64 bcm2711_defconfig

$ make ARCH=arm64 dtbs

### Device (s)

Raspberry Pi 4 Mod. B

### System

Building environment:
Hardware: MacBook Air M2 (AArch64 system)
OS: Asahi Linux
Compiler: GCC 12.1.0
Linker etc: binutils 2.38

Raspberry Pi Linux kernel: 6.1.53

### Logs

  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  LEX     scripts/dtc/dtc-lexer.lex.c
  YACC    scripts/dtc/dtc-parser.tab.[ch]
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTLD  scripts/dtc/dtc
  HOSTCC  scripts/dtc/libfdt/fdt.o
  HOSTCC  scripts/dtc/libfdt/fdt_ro.o
  HOSTCC  scripts/dtc/libfdt/fdt_wip.o
  HOSTCC  scripts/dtc/libfdt/fdt_sw.o
  HOSTCC  scripts/dtc/libfdt/fdt_rw.o
  HOSTCC  scripts/dtc/libfdt/fdt_strerror.o
  HOSTCC  scripts/dtc/libfdt/fdt_empty_tree.o
  HOSTCC  scripts/dtc/libfdt/fdt_addresses.o
  HOSTCC  scripts/dtc/libfdt/fdt_overlay.o
  HOSTCC  scripts/dtc/fdtoverlay.o
  HOSTLD  scripts/dtc/fdtoverlay
  UPD     include/config/kernel.release
  DTC     arch/arm64/boot/dts/broadcom/bcm2711-rpi-400.dtb
In file included from arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/bcm2711.dtsi:2,
                 from arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/bcm2711-rpi-400.dts:5,
                 from arch/arm64/boot/dts/broadcom/bcm2711-rpi-400.dts:1:
arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/bcm283x.dtsi:1:10: fatal error: dt-bindings/pinctrl/bcm2835.h: No such file or directory
    1 | #include <dt-bindings/pinctrl/bcm2835.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.lib:407: arch/arm64/boot/dts/broadcom/bcm2711-rpi-400.dtb] Error 1
make[1]: *** [scripts/Makefile.build:500: arch/arm64/boot/dts/broadcom] Error 2
make: *** [Makefile:1478: dtbs] Error 2
tinyreminder@Starlight:~/Development/RPiKernel/linux-rpi-6.1.y 2$ make dtbs
  DTC     arch/arm64/boot/dts/broadcom/bcm2711-rpi-400.dtb
In file included from arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/bcm2711.dtsi:2,
                 from arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/bcm2711-rpi-400.dts:5,
                 from arch/arm64/boot/dts/broadcom/bcm2711-rpi-400.dts:1:
arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/bcm283x.dtsi:1:10: fatal error: dt-bindings/pinctrl/bcm2835.h: No such file or directory
    1 | #include <dt-bindings/pinctrl/bcm2835.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.lib:407: arch/arm64/boot/dts/broadcom/bcm2711-rpi-400.dtb] Error 1
make[1]: *** [scripts/Makefile.build:500: arch/arm64/boot/dts/broadcom] Error 2
make: *** [Makefile:1478: dtbs] Error 2

pelwell, 2023-09-18:
It's fine for me:
```
$ git show 
commit e079555a4c68356e58249cfc041b28f6eb455bd5 (HEAD -> rpi-6.1.y, origin/rpi-6.1.y)
Author: Stefan Wahren <wahrenst@gmx.net>
Date:   Sat May 4 17:06:15 2019 +0200

    hwrng: iproc-rng200: Add BCM2838 support
    
    The HWRNG on the BCM2838 is compatible to iproc-rng200, so add the
    support to this driver instead of bcm2835-rng.
...
$ make ARCH=arm64 clean
  CLEAN   arch/arm64/boot
  CLEAN   arch/arm64/crypto
  CLEAN   arch/arm64/kernel/pi
  CLEAN   arch/arm64/kernel/vdso
  CLEAN   arch/arm64/kernel/vdso32
  CLEAN   arch/arm64/kernel
  CLEAN   arch/arm64/kvm/hyp/nvhe
  CLEAN   arch/arm64/kvm
  CLEAN   certs
  CLEAN   drivers/accessibility/speakup
  CLEAN   drivers/firmware/efi/libstub
  CLEAN   drivers/scsi
  CLEAN   drivers/tty/vt
  CLEAN   drivers/video/logo
  CLEAN   init
  CLEAN   kernel/debug/kdb
  CLEAN   kernel
  CLEAN   lib/raid6
  CLEAN   lib
  CLEAN   net/wireless
  CLEAN   security/apparmor
  CLEAN   usr/include
  CLEAN   usr
  CLEAN   .
  CLEAN   vmlinux.symvers modules.builtin modules.builtin.modinfo .vmlinux.objs .vmlinux.export.c
$ make ARCH=arm64 bcm2711_defconfig
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
$ make ARCH=arm64 dtbs
  SYNC    include/config/auto.conf.cmd
arch/arm64/Makefile:36: Detected assembler with broken .inst; disassembly will be unreliable
  UPD     include/config/kernel.release
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  LEX     scripts/dtc/dtc-lexer.lex.c
  YACC    scripts/dtc/dtc-parser.tab.[ch]
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/dtc/dtc-parser.tab.o
[... 290 lines cut ...]
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-dpi-hyperpixel2r.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-dpi-hyperpixel4.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-dpi-hyperpixel4sq.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-dpi-panel.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-dsi-7inch.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-dsi-lt070me05000.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-dsi-lt070me05000-v2.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-dsi-waveshare-panel.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-kippah-7inch.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-v3d.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-v3d-pi4.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vc4-kms-vga666.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vga666.dtbo
  DTCO    arch/arm64/boot/dts/overlays/vl805.dtbo
  DTCO    arch/arm64/boot/dts/overlays/w1-gpio.dtbo
  DTCO    arch/arm64/boot/dts/overlays/w1-gpio-pullup.dtbo
  DTCO    arch/arm64/boot/dts/overlays/w5500.dtbo
  DTCO    arch/arm64/boot/dts/overlays/watterott-display.dtbo
  DTCO    arch/arm64/boot/dts/overlays/waveshare-can-fd-hat-mode-a.dtbo
  DTCO    arch/arm64/boot/dts/overlays/waveshare-can-fd-hat-mode-b.dtbo
  DTCO    arch/arm64/boot/dts/overlays/wittypi.dtbo
  DTCO    arch/arm64/boot/dts/overlays/wm8960-soundcard.dtbo
  DTC     arch/arm64/boot/dts/overlays/overlay_map.dtb
$
```

tinyreminder, 2023-09-19:
Thank you for confirming.

It seems that the default Asahi Linux unarchiver was corrupting the directory structure of linux-rpi-6.1.y.zip. unzip command works fine.

$ unzip linux-rpi-6.1.y.zip && cd linux-rpi-6.1.y
$ make ARCH=arm64 bcm2711_defconfig
$ make ARCH=arm64 dtbs

Sorry for bothering with uncertain information.

closed 2023-09-19 by tinyreminder

hemixs, 2024-03-25:
> Sorry for bothering with uncertain information.

thanks, that's save me 1-2 hrs of searching for root cause. I also copied sources with tar zipper.