raspberrypi/linux #5608: 32-bit 6.5.y downstream kernel does not boot on Pi3

mdiewa, 2023-09-19:
### Describe the bug

When trying to boot a 32-bit kernel on Pi3 (from USB) based on the latest `rpi-6.5.y` branch, the kernel will oops and hang. Building a kernel from the same sources for Pi4 (also 32-bit) and booting a Pi4 works fine. Setting `dtoverlay=dwc2,dr_mode=host` in `config.txt` does not avoid the issue.

### Steps to reproduce the behaviour

- Build 32-bit kernel from `rpi-6.5.y` branch (`18f87ed77f3bc07dbf594511889e8ee36d6c2004`) for Pi3 and install it
- Connect to serial console
- Try to boot Pi3 from an USB medium

### Device (s)

Raspberry Pi 3 Mod. B

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage2
```
```
$ vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
```
Kernel on Pi4:
```
$ uname -a
Linux raspberrypi 6.5.3-v7l+ #2 SMP Tue Sep 19 12:46:48 CEST 2023 armv7l GNU/Linux

### Logs

```
[    2.244157] Unable to handle kernel NULL pointer dereference at virtual address 00000004 when read
[    2.253434] [00000004] *pgd=00000000
[    2.257150] Internal error: Oops: 5 [#1] SMP ARM
[    2.261938] Modules linked in:
[    2.265116] CPU: 1 PID: 35 Comm: kworker/u8:1 Not tainted 6.5.3-v7+ #1
[    2.271881] Hardware name: BCM2835
[    2.275408] Workqueue: events_unbound deferred_probe_work_func
[    2.281467] PC is at log_event_impl.part.0+0x44/0x84
[    2.286625] LR is at log_event_impl.part.0+0x28/0x84
[    2.291780] pc : [<808e2238>]    lr : [<808e221c>]    psr: 60000093
[    2.298269] sp : bb8b5ce8  ip : 3c534f49  fp : 00000000
[    2.303680] r10: bb8b5dd4  r9 : 0ccccb60  r8 : 817e0ba4
[    2.309095] r7 : 80ea3fc8  r6 : 00000000  r5 : 00000000  r4 : 812e03f8
[    2.315855] r3 : 9ed01000  r2 : 00000000  r1 : a0000093  r0 : a0000093
[    2.322614] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[    2.330096] Control: 10c5383d  Table: 0000406a  DAC: 00000055
[    2.336044] Register r0 information: 0-page vmalloc region starting at 0x9ec00000 allocated at iotable_init+0x0/0xf4
[    2.346960] Register r1 information: 0-page vmalloc region starting at 0x9ec00000 allocated at iotable_init+0x0/0xf4
[    2.357870] Register r2 information: NULL pointer
[    2.362748] Register r3 information: 0-page vmalloc region starting at 0x9ec00000 allocated at iotable_init+0x0/0xf4
[    2.373658] Register r4 information: non-slab/vmalloc memory
[    2.379523] Register r5 information: NULL pointer
[    2.384401] Register r6 information: NULL pointer
[    2.389278] Register r7 information: non-slab/vmalloc memory
[    2.397831] Register r8 information: slab kmalloc-2k start 817e0800 pointer offset 932 size 2048
[    2.412399] Register r9 information: non-paged memory
[    2.420308] Register r10 information: 2-page vmalloc region starting at 0xbb8b4000 allocated at kernel_clone+0x9c/0x374
[    2.437006] Register r11 information: NULL pointer
[    2.444787] Register r12 information: non-paged memory
[    2.452867] Process kworker/u8:1 (pid: 35, stack limit = 0x9feb9fff)
[    2.462200] Stack: (0xbb8b5ce8 to 0xbb8b6000)
[    2.469480] 5ce0:                   817e0800 817e0a24 40000013 817e0b80 817e0ba4 808e4540
[    2.483631] 5d00: 00000001 817e0800 817e0ba4 808c66e0 817e0800 817e0800 817e0a24 00000001
[    2.498000] 5d20: 817e0800 808c3f40 817e0800 00000000 80ebf420 817e0800 817e0ba4 808c4e4c
[    2.512620] 5d40: 817e0800 808c5ecc 817e0b80 00000000 80ebf420 808e4fd8 8171a400 817e0b80
[    2.527475] 5d60: 00000002 3f202040 00000000 00000004 00000004 dec00000 dec00008 808e9cd0
[    2.542416] 5d80: bab18310 8171a640 8171a640 00000000 0000000c bb8b5dd4 8165ce00 00038042
[    2.557360] 5da0: 8171a64c d86548fd 817e0800 817e0800 812e03f8 81631000 817e0b80 00000000
[    2.572344] 5dc0: 00000000 808e54a8 00000000 bb8b5dd4 81655600 00000000 0001dc9a 0002fa91
[    2.587466] 5de0: 81631010 81232d80 00000000 00000100 815fa400 808ff5d8 ffffffff 81657898
[    2.602742] 5e00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 d86548fd
[    2.618438] 5e20: 00000000 81631010 81232d80 00000000 00000100 815fa400 814c600d 807c1e84
[    2.634314] 5e40: 81631010 00000000 81232d80 807bf6ec 81631010 81232d80 81631010 0000000e
[    2.650312] 5e60: 00000100 807bf98c 60000013 80b1d140 812db5d4 81232d80 81631010 0000000e
[    2.666382] 5e80: 00000100 807bfad8 00000001 81232d80 bb8b5ee4 81631010 00000100 807bfc48
[    2.682738] 5ea0: 00000000 8151f200 bb8b5ee4 807bfbac 00000100 807bd94c 81630494 8151f26c
[    2.699219] 5ec0: 81f207b8 d86548fd fffffff3 81631010 00000001 81631054 8151f200 807c002c
[    2.715772] 5ee0: 81631010 81631010 00000001 d86548fd 81631010 81631010 81209d18 807be818
[    2.732325] 5f00: 81631010 81209d04 81209d18 00000000 00000100 807becb0 81209d3c 816ca400
[    2.748878] 5f20: 814c6000 81410a00 00000100 8013a7a0 81410a00 81410a00 81410a20 816ca400
[    2.765439] 5f40: 81410a00 816ca42c 81410a20 81103d40 00000088 815fa400 81410a00 8013ac9c
[    2.781998] 5f60: 815fa400 81255847 816db754 816db640 815fa400 8013aa60 816ca400 816db740
[    2.798550] 5f80: bb83dedc 00000000 00000000 8014366c 816db640 80143584 00000000 00000000
[    2.815105] 5fa0: 00000000 00000000 00000000 801000e4 00000000 00000000 00000000 00000000
[    2.831659] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    2.848211] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[    2.864766]  log_event_impl.part.0 from bcm2835_sdhost_set_ios+0x5c/0x108
[    2.875845]  bcm2835_sdhost_set_ios from mmc_power_up.part.0+0x48/0x158
[    2.886711]  mmc_power_up.part.0 from mmc_start_host+0x98/0xa4
[    2.896731]  mmc_start_host from mmc_add_host+0x80/0xb4
[    2.906064]  mmc_add_host from bcm2835_sdhost_add_host+0x308/0x4e8
[    2.916335]  bcm2835_sdhost_add_host from bcm2835_sdhost_probe+0x2f0/0x450
[    2.927290]  bcm2835_sdhost_probe from platform_probe+0x64/0xc4
[    2.937205]  platform_probe from really_probe+0xd0/0x2e0
[    2.946433]  really_probe from __driver_probe_device+0x90/0x1a4
[    2.956231]  __driver_probe_device from driver_probe_device+0x38/0x10c
[    2.966619]  driver_probe_device from __device_attach_driver+0x9c/0x110
[    2.977059]  __device_attach_driver from bus_for_each_drv+0x8c/0xe4
[    2.987097]  bus_for_each_drv from __device_attach+0xb4/0x1d8
[    2.996552]  __device_attach from bus_probe_device+0x90/0x94
[    3.005860]  bus_probe_device from deferred_probe_work_func+0x80/0xac
[    3.015941]  deferred_probe_work_func from process_one_work+0x22c/0x4ec
[    3.026169]  process_one_work from worker_thread+0x23c/0x560
[    3.035375]  worker_thread from kthread+0xe8/0x104
[    3.043631]  kthread from ret_from_fork+0x14/0x30
[    3.051714] Exception stack(0xbb8b5fb0 to 0xbb8b5ff8)
[    3.060087] 5fa0:                                     00000000 00000000 00000000 00000000
[    3.074758] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    3.089372] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    3.099284] Code: e597c000 e783c202 e0833202 e594200c (e592c004)
```

popcornmix, 2023-09-19:
Duplicate of https://github.com/raspberrypi/linux/issues/5573

pelwell, 2023-09-19:
I'm not so sure it's a duplicate - OP states that the dwc2 overlay doesn't fix it, and #5573 doesn't mention the bcm2835-sdhost driver.

6.5.3-v7+ is booting for me from USB (the SD card in a card reader) on a 3B. Can you post the full logs up to the point of failure? And the contents of your config.txt?

pelwell, 2023-09-19:
I can see a possible code path that could lead to such a fault, but with a valid 3B DTB it shouldn't occur.

mdiewa, 2023-09-20:
> I can see a possible code path that could lead to such a fault, but with a valid 3B DTB it shouldn't occur.

I double-checked the DTB and only then realized that I didn't install it correctly. The location in the source tree has changed in 6.5 from `arch/arm/boot/dts` to `arch/arm/boot/dts/broadcom` in line with arm64. I forgot about this and running `sudo cp arch/arm/boot/dts/*.dtb /mnt/boot/` didn't copy anything, which I failed to realize. After actually copying the DTBs to the boot disk the kernel boots fine. Thanks and sorry for the noise, I'll close the issue.

closed 2023-09-20 by mdiewa