raspberrypi/linux #5611: Wifi, hdmi audio, Bluetooth, audio broke on 6.1.45 kernel

Amitatreya, 2023-09-22:
### Describe the bug

I used prebuilt images on rpi 4b, all features are passing.
Checked out the kernel version 6.1.45, built and flashed the images as per rpi documentation, wifi, Bluetooth, audio and hdmi audio, fails to work, it looks drivers are not loading.
Kindly help and guide further.

### Steps to reproduce the behaviour

I used prebuilt images on rpi 4b, all features are passing.
Checked out the kernel version 6.1.45, built and flashed the images as per rpi documentation, wifi, Bluetooth, audio and hdmi audio, fails to work, it looks drivers are not loading.

### Device (s)

Raspberry Pi 4 Mod. B

### System

I used prebuilt images on rpi 4b, all features are passing.
Checked out the kernel version 6.1.45, built and flashed the images as per rpi documentation, wifi, Bluetooth, audio and hdmi audio, fails to work, it looks drivers are not loading.

pelwell, 2023-09-22:
Post the steps you followed from your `history`.

Amitatreya, 2023-09-29:
from: https://www.raspberrypi.com/documentation/computers/linux_kernel.html
build 32 bit:
KERNEL=kernel7l
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2711_defconfig
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs
flash:
mkdir mnt
mkdir mnt/fat32
mkdir mnt/ext4
sudo mount /dev/sdb1 mnt/fat32
sudo mount /dev/sdb2 mnt/ext4
sudo env PATH=$PATH make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=mnt/ext4 modules_install
sudo cp mnt/fat32/$KERNEL.img mnt/fat32/$KERNEL-backup.img
sudo cp arch/arm/boot/zImage mnt/fat32/$KERNEL.img
sudo cp arch/arm/boot/dts/*.dtb mnt/fat32/
sudo cp arch/arm/boot/dts/overlays/*.dtb* mnt/fat32/overlays/
sudo cp arch/arm/boot/dts/overlays/README mnt/fat32/overlays/
sudo umount mnt/fat32
sudo umount mnt/ext4
place the sdcard on rpi4b
boot the board and observe wifi with ifconfig, only eth0 and lo found, no wifi

popcornmix, 2023-09-29:
Can you report output of `uname -a`?

Note: that pi4 defaults to a 64-bit kernel (even on 32-bit image) on recent versions of RPiOS.

Amitatreya, 2023-09-30:
6.1.45

pelwell, 2023-09-30:
No, the whole line.

Amitatreya, 2023-09-30:
I shifted to 6.5 kernel, the issue is visible on 6.5 kernel as well.
msl@raspberrypi:~$ uname -ar
Linux raspberrypi 6.5.0-rc6-v7l+ #6 SMP Sat Sep 30 18:54:21 IST 2023 armv7l GNU/Linux