raspberrypi/linux #5615: Mipi-dbi-spi display on SPI interface is non-working in X/KMS

aBUGSworstnightmare-rpi, 2023-09-26:
### Describe the bug

I have a small 240x240pixels display with a controller which is MIPI-DBI compliant (ST7789VW). Wrote a custom tinyDRM compliant driver for the display which was only working when booted to CLI; in X I was unable to get the desktop shown on the display.
Find details on my custom driver + overlay: here https://forums.raspberrypi.com/viewtopic.php?t=355870

Then decided to switch to the in-tree driver and make use of 'panel-mipi-dbi.c' + it's related overlay instead to check if my custom driver has issues, or, if the result is the same.

I've created a panel.bin file in lib/firmware which has the required initialization commands for my display.

Tested my driver as well as the upstream driver of below kernels; non of them (!)  is starting the SPI display in X. The display works when booted to CLI, when booting to GUI it works as long as FW is active, it's switched off/fails when KMS takes over:
- 6.1.5x (log below are from the current kernel)
- 6.5.5
- 6.6.0-rc3

### Steps to reproduce the behaviour

Display which I'm currently testing with: https://www.waveshare.com/pico-lcd-1.3.htm

Any MIPI-DBI compliant display will do!

1.) Prepare a 'panel.bin' file according to the requirements explained in https://github.com/raspberrypi/linux/blob/cc08810f89e52337a99cc6ae5f53f08588357c5f/drivers/gpu/drm/tiny/panel-mipi-dbi.c
The default overlay (https://github.com/raspberrypi/linux/blob/rpi-6.1.y/arch/arm/boot/dts/overlays/mipi-dbi-spi-overlay.dts) expects the file with parameters/commands for initialization from the file 'panel.bin' in '/lib/firmware'

2.) Change config.txt to start the SPI display with the right set of GPIO configuration

3.) see the display working in CLI and failing in X

### Device (s)

Raspberry Pi CM4, Raspberry Pi CM4 Lite

### System

pi@WaveC-2GB-NVMe:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage4

pi@WaveC-2GB-NVMe:~ $ vcgencmd version
Apr 25 2023 18:26:03 
Copyright (c) 2012 Broadcom
version d7f9c2b4ef7e4a8c0b04374a879ce89d7a948453 (clean) (release) (start)

pi@WaveC-2GB-NVMe:~ $ uname -a
Linux WaveC-2GB-NVMe 6.5.5-v8+ #1 SMP PREEMPT Tue Sep 26 09:24:49 CEST 2023 aarch64 GNU/Linux

### Logs

System Information
------------------

Raspberry Pi Compute Module 4 Rev 1.1
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"

Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage4

Linux WaveC-2GB-NVMe 6.5.5-v8+ #1 SMP PREEMPT Tue Sep 26 09:24:49 CEST 2023 aarch64 GNU/Linux
Revision	: b03141
Serial		: 10000000ec0f48c5
Model		: Raspberry Pi Compute Module 4 Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0, libcamera interfaces=0

Videocore information
---------------------

Apr 25 2023 18:26:03 
[... 1197 lines cut ...]
# ant2 = external
# ---------------------------------------------
dtparam=ant2

# ---------------------------------------------
# enable USB in host mode on CM4
# ---------------------------------------------
dtoverlay=dwc2,dr_mode=host

# ---------------------------------------------
# uncomment to enable overclocking for RPi4/CM4/Pi400
# ---------------------------------------------
arm_freq=2200

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
dtoverlay=mipi-dbi-spi,spi0-0,speed=48000000
dtparam=write-only,cpha,cpol
dtparam=width=240,height=240,width-mm=23,height-mm=23
dtparam=reset-gpio=27,dc-gpio=25,backlight-gpio=24

aBUGSworstnightmare-rpi, 2023-09-26:
Here you can find a video of the display when powered up
https://drive.google.com/file/d/1bTyH07OEvgjPvhh29egJpZJOiQuwhytR/view?usp=share_link

6by9, 2023-09-26:
> Wrote a custom tinyDRM compliant driver for the display which was only working when booted to CLI; in X I was unable to get the desktop shown on the display.

If it's working at the console, then it's not a kernel issue.

X doesn't perform brilliantly with multiple DRM cards - it has concepts of source and sink providers, as reported by ```xrandr --listproviders```. As I'd linked in the forum thread, I've been down that road before - https://forums.raspberrypi.com/viewtopic.php?p=1837374#p1837374

I'd suggest you wait for Bookworm as it uses Wayfire and Wayland which does work nicely with mutliple DRM cards.

aBUGSworstnightmare-rpi, 2023-09-26:
Neven been able to get the display started in X when following your suggestions; KMS will always end with a log-on box and a black SPI display.

There is no other display connected (No HDMI/DSI/DPI) during my testing, so how to make sure there is only one DRM card - the SPI - display?

Got It working once by accident when the GPU binding fatally failed, but was not able to reproduce/take logs (as I wasn't prepared for it to work).

From my Bookworm testing I figured out that Wayfire and Wayland have other issues. Anyhow, if you can share a link (via email) to latest Bookworm Beta (resigned from beta testing program because of various resonsl) I'm willing to test it with 6.5 kernel again.

6by9, 2023-09-26:
A comment from a colleague was that they needed a file /etc/X11/xorg.conf.d/99-vc4.conf:
```
Section "OutputClass"
    Identifier "vc4"
    MatchDriver "vc4"
    Driver "modesetting"
    Option "PrimaryGPU" "true"
EndSection
```
to get vc4 selected correctly as the primary X display.

You don't only have one card - v3d should exist and be doing the composition, with your SPI driver doing the display.

aBUGSworstnightmare-rpi, 2023-09-26:
Created a file with the content shown.
Also checked/setup providers by
`pi@WaveC-2GB-NVMe:~ $ DISPLAY=:0 xrandr --listproviders
Providers: number : 2
Provider 0: id: 0x47 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 6 outputs: 2 associated providers: 0 name:modesetting
Provider 1: id: 0x79 cap: 0xf, Source Output, Sink Output, Source Offload, Sink Offload crtcs: 1 outputs: 1 associated providers: 0 name:modesetting
pi@WaveC-2GB-NVMe:~ $ xrandr  --setprovideroutputsource 1 0
`
Used screen configuration editor for switching the SPI on - not working -> back to log-on screen (which is funny as well as my OS is setup for auto-login).

`pi@WaveC-2GB-NVMe:~ $ sudo cat /sys/kernel/debug/device_component/gpu
aggregate_device name                                  status
-------------------------------------------------------------
gpu                                                     bound

device name                                            status
-------------------------------------------------------------
fe400000.hvs                                            bound
fef00700.hdmi                                           bound
fef05700.hdmi                                           bound
fe004000.txp                                            bound
fe206000.pixelvalve                                     bound
fe207000.pixelvalve                                     bound
fe20a000.pixelvalve                                     bound
fe216000.pixelvalve                                     bound
fec12000.pixelvalve                                     bound
pi@WaveC-2GB-NVMe:~ $ sudo cat /sys/kernel/debug/devices_deferred
`

`[    7.594520] panel-mipi-dbi-spi spi0.0: supply power not found, using dummy regulator
[    7.667138] panel-mipi-dbi-spi spi0.0: supply io not found, using dummy regulator
[    7.689118] [drm] Initialized panel-mipi-dbi 1.0.0 20220103 for spi0.0 on minor 2
[    7.696437] Bluetooth: hci0: BCM: features 0x2f
[    7.702338] Bluetooth: hci0: BCM43455 37.4MHz Raspberry Pi 3+-0190
[    7.702376] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0382
[    7.936150] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    8.270926] Console: switching to colour frame buffer device 30x30
[    8.306546] panel-mipi-dbi-spi spi0.0: [drm] fb0: panel-mipi-dbid frame buffer device`

aBUGSworstnightmare-rpi, 2023-09-26:
@6by9: your post on the vc4.conf gave me the idea to do something else

`sudo nano /etc/X11/xorg.conf.d/99-fbdev.conf`
adding below to the file

> Section "Device"
  Identifier "FB Driver"
  Driver "fbdev"
  Option "fbdev" "/dev/fb0"
EndSection

saving and reboot kicks the display alive, but is for sure not what we want.

> pi@WaveC-2GB-NVMe:~ $ xrandr --verbose
> xrandr: Failed to get size of gamma for output default
> Screen 0: minimum 240 x 240, current 240 x 240, maximum 240 x 240
> default connected 240x240+0+0 (0x51e) normal (normal) 0mm x 0mm
> 	Identifier: 0x51d
> 	Timestamp:  9894
> 	Subpixel:   unknown
> 	Clones:    
> 	CRTC:       0
> 	CRTCs:      0
> 	Transform:  1.000000 0.000000 0.000000
> 	            0.000000 1.000000 0.000000
> 	            0.000000 0.000000 1.000000
> 	           filter: 
> 	non-desktop: 0 
> 		supported: 0, 1
>   240x240 (0x51e)  0.000MHz *current
>         h: width   240 start    0 end    0 total  240 skew    0 clock   0.00KHz
>         v: height  240 start    0 end    0 total  240           clock   0.00Hz

[image]
Looks like I need to figure out which conf file in KMS needs to be changed/added.

aBUGSworstnightmare-rpi, 2023-10-13:
Just confirmed that mipi-dbi-spi is also not starting the display on Bookworm 6.5.7 kernel under Wayland.
working during boot/when. booted to CLI, but non-working under Wayland

[](https://drive.google.com/file/d/1H0mK421wtEtm6LPwuaVfl5ewy0U3f7KS/view)

[](https://drive.google.com/file/d/1KhCgEMtNpLK2PM0R7abZGCtfajCSSgte/view)

sdhizumi, 2024-05-30:
I have same issue.
I try to use ILI9341 display on Pi OS bookworm with Raspberry Pi Zero 2 W.

I installed Pi OS Lite bookworm to my microSD.
I build my panel.bin using [this sample](https://github.com/notro/panel-mipi-dbi/wiki/Displays#adafruit-ili9341-pitft-tft-lcd-screen) and placed on /lib/firmware

I wired my display and set this on config.txt
```
dtparam=spi=on

dtoverlay=mipi-dbi-spi,spi0-0,speed=70000000
dtparam=width=320,height=240
dtparam=reset-gpio=25,dc-gpio=24
dtparam=backlight-gpio=23
```
After reboot, CLI messages are displaying on my display.
I found these logs from dmesg
```
[    1.743999] simple-framebuffer 1ebaf000.framebuffer: fb0: simplefb registered!
[   12.102686] panel-mipi-dbi-spi spi0.0: [drm] fb0: panel-mipi-dbid frame buffer device
```
It seems my display are recognized fb0 (=first?) flamebuffer device. It looks nothing has problems.

I installed
```
sudo apt install mate-desktop-environment lightdm lightdm-gtk-greeter
```
I change login option by raspi-config, from console to graphical, reboot, there is nothing display after console messages.

I tried same steps on Pi OS Lite bullseye (kernel 6.1), It is working. I can see MATE desktop.

6by9, 2024-06-01:
> Edit: My friend tested with Pi 5, Pi 4, and Pi 3B+. Pi 5 and Pi 4 are working but Pi 3B+ is not.

Pi 4 & 5 on Bookworm default to using Wayfire instead of X.
Wayfire supports multiple display devices far better than X, but has a significant overhead in use of 3D hardware for rendering.
We are beta testing labwc which has much lower 3D requirements on the lower powered Pis. See https://forums.raspberrypi.com/viewtopic.php?p=2223184#p2223184 for basic install instructions.

As I said in https://github.com/raspberrypi/linux/issues/5615#issuecomment-1735501138 if it works in the console, then it's not a kernel issue.

sdhizumi, 2024-06-02:
> Pi 4 & 5 on Bookworm default to using Wayfire instead of X.
Thank you! This was really good hint for me!

I change from X to Wayland in raspi-config on Pi Zero 2 W, and then I run `sway` from CLI, it works with mipi-dbi-spi display! (weston and cage also works)
I can't run PIXEL desktop (something problems about OpenGL support on Pi 3/Zero2's SoC?). But it is enough for me at this time.