raspberrypi/linux #5616: BT SCO over PCM back-channel (RPI4, CM4)

mikegapinski, 2023-09-27:
### Describe the bug

Hi,

I am looking for a possibility to use BT SCO over PCM on RPI4/CM4, in theory it should be possible but might require some config/firmware/kernel changes to work. 

Background: BT SCO over HCI is used on Linux based systems but that mode is not supported in Android based distros - no support in the BT stack.

The chip used for BT/Wi-Fi on the Pi is used on Android on other boards, so the vendor should be able to provide some sort of integration guidance, the precompiled proprietary HAL etc. The generic btlinux hal that is working with krnbt supports playback without issues however the microphone is not captured anywhere - you always get an empty buffer with the recording even though the SCO channel is established (a dummy phone call is visible on the connected device).

### Steps to reproduce the behaviour

- Download any Android image for the Pi, Glodroid, Lineage, Emteria etc
- Pair a bluetooth headset
- Use a sound recorder app or jump on a VOIP call
- You'll hear the audio but nothing would be recorded on the device

### Device (s)

Raspberry Pi 4 Mod. B, Raspberry Pi CM4

### System

5.15, 6.x

### Additional context

I think that if it is possible, sound/soc/codecs/bt-sco.c can be used, but it has to be configured in the device tree in the sound section

https://kernel.googlesource.com/pub/scm/linux/kernel/git/davem/sparc/+/refs/heads/master/Documentation/devicetree/bindings/sound/bt-sco.txt

pelwell, 2023-09-27:
Where are you expecting this PCM data to appear? The Bluetooth PCM pins are not connected on CM4, and have not been connected on any Pi since the Pi 3B rev 1.2.

pelwell, 2023-09-27:
I don't think it would need to be proprietary, just hooked up to a standard I2S interface. But it isn't on Raspberry Pis.

mikegapinski, 2023-09-27:
Thanks for the explanation, it might be worth considering connecting the PCM pins on newer boards when designing them. Maybe there was no reason to do it before since bluez does not need it

closed 2023-09-27 by mikegapinski

pelwell, 2023-09-27:
They were dropped because we had no use for them and because they occupy precious board space and GPIOs; the routing around the WiFi/BT combo chip is tight enough as it is, and we might even have needed a second/larger GPIO expander if we had to sacrifice for more SoC GPIOs.