raspberrypi/linux #5619: kernel 6.1.53-v8+ broke official 7'' touchscreen

rafael2k, 2023-09-28:
### Describe the bug

The touchscreen was working flawless for years... until kernel 6.1.53 broke it.

 [    8.914075] edt_ft5x06 10-0038: supply vcc not found, using dummy regulator
[    8.914404] edt_ft5x06 10-0038: supply iovcc not found, using dummy regulator
[    8.915006] edt_ft5x06 10-0038: touchscreen probe failed
[    8.915382] edt_ft5x06: probe of 10-0038 failed with error -5

More reports at:
https://forums.raspberrypi.com/viewtopic.php?t=356352

### Steps to reproduce the behaviour

Just use a kernel  6.1.53 or newer ( 6.1.54 has the same problem).
Using Raspberry OS Bullseye 64 bit here. 

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage2

Sep 21 2023 18:42:00 
Copyright (c) 2012 Broadcom
version 3094eda5aadf2c719edd2c68d1900884abc3dbca (clean) (release) (start)

Linux blablabla 6.1.54-v8+ #1682 SMP PREEMPT Wed Sep 20 15:19:57 BST 2023 aarch64 GNU/Linux

### Logs

 [    8.914075] edt_ft5x06 10-0038: supply vcc not found, using dummy regulator
[    8.914404] edt_ft5x06 10-0038: supply iovcc not found, using dummy regulator
[    8.915006] edt_ft5x06 10-0038: touchscreen probe failed
[    8.915382] edt_ft5x06: probe of 10-0038 failed with error -5

### Additional context

-

pelwell, 2023-09-29:
Do you have a Last Known Good kernel version?

rafael2k, 2023-09-29:
I was using something like 6.1.3x, and it was all good. I can try to pinpoint at least the minor version which broke. Do you happen to know how to use rpi-update to make it install older kernel versions?
Of course I can just compile the kernel... but then I'll take much more time.

pelwell, 2023-09-29:
I see that 6.1.21 is working - I'll bisect from there.

rafael2k, 2023-09-29:
6.1.21 is definitely working, indeed, as available via the apt repository (from Raspberry OS) or from the official Raspberry OS images.

rafael2k, 2023-09-29:
for the reference, my config.txt:
```
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=audio=off
display_auto_detect=1
dtoverlay=vc4-kms-v3d,noaudio
max_framebuffers=2
arm_64bit=1
disable_overscan=1
disable_splash=1
otg_mode=1

[pi4]
arm_boost=1
# this is for my RTC and bit-banged i2c I use
dtoverlay=i2c-rtc-gpio,ds1307,i2c_gpio_delay_us=10,bus=2,i2c_gpio_sda=13,i2c_gpio_scl=6
# my soundcard
dtoverlay=audioinjector-wm8731-audio
dtoverlay=i2s-mmap
avoid_warnings=1
disable_poe_fan=1
force_eeprom_read=0
dtoverlay=disable-bt
```

pelwell, 2023-09-29:
This is caused by vc4-kms-dsi-7inch overlay, which got damaged between 6.1.47 and 6.1.48, when the support for i2c_csi_dsi0 was added, by the addition of another level of fragment merging.

Fragments should be thought of as being merged in order, where the order is the order in which they appear in the file, not by the numbering of the fragments. However, the firmware and other overlay tools have an additional rule saying that fragments that patch things within the overlay ("intra" fragments) should be merged before fragments which patch the base DTB ("extra" fragments). What it can't do is detect and correctly order a chain of patches.

In this case, and fragment 5 patched fragment 12, fragment 12 patched fragment 13, and fragment 13 patched the base DTB - which sounds OK written like that, but not when fragment 13 appears after fragment 13 in the file. At the point the overlay is applied, the intra fragments get merged first, with 12 patching 13 and 5 patching 12 _in that order_, so the contributions from fragment 5 - `vcc-supply` and `reset-gpio` - get lost, disabling the touchscreen by keeping it reset.

Combining fragments 12 and 13 removes the extra level of dependency, and the disabling of fragment 12 can be replaced by setting its `status` property to `disabled`.

#5622 is my proposed solution. If you wait about an hour until the build checks complete you'll be able to run `sudo rpi-update pulls/5622` to confirm that it fixes the problem.

pelwell, 2023-09-29:
[ Previous description updated to correct a few errors ]

rafael2k, 2023-09-29:
I will test ASAP. Thanks a lot!

rafael2k, 2023-09-29:
Tested and confirmed it works! Yay!

pelwell, 2023-09-29:
Thanks - that patch will be in all future kernel builds.

xXhoobaXx, 2023-09-29:
Hey , thank you for following the issue, been waiting for over a week for a solution, can you help me how to apply the fix please

pelwell, 2023-09-29:
Apply it to what? If you are building your own kernel then just clone this repo and you'll get the fix. If you just want to install a kernel with the fix run `sudo rpi-update pulls/5622`. If you don't have rpi-update, first run `git clone https://github.com/raspberrypi/rpi-update.git` and get it from there.

xXhoobaXx, 2023-09-29:
Thank you @pelwell , touch screen and camera v3 are working perfectly now , and thank you @rafael2k for opening the case here to solve the problem

rafael2k, 2023-09-30:
Yay! @pelwell, can we close the issue?

closed 2023-09-30 by pelwell

hydazz, 2023-10-26:
still experiencing this on `6.1.58-v8`, also tried `rpi-update pulls/5622` (back to `6.1.54-v8+`), same error:

```
[   11.133848] edt_ft5x06 10-0038: touchscreen probe failed
[   11.134136] edt_ft5x06: probe of 10-0038 failed with error -5
```
RPI4B, Bookworm (PIOS Lite) 64Bit

pelwell, 2023-10-26:
After running `rpi-update pulls/5622` again (and rebooting, of course), what do the following commands report?:
```
$ dmesg | grep edt_ft5x06`
$ ls -l /proc/device-tree/soc/i2c0mux/i2c@1/ts@38
```

hydazz, 2023-10-26:
> After running `rpi-update pulls/5622` again (and rebooting, of course), what do the following commands report?:
> 
> ```
> $ dmesg | grep edt_ft5x06`
> $ ls -l /proc/device-tree/soc/i2c0mux/i2c@1/ts@38
> ```

```
alex@livingroom-dashboard:~ $ uname -r
6.1.54-v8+
alex@livingroom-dashboard:~ $ dmesg | grep edt_ft5x06
[   10.054085] edt_ft5x06 10-0038: supply iovcc not found, using dummy regulator
[   10.397893] edt_ft5x06 10-0038: touchscreen probe failed
[   10.398163] edt_ft5x06: probe of 10-0038 failed with error -5
alex@livingroom-dashboard:~ $ ls -l /proc/device-tree/soc/i2c0mux/i2c@1/ts@38
total 0
-r--r--r-- 1 root root 15 Oct 26 20:18 compatible
-r--r--r-- 1 root root  3 Oct 26 20:18 name
-r--r--r-- 1 root root  4 Oct 26 20:18 phandle
-r--r--r-- 1 root root  4 Oct 26 20:18 reg
-r--r--r-- 1 root root 12 Oct 26 20:18 reset-gpio
-r--r--r-- 1 root root  0 Oct 26 20:18 touchscreen-inverted-x
-r--r--r-- 1 root root  0 Oct 26 20:18 touchscreen-inverted-y
-r--r--r-- 1 root root  4 Oct 26 20:18 touchscreen-size-x
-r--r--r-- 1 root root  4 Oct 26 20:18 touchscreen-size-y
-r--r--r-- 1 root root  4 Oct 26 20:18 vcc-supply
alex@livingroom-dashboard:~ $
```

pelwell, 2023-10-26:
Thanks - to me that shows that the overlay is doing what is should (both vcc-supply and reset-gpio are present). The fact that you are getting errors on the latest(ish) kernel and 6.1.21 suggest, while for other people it is working, suggests a hardware/wiring fault in your setup.

hydazz, 2023-10-26:
> failed with error -5

does `failed with error -5` point to anything?

pelwell, 2023-10-26:
> does failed with error -5 point to anything?

    #define	EIO		 5	/* I/O error */

Yes - an I2C access is failing, i.e. a hardware problem. Error -121 is similar:

    #define	EREMOTEIO	121	/* Remote I/O error */

PinkFreud, 2023-10-26:
> Thanks - to me that shows that the overlay is doing what is should (both vcc-supply and reset-gpio are present). The fact that you are getting errors on the latest(ish) kernel and 6.1.21 suggest, while for other people it is working, suggests a hardware/wiring fault in your setup.

I'm chiming in with another unresponsive touchscreen.  Same errors for me:
```
[   10.540015] edt_ft5x06 10-0038: supply vcc not found, using dummy regulator
[   10.540342] edt_ft5x06 10-0038: supply iovcc not found, using dummy regulator
[   10.541029] edt_ft5x06 10-0038: touchscreen probe failed
[   10.541435] edt_ft5x06: probe of 10-0038 failed with error -5
```

This Pi was running the bullseye image up until yesterday, and the touchscreen worked without issue.  Once I booted bookworm, the touchscreen was unresponsive, and all attempts to revive it, including the aforementioned `rpi-update pulls/5622` fail for me - rpi-update succeeds, but a reboot doesn't bring the touchscreen back to life.

pelwell, 2023-11-02:
I just tried this with a 32-bit Bookworm image running the latest stable kernel on a 4B:
```
pi@raspberrypi:~$ uname -a
Linux raspberrypi 6.1.58-v7l+ #1694 SMP Mon Oct 30 11:51:22 GMT 2023 armv7l GNU/Linux
```
Attaching only the display ribbon, 5V and GND, and making no configuration changes, it just works - I can click, double-click and drag. Relevant messages in the kernel log are:
```
[   11.679899] edt_ft5x06 10-0038: supply iovcc not found, using dummy regulator
[   12.004473] input: generic ft5x06 (00) as /devices/platform/soc/fe205000.i2c/i2c-22/i2c-10/10-0038/input/input2
```
`config.txt` includes `dtoverlay=vc4-kms-v3d` and `disable_fw_kms_setup=1` (removing the latter doesn't stop it working).

Switching the same image and display over to a 3B+ I get the same result (just slower), although of course it's running the pre-Wayland lxpanel desktop:
```
[   15.761012] edt_ft5x06 10-0038: supply iovcc not found, using dummy regulator
[   16.365789] input: generic ft5x06 (00) as /devices/platform/soc/3f205000.i2c/i2c-11/i2c-10/10-0038/input/input1
```

pelwell, 2023-11-02:
(To install the latest stable kernel, run `sudo rpi-update stable`, having first backed up any important data)

MartinHajda, 2023-12-01:
Thanks to @rafael2k for the thread and @pelwell for fixing the problem, the display now works as well as before. Good work!

barajaam, 2023-12-30:
Touch Screen on 6.1.69-v8+ #1711 not working any help is appreciated.