raspberrypi/linux #5621: WEB UPLOAD SYSTEM WITH RASPBERRY PI 3B+ do not want to read / write to usb ssd drive and or flash / sd card !!!!

karadevnet, 2023-09-29:
### Describe the bug

Project : WEB UPLOAD SYSTEM WITH RASPBERRY PI 3B+
AND 2023-05-03-raspios-bullseye-armhf-full.img
Hardware : raspberry pi 3b+ and ssd apacer 1TB usb to sata outside box
Boot sd card Kingston 32GB, pi imager to upload img file downloaded by torent
GOAL : upload / download / delete files to SSD drive trough simple web site in local lan system with link IP : 192.168.88.249/upload
Router : mikrotik hap lite – best of 😊 no problem with this
Pi static ip : 192.168.99.249
Software WEB SERVER : apache 2 + php7.4.33 instaled by bash command
user@user : sudo apt install apache2
user@user : sudo apt install php libapache2-mod-php
all work fine until start to make my web system with external ssd drive or any of drives attached to usb ports.
Link for php file git hub https://github.com/jcampbell1/simple-file-manager
System is good, but linux can not see sym links outside from standart apache folder
/var/www/html/upload
Php link works fine will show in picture bellow. I know that apache user is www-data and need to own sym link and target mount folder and/or all content inside in mount folder.
But when execute:
sudo chown -R www-data:www-data /media/pi/ADATA_8_GB
command is show to executed, no return errors and/or warning messages but user and group are same. Can be seen with
ls -l
like Is here:
[image]

Icon of link wit name ADATA_8_GB is not like a other folder in list and is not active. When click on it give forbidden 403 error
[image]

I spend one week to read all data in internet for this problem. Try it to make is like a new virtual host with apache conf file all is written in internet I try it. NO SUCCESS !!!!

I think is here have some little fix need to be done to make it all system works like charm. I can not understand what the problem is, but all y read and try from internet not works.

Linux or apache2 or php x.x.x.x or some else can not allow to work with extended drives attached to usb ports.
Try it with ntfs, ext4, fat32 no success and no success and that is it !!! ☹

After one week fight with all powers that I have I decide to write you for this and think this is best way
I know like all world that the pi 5 is here. It is crucial for automation for all users to make it work fine.
Because all my projects with lan 10/100 communications depends of this to can read write execute data from extended usb devices like ssd sata attached to usb.

I am sure and will be happy like all users to here that is worked in your desk office first and way to make it work in my.
Best regards to all developers. Pi 5 is great and strong will be mile stone to all my projects with automation, electronics and programming

My web site is [www.karadev.net](http://www.karadev.net/) – there have many info about me. Also have a github with C_SHARP_RPI2040_LITTLE_FS_UPLOADER for raspberry pi pico file system. I made it 5-6 months ago.
https://github.com/karadevnet 

I work a lot with PICO it is great.
NEED THIS TO RUN for all others automation system that I produce and develop 😊
If any info need or need to test some way to give a data for fixing this let me know
I ma ready to help to fix, test, develop and make it run it

### Steps to reproduce the behaviour

install brand new distro from 2023-05-03-raspios-bullseye-armhf-full.img torent download
Software WEB SERVER : apache 2 + php7.4.33 instaled by bash command
user@user : sudo apt install apache2
user@user : sudo apt install php libapache2-mod-php

make a folder upload and save in it index.php file of web upload system
make a sym link to adata flash drive and / or ssd / sd card to access drives
use chown to change sym link user from root to www-data:www-data

and other 1000 try command and files for virtual host, but no success :(
need to make some sort of easy menager to make a extended drives work like a local folders that sym links need to do but some easy way to set up this.
and some sort of software to make a pi like a small hosting company for 1000 web sites if user need this
with pi 5 that will be crucial for user and automation

### Device (s)

Raspberry Pi 3 Mod. B+

### System
[... 437 lines cut ...]
[   18.707562] Bluetooth: HCI device and connection manager initialized
[   18.707584] Bluetooth: HCI socket layer initialized
[   18.707594] Bluetooth: L2CAP socket layer initialized
[   18.707614] Bluetooth: SCO socket layer initialized
[   18.742434] Bluetooth: HCI UART driver ver 2.3
[   18.742460] Bluetooth: HCI UART protocol H4 registered
[   18.742554] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   18.742713] Bluetooth: HCI UART protocol Broadcom registered
[   19.569171] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   19.569195] Bluetooth: BNEP filters: protocol multicast
[   19.569212] Bluetooth: BNEP socket layer initialized
[   19.575503] Bluetooth: MGMT ver 1.22
[   19.603243] NET: Registered PF_ALG protocol family
[   19.660560] cryptd: max_cpu_qlen set to 1000
[   20.067027] Bluetooth: RFCOMM TTY layer initialized
[   20.067064] Bluetooth: RFCOMM socket layer initialized
[   20.067087] Bluetooth: RFCOMM ver 1.11
[   29.025557] EXT4-fs (sda1): recovery complete
[   29.448418] EXT4-fs (sda1): mounted filesystem with ordered data mode. Quota mode: none.
pi@pi:~ $ 

### Additional context

please help me in good way. not just to give a command and say try this try that :(
if is need i will give to some one remote control directly to my pi 3b+

pelwell, 2023-09-29:
This is not the right place to ask for help unless there is a clear bug in the kernel. The Raspberry Pi forums are much more likely to be helpful - try Troubleshooting as a starting point: https://forums.raspberrypi.com/viewforum.php?f=28

However, you say:

> I know that apache user is www-data and need to own **sym link and target mount** folder and/or all content inside in mount folder.
> But when execute:
> sudo chown -R www-data:www-data [...]
ls -l

Where are you changing the ownership of the symbolic link? /var/www/html/upload/ADATA_8_GB is the symbolic link and 
/media/pi/ADATA_8GB is the target directory, but you are only chowning the target and "ls -l"ing the symbolic link.

closed 2023-09-29 by pelwell

karadevnet, 2023-09-29:
as you can see from this picture and all data and mount folder are same user of apache www-data.
the bug is there, but i can not get it. i am not a programmer of kernel, apache or else. need to check it deeply to find it and fix it. i know about forums very well, but have not reason to write there, because i already know and make what they will say to me. check it, fixit and make it work !!! all linux and pi 4, 5, ++ waiting for this!!!
[image]

pelwell, 2023-09-29:
> > Where are you changing the ownership of the symbolic link?

```
$ sudo chown -R www-data:www-data /var/www/html/upload/ADATA_8_GB
$ ls -l /var/www/html/upload/ADATA_8_GB
$ ls -l /var/www/html/upload/ADATA_8_GB/
```

karadevnet, 2023-09-29:
no no . talk about 
sudo chown -R www-data:www-data /media/pi/SDCARD_EXT4_16GB
but what i plug in to list in web page and set ownership to www-data:www-data is not work