raspberrypi/linux #5623: USB current limitations start with kernel 6.5.5

df8oe, 2023-09-30:
### Describe the bug

Starting with kernel 6.5.5 most of my USB devices can no longer be used because of current lilimtations. Setup was working since years without any problems, all 4 ports have devices connected, all devices do have much less power consumption (level shifters, adaptors to RS232 and RS485).

### Steps to reproduce the behaviour

connect USB devices to all ports

### Device (s)

Raspberry Pi 3 Mod. B

### System

OS:
Arch Linux

Firmware:
Sep 21 2023 18:44:30 
Copyright (c) 2012 Broadcom
version 3094eda5aadf2c719edd2c68d1900884abc3dbca (clean) (release) (start)

Kernel:
Linux raspserver 6.5.5-2-rpi-ARCH #1 SMP PREEMPT Thu Sep 28 12:43:30 EDT 2023 aarch64 GNU/Linux
(the issue started with kernel 6.5.5-1-rpi-ARCH #1 SMP PREEMPT)

### Logs

If I only connect a 4 port hub (without anything applied:
usb 1-1.2: can't connect bus-powered hub to this port Hub

If I connect an USB to RS485 adaptor:
usb 1-1.2: rejected 1 configuration due to insufficient available bus power

### Additional context

If i use

max_usb_current=1

in config.txt everything is working as before. This line was not neccessary up to kernel 6.5.4

1caue, 2023-09-30:
[image]

henkv1, 2023-10-14:
I have the same issue, but setting max_usb_current=1 did not fix it for me. I needed to execute echo 1 >  /sys/bus/usb/devices/1-1.3/bConfigurationValue to make the USB device usable.

my system is:
Raspberry Pi 3 Model B Rev 1.2
Kernel 6.5.7-1-rpi-ARCH
raspberrypi-firmware 20230921-1

Kernel messages:
```
[    3.344605] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    3.344946] Indeed it is in host mode hprt0 = 00001101
[    3.553444] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
[    3.553499] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.554938] hub 1-1:1.0: USB hub found
[    3.555185] hub 1-1:1.0: 5 ports detected
[    3.555754] hub 1-1:1.0: insufficient power available to use all downstream ports
[    3.844612] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
[    3.945203] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
[    3.945251] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    3.948754] smsc95xx v2.0.0
[    4.075323] SMSC LAN8700 usb-001:003:01: attached PHY driver (mii_bus:phy_addr=usb-001:003:01, irq=184)
[    4.076833] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:db:85:9a
[    4.160619] usb 1-1.3: new high-speed USB device number 4 using dwc_otg
[    4.276333] usb 1-1.3: New USB device found, idVendor=148f, idProduct=7601, bcdDevice= 0.00
[    4.276385] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.276408] usb 1-1.3: Product: 802.11 n WLAN
[    4.276428] usb 1-1.3: Manufacturer: MediaTek
[    4.276447] usb 1-1.3: SerialNumber: 1.0
[    4.277239] usb 1-1.3: rejected 1 configuration due to insufficient available bus power
[    4.277274] usb 1-1.3: no configuration chosen from 1 choice
[    4.277298] usb 1-1.3: No support over 100mA
```

df8oe, 2023-10-14:
Do you place it in the [all] - section? I have multiple rpis and this worked for all of them...

henkv1, 2023-10-14:
Putting it in the [all] section did not make any difference:
```
[root@alarm ~]# cat /boot/config.txt 
# See /boot/overlays/README for all available options

initramfs initramfs-linux.img followkernel

display_auto_detect=1

dtoverlay=vc4-kms-v3d,cma-256
dtoverlay=rpivid-v4l2
disable_overscan=1
arm_64bit=1

[all]
max_usb_current=1
```

df8oe, 2023-10-14:
Strange. I cannot reproduce.

yi-yunseok, 2023-10-15:
+1 same issue.

p-v-a, 2023-12-06:
Facing the same issue, but in my case situation is even worse. I was using USB to SATA bridge to boot Pi from SSD:
```
[  165.271878] usb 1-1.2: USB disconnect, device number 6
[  176.757342] usb 1-1.2: new high-speed USB device number 7 using dwc_otg
[  176.939015] usb 1-1.2: New USB device found, idVendor=152d, idProduct=0576, bcdDevice=15.07
[  176.939055] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  176.939078] usb 1-1.2: Product: USB to ATA/ATAPI Bridge
[  176.939099] usb 1-1.2: Manufacturer: JMicron
[  176.939119] usb 1-1.2: SerialNumber: 0123456789ABCDEF
```

and after update to rpi-6.6.y, raspberry couldn't longer boot. ending up in initramfs
```
Begin: Waiting for root file system ... Begin: Running /scripts/local-block ... done.
done.
Gave up waiting for root file system device.  Common problems:
 - Boot args (cat /proc/cmdline)
   - Check rootdelay= (did the system wait long enough?)
 - Missing modules (cat /proc/modules; ls /dev)
ALERT!  UUID=91ce9fc6-02 does not exist.  Dropping to a shell!
```

Though there were no USB power related messages visible, I might just missed them. I have to use spare USB->SATA bridge to make system bootable again (still powered via USB port).

After plugging in old USB->SATA adaptor, I've seen the following in dmesg:
```
[  176.757342] usb 1-1.2: new high-speed USB device number 7 using dwc_otg
[  176.939015] usb 1-1.2: New USB device found, idVendor=152d, idProduct=0576, bcdDevice=15.07
[  176.939055] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  176.939078] usb 1-1.2: Product: USB to ATA/ATAPI Bridge
[  176.939099] usb 1-1.2: Manufacturer: JMicron
[  176.939119] usb 1-1.2: SerialNumber: 0123456789ABCDEF
[  176.939849] usb 1-1.2: rejected 1 configuration due to insufficient available bus power
[  176.939879] usb 1-1.2: no configuration chosen from 1 choice
[  176.939901] usb 1-1.2: No support over 100mA
```

and after
```
root@only-slightly-bent:~# echo 1 > /sys/bus/usb/devices/1-1.2/bConfigurationValue
```

I finally was able to recognise adaptor, just an adaptor, as there were no disk connected to adaptor at that time.

```
[  341.363708] usb 1-1.2: new config #1 exceeds power limit by 400mA
[  341.364413] usb 1-1.2: The driver for the USB controller dwc_otg_hcd does not support scatter-gather which is
[  341.364446] usb 1-1.2: required by the UAS driver. Please try an other USB controller if you wish to use UAS.
[  341.364614] usb 1-1.2: The driver for the USB controller dwc_otg_hcd does not support scatter-gather which is
[  341.364638] usb 1-1.2: required by the UAS driver. Please try an other USB controller if you wish to use UAS.
[  341.364660] usb-storage 1-1.2:1.0: USB Mass Storage device detected
[  341.366134] scsi host1: usb-storage 1-1.2:1.0
[  342.398058] scsi 1:0:0:0: Direct-Access     JMicron  Generic          1507 PQ: 0 ANSI: 6
[  342.398963] sd 1:0:0:0: Attached scsi generic sg1 type 0
```

Same adapted works fine with the same disk when connected to the same Pi running rpi-6.1.y, as well as when I connect it into my PC.

p-v-a, 2023-12-06:
> Diff of `sudo lsusb -v` for the first hub on a 3b+:

> 
> Device status looks an awful lot like garbage...

What i noticed as well, is all devices I have problem with are USB3.0
SATA adaptor is USB3, while troubleshooting I have tried connecting adaptor via USB3.0 hub, which wasn't even get detected.
SATA adaptor that worked is 12 years old USB2 device.

Just an observation, not sure if it's related.

popcornmix, 2023-12-08:
Can you test to see if See: https://github.com/raspberrypi/linux/pull/5775 helps?
It is now merged on 6.5, 6.6 and 6.7 kernels.

p-v-a, 2023-12-09:
Seems to work now! I can now boot from SSD, via SATA-to-USB adapter.