raspberrypi/linux #5624: system hang on upgrading to kernel 6.1.54-2-aarch64

GazzaC, 2023-09-30:
### Describe the bug

when using dual head displays with 
dtoverlay=vc4-fkms-v3d-pi4
dtoverlay=rpivid-v4l2
in config.txt 
system will lock with solid green light
works fine on kernel 6.1.54-1-aarch64 and below

### Steps to reproduce the behaviour

upgrade system to kernel version linux-rpi-6.1.54-2-aarch64

### Device (s)

Raspberry Pi 4 Mod. B

### System

OS - Arch Linux - 6.1.54-2-rpi-ARCH #1 SMP PREEMPT Tue Sep 19 11:55:52 MDT 2023 aarch64 GNU/Linux

# /opt/vc/bin/vcgencmd version
Sep 21 2023 18:42:00 
Copyright (c) 2012 Broadcom
version 3094eda5aadf2c719edd2c68d1900884abc3dbca (clean) (release) (start)

### Additional context

[archlinuxarm.org](https://archlinuxarm.org/forum/viewtopic.php?f=15&t=16616&p=71620#p71620)

popcornmix, 2023-10-04:
I think you should report this to Arch maintainers.
We don't know what the difference is between 6.1.54-1-aarch64  and rpi-6.1.54-2-aarch64.

If Arch can confirm exactly what the change is (e.g. config options or a bump in git hash) and the regression came from this repo, we can investigate further.

GazzaC, 2023-10-04:
Ok, i have reported there as a first instance. Here is URL [https://archlinuxarm.org/forum/viewtopic.php?f=15&t=16616&p=71639#p71639](https://archlinuxarm.org/forum/viewtopic.php?f=15&t=16616&p=71639#p71639)

graysky2, 2023-10-05:
> If Arch can confirm exactly what the change is (e.g. config options or a bump in git hash) and the regression came from this repo, we can investigate further.

I can: see [here](https://github.com/archlinuxarm/PKGBUILDs/commit/9a7666279d57b80deeef34558b8080d9f8a4330c)
6.1.54-1 hash = fad58933544bb2a7b7db92847c25c79a83171fa6
6.1.54-2 hash = 9730a2e44d9e9b19cc126e757fc271f377e5cc18

You can see the differences in the config options at the link above.  ArchARM closely follows your defconfigs.  Basically, we just add a few options to them.

pelwell, 2023-10-05:
The differences amount to the Pi5 mega-merge and:
```
cc08810f89e52 overlays: Add a sample hat_map
df4d584e66a66 Revert "drm/edid: Add option to report basic audio support with a generic edid"
```

graysky2, 2023-10-06:
Thanks for clarifying the diffs.  Is there anything commit within there you might suspect as causing the issue the OP could revert and check functionality?

pelwell, 2023-10-06:
Not really - there have been many changes to the VC4 driver, but I wouldn't like to guess which one (if any) it might be - it's not my specialist subject. Step one is going to be to reproduce the problem.

popcornmix, 2023-10-06:
I note you are running the fkms driver - any reason (it is deprecated)?
Can you switch to kms driver and report if hang still occurs?

GazzaC, 2023-10-06:
Hey guys, my use case is dual head or (extended display). I use Remmina with multi-monitor via VPN for my work..etc, its just greener energy :) .. I been running Arch like this for many years for local desktop as well.
I could only get the pi4 to run extended desktop with the fkms driver. I don't know of any other way to do dual displays (extended desktop) on pi4.
I'll ask on Arch forum if anyone else is experiencing issue, or, I am happy to reproduce the issue if required.

spockfish, 2023-10-07:
I can confirm this issue.

This occured between 6.1.52 and 6.1.54.
As I'm using a Buildroot environment I can guarantee that the only thing that has changed is the kernel.

Commenting out the driver (vc4-fkms-v3d-pi4) fixes the issue.

spockfish, 2023-10-07:
I can also confirm that with the 'vc4-kms-v3d-pi4' the system is *not* hanging.

But no picture, I suspect that's related to the Qt application.

popcornmix, 2023-10-07:
> I could only get the pi4 to run extended desktop with the fkms driver. I don't know of any other way to do dual displays (extended desktop) on pi4.

The kms driver should do everything the fkms does (including dual desktop).

GazzaC, 2023-10-07:
> > I could only get the pi4 to run extended desktop with the fkms driver. I don't know of any other way to do dual displays (extended desktop) on pi4.
> 
> The kms driver should do everything the fkms does (including dual desktop).

Ok, nice. I would love to know how to do that. All the config.txt options i have tried with kms only give me single display. Wish i knew how to turn that on.

GazzaC, 2023-10-07:
Ok. I managed to turn on dual displays under kms by reading thread https://github.com/raspberrypi/linux/issues/5178

my config.txt to do that is - just the relevant parts

dtoverlay=vc4-kms-v3d
display_auto_detect=1
dtparam=audio=on
disable_overscan=1
hdmi_force_hotplug:0=1

cant do this now 
$ tvservice
tvservice is not supported when using the vc4-kms-v3d driver.
Similar features are available with standard linux tools
such as kmsprint from kms++-utils

I am not 100% sure the above is the correct way to get dual displays, but its working for me.

Anyway back to the issue, does this mean the fkms driver and it being deprecated. Does this mean it might be kaput now?

GazzaC, 2023-10-08:
Ok, I am a bit confused. If i use the kms setting above all is well on kernel 6.1.54.-1 if i upgrade to 6.1.54-2 a cpl of things
1. lightdm doesn't work on both screens with mouse pointer for sign in (just primary)
2. I can only get mirroring to function, no extended desktop, although in settings -> display i see the screens but the id's are both 1 when i turn on identification.

popcornmix, 2023-10-19:
@GazzaC can you test again with a kernel with https://github.com/raspberrypi/linux/pull/5654 merged?
That fixed an issue with extended desktop appearing mirrored.

graysky2, 2023-10-19:
@GazzaC - you can find that in the latest kernel which should be queued for building now. See: https://github.com/archlinuxarm/PKGBUILDs/commit/f508f1aa1b8cad754bafc134f9733dbf3f4de4e2

GazzaC, 2023-10-20:
Yep, thanks guys. Just waiting for it etc. Will let you know how it goes.

GazzaC, 2023-10-20:
Hi Guys,
This works fine. I gave it a good test with X11 etc, different WM's desktops, remmina (multi displays) etc. Works just like the previous setup i had using FKMS. I am now switched over to KMS. Note: FKMS still hangs system, if trying to use.

I genuinely appreciate this fix. Thanks heaps.

For any readers curious about the setup, here are the cmdline.txt and config.txt setting that worked for me. 
Because i use the 3.5mm jack for a headset.
with FKMS driver the audio jack just worked, with KMS, add snd_bcm2835.enable_headphones=1

cmdline.txt - using headset (3.5mm audio jack) & HDMI audio available
`root=UUID=<your UUID> rootflags=data=writeback,commit=60,noatime,nodiratime rw rootwait snd_bcm2835.enable_headphones=1 console=serial0,115200 console=tty1 selinux=0 smsc95xx.turbo_mode=N dwc_otg.lpm_enable=0 kgdboc=serial0,115200`

config.txt - using active cooling
```
# See /boot/overlays/README for all available options

dtoverlay=vc4-kms-v3d-pi4
display_auto_detect=1
initramfs initramfs-linux.img followkernel

# Uncomment to enable bluetooth
dtparam=krnbt=on
# sound
dtparam=audio=on
disable_overscan=1
arm_64bit=1
hdmi_force_hotplug:1=1
hdmi_force_hotplug:0=1
hdmi_enable_4kp60=0

[pi4]
# Run as fast as firmware / board allows
#arm_boost=1
over_voltage=10
arm_freq=2300
gpu_freq=900
v3d_freq=900
# usb tweaks
otg_mode=1
```
Cheers,

popcornmix, 2023-10-20:
> dtoverlay=vc4-kms-v3d-pi4

`dtoverlay=vc4-kms-v3d` is preferred. It will work on any model of a pi (choosing the appropriate board specific variation).

GazzaC, 2023-10-20:
ok, great. thanks.

GazzaC, 2023-10-22:
no issues after testing for a few days, closing

closed 2023-10-22 by GazzaC