raspberrypi/linux #5625: gpiomem

bulletmark, 2023-10-02:
### Describe the bug

On Arch ARM, since the update of `linux-rpi` from `6.1.54-1 -> 6.1.54-2` 2 days ago, I can not set `/dev/gpiomem` ownership using a [udev rule](https://github.com/bulletmark/wccontrol/blob/master/gpio.rules) that I have used for years, so my app fails with a `RuntimeError: No access to /dev/mem.`

Downgrading to the previous version of `linux-rpi` (`6.1.54-2 -> 6.1.54-1`) fixes the issue.

### Steps to reproduce the behaviour

As above.

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
$ uname -a
Linux pi4 6.1.54-1-rpi-ARCH #1 SMP Tue Sep 19 12:15:06 MDT 2023 armv7l GNU/Linux

$ sudo vcgencmd version
Sep 21 2023 18:42:00 
Copyright (c) 2012 Broadcom
version 3094eda5aadf2c719edd2c68d1900884abc3dbca (clean) (release) (start)
```

bulletmark, 2023-10-02:
I should add that I tried some other udev rule incantations to change group/permissions that I found around the web, including the [one in the Arch ARM wiki](https://archlinuxarm.org/wiki/Raspberry_Pi), but none worked either with that new kernel.

6by9, 2023-10-02:
The driver has changed to a more generic version that works with RP1 on Pi5.
https://github.com/raspberrypi/linux/commit/fdf9cab5eaa849e90b12e17718bc47130a91433c
https://github.com/raspberrypi/linux/commit/27bda80061b46e18fe83be11228df5365363b377
The device name is now rpi-gpiomem rather than bcm2835-gpiomem .

pelwell, 2023-10-02:
Indeed. The RPiOS udev rule uses "*gpiomem*" to match both.

bulletmark, 2023-10-02:
After @6by9's comment I changed:

    SUBSYSTEM=="bcm2835-gpiomem", GROUP="gpio", MODE="0660"

to:

     SUBSYSTEM=="rpi-gpiomem", GROUP="gpio", MODE="0660"

but that didn't make any difference. Then after @pelwell's comment, I changed it to:

    SUBSYSTEM=="gpiomem", GROUP="gpio", MODE="0660"

and that worked.

It's a real shame the old name is not available as some kind of back-wards compatible alias.

pelwell, 2023-10-02:
Unfortunately Linux drivers only have a single name - there is no concept of a fallback name, unlike Device Tree. We could have retained the old driver name, but a) it is a completely new driver written, by us, and b) it really isn't bcm2835-specific.

pelwell, 2023-10-02:
> Then after @pelwell's comment, I changed it to:

    SUBSYSTEM=="gpiomem", GROUP="gpio", MODE="0660"

Interesting - my suggestion was actually to use `"*gpiomem*"`, but the GitHub interpreted the asterisks as MarkDown markup, hence the italics in my post above.

bulletmark, 2023-10-02:
Tried it with the stars and that works also.

closed 2023-10-12 by pelwell