raspberrypi/linux #5626: [RFE] add missing support for pcie endpoint mode

cz172638, 2023-10-02:
### Describe the bug

[pcie-brcmstb code](/raspberrypi/linux/blob/rpi-6.1.y/drivers/pci/controller/pcie-brcmstb.c#L1337) says pcie controller defaults to endpoint mode. with rpi as one of the most spread diy platforms, compute module 4 or rpi5 with its pcie connector is ideal platform for poor-man's compute accelerator.
but lack of documentation (esp. register map) makes such development slow and only based on reverse engineering / trial'n'error effort. would be great if raspberry pi foundation should put some effort in this.

### Steps to reproduce the behaviour

boot cm4

### Device (s)

Raspberry Pi CM4

### System

not needed as it is not yet implemented

pelwell, 2023-10-02:
The IP behind the PCIe block may support endpoint mode, but the instances found in BCM2711 (Pi 4) and BCM2712 (Pi 5) do not.

closed 2023-10-02 by pelwell

ohault, 2023-12-09:
Could it be considered for both
- firmware upgrade for rpi4
- firmware upgrade fir rpi5
- build in for Compute Module v.next

pelwell, 2023-12-09:
No, it can't be made to work - it is not possible.