raspberrypi/linux #5643: raspberrypi-kernel-headers missing in 2023-10-10 bookwork image (64bit, armhf)

delijati, 2023-10-11:
### Describe the bug

In the current build from 2023-10-10 there is a package missing error for "raspberrypi-kernel-headers". Also i can't see the deb package on https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/

### Steps to reproduce the behaviour

Use current image 2023-10-10

### Device (s)

Raspberry Pi CM4

### System

2023-10-10

XECDesign, 2023-10-11:
For bookworm, we have switched to debian-style packaging for the kernel. If you want the headers for kernel8.img, you'd install `linux-headers-rpi-v8`.

delijati, 2023-10-11:
Hmm i tried that [1] but still get [2] 

[1] linux-headers-rpi-v8 is already the newest version (1:6.1.54-1+rpt2).
[2] make[1]: *** /lib/modules/6.1.54-v8+/build: No such file or directory.  Stop.

delijati, 2023-10-11:
ahh i see its now also called `/lib/modules/6.1.0-rpi4-rpi-v8`

XECDesign, 2023-10-11:
Right, it should match the output of `uname -r`

delijati, 2023-10-11:
Perfect it works :)

neverkid, 2023-12-05:
can 6.1.0-rpi6-rpi-v8 support AppArmor module？

XECDesign, 2023-12-05:
As far as I can tell, apparmor is already compiled in. But, that's not related to this issue, so it's probably best discussed on the forum.

neverkid, 2023-12-07:
Thank you for your answer. I didn't realize that the main issue has nothing to do with it. Thank you for the reminder.