raspberrypi/linux #5649: CM4: Oops on vc4_platform_drm_probe

michael1413, 2023-10-13:
### Describe the bug

Kernel Ooops on loading the VC4 driver with Kernel 6.1.57.
With kernel 6.1.23 everything is still fine.

### Steps to reproduce the behaviour

Kernel settings:
[config.zip](https://github.com/raspberrypi/linux/files/12893152/config.zip)

Boot config:
[config.txt](https://github.com/raspberrypi/linux/files/12893218/config.txt)

RPI-Firmware
a3b20e7dd41aa174ae50b872c8c9c66beafd63e2

### Device (s)

Raspberry Pi CM4

### System

Raspberry CM4

vcos_build_version     = 3094eda5aadf2c719edd2c68d1900884abc3dbca (clean)
vcos_build_branch      = bcm2711_2
debug_sym: ReadVideoCoreMemoryBySymbol: Symbol not found: 'vcos_build_application'
vcos_build_application = ?
vcos_build_date        = Sep 21 2023
vcos_build_time        = 18:42:00
vcos_build_user        = dom
vcos_build_hostname    = buildbot
vcos_build_platform    = raspberrypi_linux
debug_sym: ReadVideoCoreMemoryBySymbol: Symbol not found: 'vcos_build_type'
vcos_build_type        = ?

Linux version 6.1.57-v7l (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot 2023.08-655-g07a03a8549-dirty) 13.2.0, GNU ld (GNU Binutils) 2.39) #3 SMP PREEMPT Fri Oct 13 13:31:16 CEST 2023

### Logs

 
[    2.522595] Unable to handle kernel NULL pointer dereference at virtual addre                                                                                                                                                                                                                                                                     ss 00000004
[    2.522602] [00000004] *pgd=02974003, *pmd=00000000
[    2.522622] Internal error: Oops: 206 [#1] PREEMPT SMP ARM
[    2.522628] Modules linked in: vc4(+) snd_soc_hdmi_codec snd_soc_core rtc_ds1                                                                                                                                                                                                                                                                     307 regmap_i2c brcmfmac snd_pcm_dmaengine mcp251x snd_pcm dwc2(+) brcmutil cpufr                                                                                                                                                                                                                                                                     eq_dt can_dev cfg80211 snd roles raspberrypi_cpufreq raspberrypi_hwmon rfkill so                                                                                                                                                                                                                                                                     undcore i2c_bcm2835 spi_bcm2835 drm_display_helper drm_dma_helper drm_kms_helper                                                                                                                                                                                                                                                                      nvmem_rmem uio_pdrv_genirq uio
[    2.522709] CPU: 2 PID: 134 Comm: udevd Not tainted 6.1.57-v7l #3
[    2.522716] Hardware name: BCM2711
[    2.522720] PC is at component_master_add_with_match+0x24/0x100
[    2.522738] LR is at vc4_platform_drm_probe+0xa8/0xd4 [vc4]
[    2.522821] pc : [<c075c790>]    lr : [<bf59db98>]    psr: 600f0013
[    2.522825] sp : f0aa9bc8  ip : f0aa9bf0  fp : f0aa9bec
[    2.522829] r10: 00000000  r9 : bf5c1a10  r8 : bf5c1a10
[    2.522833] r7 : bf5c1a14  r6 : c16b3c10  r5 : bf5ceb00  r4 : 00000000
[    2.522837] r3 : c281a480  r2 : 00000000  r1 : bf5c1a14  r0 : 00000000
[    2.522841] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[    2.522848] Control: 30c5383d  Table: 0294b2c0  DAC: 55555555
[    2.522851] Register r0 information: NULL pointer
[    2.522860] Register r1 information: 66-page vmalloc region starting at 0xbf5                                                                                                                                                                                                                                                                     9a000 allocated at load_module+0x9b0/0x1f6c
[    2.522883] Register r2 information: NULL pointer
[    2.522890] Register r3 information: slab task_struct start c281a480 pointer                                                                                                                                                                                                                                                                      offset 0
[    2.522905] Register r4 information: NULL pointer
[    2.522912] Register r5 information: 66-page vmalloc region starting at 0xbf5                                                                                                                                                                                                                                                                     9a000 allocated at load_module+0x9b0/0x1f6c
[    2.522924] Register r6 information: slab kmalloc-1k start c16b3c00 pointer o                                                                                                                                                                                                                                                                     ffset 16 size 1024
[    2.522940] Register r7 information: 66-page vmalloc region starting at 0xbf5                                                                                                                                                                                                                                                                     9a000 allocated at load_module+0x9b0/0x1f6c
[    2.522952] Register r8 information: 66-page vmalloc region starting at 0xbf5                                                                                                                                                                                                                                                                     9a000 allocated at load_module+0x9b0/0x1f6c
[    2.522963] Register r9 information: 66-page vmalloc region starting at 0xbf5                                                                                                                                                                                                                                                                     9a000 allocated at load_module+0x9b0/0x1f6c
[    2.522974] Register r10 information: NULL pointer
[    2.522980] Register r11 information: 2-page vmalloc region starting at 0xf0a                                                                                                                                                                                                                                                                     a8000 allocated at kernel_clone+0xb4/0x39c
[    2.522992] Register r12 information: 2-page vmalloc region starting at 0xf0a                                                                                                                                                                                                                                                                     a8000 allocated at kernel_clone+0xb4/0x39c
[    2.523001] Process udevd (pid: 134, stack limit = 0xb4000e63)
[... 50 lines cut ...]
[    2.523416]  r7:c12c31f0 r6:c07663f4 r5:bf5ce09c r4:00000000
[    2.523420]  bus_for_each_dev from driver_attach+0x2c/0x30
[    2.523436]  r6:00000000 r5:c3006380 r4:bf5ce09c
[    2.523439]  driver_attach from bus_add_driver+0x174/0x200
[    2.523470]  bus_add_driver from driver_register+0x98/0x12c
[    2.523502]  r9:00000000 r8:f0aa9ea0 r7:00000000 r6:c3482200 r5:00000000 r4:b                                                                                                                                                                                                                                                                     f5ce09c
[    2.523506]  driver_register from __platform_driver_register+0x2c/0x34
[    2.523535]  r5:c281a480 r4:00000000
[    2.523545]  __platform_driver_register from vc4_drm_register+0x58/0x1000 [vc                                                                                                                                                                                                 [... cut ...]
[    2.523876] 9fc0: 00020000 00000000 00000000 0000017b 01841728 004426f4 00020                                                                                                                                                                                                                                                                     000 004425ca
[    2.523880] 9fe0: bee9f8d8 bee9f8c8 b6e68d8c b6acedc0
[    2.523885]  r6:00000000 r5:00000000 r4:00020000
[    2.523890] Code: e28dd004 e1a06000 e1a07001 e1a00002 (e5921004)

6by9, 2023-10-13:
Checking your config.txt, you appear to be running FKMS.
Whilst we will still provide some support for it, it is deprecated.

6by9, 2023-10-13:
rpi-firmware bf601ccd8032d47f42fc91b9de2bc0e253f61ba7 (6.1.54) is fine
(6f1dcc758829cfd56d7c2f16513aaefa876902b2 was only a firmware change)
cac01bed1224743104cb2a4103605f269f207b1a (Add support for Pi5) locks up on loading FKMS. So it's probably another "setting of registers when we shouldn't be" issue.

6by9, 2023-10-13:
Chasing so far appears to point to it stalling as it creates the framebuffer bo. 
```
[   35.805571] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[   35.805574] Call trace:
[   35.805577]  dump_backtrace+0xfc/0x108
[   35.805589]  show_stack+0x20/0x30
[   35.805593]  dump_stack_lvl+0x8c/0xb8
[   35.805601]  dump_stack+0x18/0x34
[   35.805605]  drm_gem_dma_create+0x3c/0x158 [drm_dma_helper]
[   35.805624]  drm_gem_dma_create_with_handle+0x60/0xe0 [drm_dma_helper]
[   35.805633]  drm_gem_dma_dumb_create_internal+0xb8/0xf4 [drm_dma_helper]
[   35.805643]  vc5_dumb_create+0xb0/0xe4 [vc4]
[   35.805706]  drm_mode_create_dumb+0x98/0xb8 [drm]
[   35.805862]  drm_client_framebuffer_create+0xc4/0x218 [drm]
[   35.805932]  drm_fb_helper_generic_probe+0xa0/0x250 [drm_kms_helper]
[   35.806000]  __drm_fb_helper_initial_config_and_unlock+0x2f8/0x550 [drm_kms_helper]
[   35.806034]  drm_fbdev_client_hotplug+0x120/0x1c0 [drm_kms_helper]
[   35.806067]  drm_client_register+0x60/0xa8 [drm]
[   35.806137]  drm_fbdev_generic_setup+0xbc/0x168 [drm_kms_helper]
[   35.806170]  vc4_drm_bind+0x3ac/0x3c8 [vc4]
[   35.806204]  try_to_bring_up_aggregate_device+0x170/0x1d8
[   35.806211]  component_master_add_with_match+0xb8/0x100
[   35.806216]  vc4_platform_drm_probe+0xcc/0x110 [vc4]
[   35.806248]  platform_probe+0x70/0xe0
[   35.806255]  really_probe+0xc4/0x2b0
[   35.806262]  __driver_probe_device+0x80/0x120
[   35.806268]  driver_probe_device+0x44/0x128
[   35.806274]  __driver_attach+0x7c/0x130
[   35.806279]  bus_for_each_dev+0x78/0xd0
[   35.806285]  driver_attach+0x2c/0x38
[   35.806290]  bus_add_driver+0x15c/0x210
[   35.806295]  driver_register+0x6c/0x128
[   35.806299]  __platform_driver_register+0x30/0x40
[   35.806303]  vc4_drm_register+0x64/0x1000 [vc4]
[   35.806336]  do_one_initcall+0x50/0x2a8
[   35.806342]  do_init_module+0x50/0x1f8
[   35.806350]  load_module+0x1920/0x1d68
[   35.806356]  __do_sys_init_module+0x19c/0x208
[   35.806362]  __arm64_sys_init_module+0x24/0x30
[   35.806368]  invoke_syscall+0x4c/0x110
[   35.806374]  el0_svc_common.constprop.0+0x68/0x128
[   35.806379]  do_el0_svc+0x34/0xd0
[   35.806384]  el0_svc+0x30/0x98
[   35.806390]  el0t_64_sync_handler+0xb8/0xc0
[   35.806395]  el0t_64_sync+0x18c/0x190
```
It's dying in
```
	if (dma_obj->map_noncoherent) {
		dma_obj->vaddr = dma_alloc_noncoherent(drm->dev, size,
						       &dma_obj->dma_addr,
						       DMA_TO_DEVICE,
						       GFP_KERNEL | __GFP_NOWARN);
	} else {
		dma_obj->vaddr = dma_alloc_wc(drm->dev, size,
					      &dma_obj->dma_addr,
					      GFP_KERNEL | __GFP_NOWARN);
	}
```
https://github.com/raspberrypi/linux/blob/rpi-6.1.y/drivers/gpu/drm/drm_gem_dma_helper.c#L146-L155 (I'm expecting it to take the else clause).

I am also seeing the odd stall waiting on a spinlock as it tries to request_irq for the SMI interrupt. I guess it could be a random buffer overrun, so depending on exactly how the compiler handles things, it varies what gets trashed.

spockfish, 2023-10-13:
Could this be related to #5624? Which is also about FKMS.

popcornmix, 2023-10-19:
@spockfish quite likely related. Fixing fkms would be nice, but the real solution is to switch the actually supported kms driver.

michael1413, 2023-10-20:
> @spockfish quite likely related. Fixing fkms would be nice, but the real solution is to switch the actually supported kms driver.

if it were that easy

004552.421: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
004552.672: dterror: not a valid FDT - err -9
004552.700: Failed to load overlay 'vc4-kms-v3d-pi4'

pelwell, 2023-10-20:
Please upload `overlays/vc4-kms-v3d-pi4.dtbo`  from your image somewhere, or at least the output of `md5sum /boot/overlays/vc4-kms-v3d-pi4.dtbo`.

spockfish, 2023-10-20:
> @spockfish quite likely related. Fixing fkms would be nice, but the real solution is to switch the actually supported kms driver.

Agreed. I've been testing my app (Qt embedded app, no X, directly DRM) on the kms driver, but that's less stable and results (after a few hours) in an kernel error:

`vc4-drm gpu: [drm] *ERROR* Failed to allocate DLIST entry: -28`

spockfish, 2023-10-20:
> > vc4-drm gpu: [drm] _ERROR_ Failed to allocate DLIST entry: -28
> 
> If you can find a reiable way to reproduce this issue, I'm sure it can be fixed, but it's not something I've seen so far.

Yeah. That's the issue. It just pops up after running for hours. So no direct clue what's causing it: the application just shows some information that's sometimes being updated, nothing fancy.

I'm wondering if this isn't related to memory mgt, which for some reason is a little bit more sensitive than with the old driver. Purely speculating here ofc. Right now I've only got a '`gpu_mem=128`' set, but playing around with the value does not change the behaviour.

6by9, 2023-10-20:
DLIST (Display List) is a totally separate memory within the HVS used for holding the list of planes to display.

Unless you're presenting a large number of planes to DRM to render, then I wouldn't have expected you to hit the limit at all. X will normally only ever present a maximum of 2 per display - the desktop, and the cursor.

How many planes are you presenting, and of what image format?

popcornmix, 2023-10-20:
> Right now I've only got a 'gpu_mem=128' set, but playing around with the value does not change the behaviour.

Just remove that line. gpu_mem use is largely deprecated (it it only now used by firmware side h264 decode, and the default is sufficient for playing BluRay video). Setting it higher is almost certainly just wasting that memory (no one else can use it).

michael1413, 2023-10-20:
> Please upload `overlays/vc4-kms-v3d-pi4.dtbo` from your image somewhere, or at least the output of `md5sum /boot/overlays/vc4-kms-v3d-pi4.dtbo`.

I used the actual version from raspberrypi/firmware
With the kernel version of the vc4-kms-v3d-pi4.dtbo the error is gone but i get only a blank screen.
I think the problem is that the EDID can't be read.

004082.768: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004083.786: hdmi: HDMI1:EDID giving up on reading EDID block 0
004088.827: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004089.842: hdmi: HDMI1:EDID giving up on reading EDID block 0
004089.861: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
004089.876: HDMI0: hdmi_pixel_encoding: 300000000
004089.889: HDMI1: hdmi_pixel_encoding: 300000000

hdmi_group=2
hdmi_mode=87
hdmi_cvt=800 1280 60 5 0 0 1

I have tried to load the edid over the cmdline " drm.edid_firmware=HDMI-A-1:display.dat" but without result.

pelwell, 2023-10-20:
It sounds like your firmware download has failed in some way. Try this:
```
$ wget https://github.com/raspberrypi/firmware/raw/master/boot/overlays/vc4-kms-v3d-pi4.dtbo
$ md5sum vc4-kms-v3d-pi4.dtbo
4ffeeb5d8d6bbf49f2c97b6340804773 vc4-kms-v3d-pi4.dtbo
```
which is exactly the same as you get from the top-of-tree kernel.

popcornmix, 2023-10-20:
> 004082.768: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004083.786: hdmi: HDMI1:EDID giving up on reading EDID block 0
004088.827: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
004089.842: hdmi: HDMI1:EDID giving up on reading EDID block 0

HDMI1 is the second hdmi connector, so if you are not using it, a failed edid read is expected.
It is HDMI0 that is important.

> hdmi_group=2
hdmi_mode=87
hdmi_cvt=800 1280 60 5 0 0 1

Remove those - they apply to deprecated fkms driver.

When you boot with blank screen do you have ssh access?
If so post output of `raspinfo` (e.g. pipe it through `pastebinit`)

Also if you connect to hdmi and have no boot media, you should get a hdmi diagnostic screen.
What does the "display:" line show.

spockfish, 2023-10-20:
> DLIST (Display List) is a totally separate memory within the HVS used for holding the list of planes to display.
>
> Unless you're presenting a large number of planes to DRM to render, then I wouldn't [...]

uhhhhh.... to be honest: I dunno :-)
We're talking about an Qt application for the original touchscreen that shows player information (RoPieee, a Raspberry Pi player for Roon). I use Qt on an a high abstract level and don't deal with planes directly.

And with the old driver it is extremely stable: it can run for weeks. 
Just swapping out the driver causes this crash.

6by9, 2023-10-20:
```kmsprint``` will tell you what is on the display.

It's possible there's a memory leak in the dlist allocator, but we need a way to reproduce or at least identify the problem.
I'll see if I can add a debugfs node that dumps the allocations.

popcornmix, 2023-10-20:
A search shows the DLIST message has occurred for one forum user: 
https://forums.raspberrypi.com/viewtopic.php?t=357826

(mentioned in two thread, but from same user).

Not seeing any other hits, so it's not widespread.
Would be good to understand what use provokes it.

spockfish, 2023-10-20:
Is the 'max_framebuffers=2' could have something to do with it?
I've copy/pasted this from some examples that uses the KMS driver, but in the FKMS setup this setting is not present.

Gonna test this... writing this down it seems stupid that I didn't think of this before :-)

6by9, 2023-10-20:
Found the commit that breaks FKMS - https://github.com/raspberrypi/linux/commit/3fb20da9f3bff93f20b1a1fe2c950c6c8f4d1e1e

We're still requesting the interrupt, but then not handling it. A simple revert seems to work on Pi4. FKMS isn't officially supported on Pi5, but I may look into whether we can handle it better.

michael1413, 2023-10-20:
> HDMI1 is the second hdmi connector, so if you are not using it, a failed edid read is expected. It is HDMI0 that is important.
>
>
> Remove those - they apply to deprecated fkms driver.
>
> When you boot [...]

[Raspinfo logfile](https://bluegate.go-sys.de/download/test.log)

popcornmix, 2023-10-20:
> hdmi_group=2
hdmi_mode=87
hdmi_cvt=800 1280 60 5 0 0 1
hdmi_force_hotplug=1

as I said before, remote those, they apply to deprecated firmware driver

> /sys/class/drm/card[0-9]-*
No EDID found

What display are you connected to? You are using a CM4? Is this though RPi I/O board, or your own board?
Is it asserting hotplug? (e.g. can you probe the line) Is it providing an edid?

> I have tried to load the edid over the cmdline " drm.edid_firmware=HDMI-A-1:display.dat" but without result.

show output of `base64 display.dat`.
Also check dmesg for lines containing edid (it should report a few times loading edid from display.dat.

When booted what is the output of `kmsprint` (initially with no `video=` or `drm.edid_firmware` line, then with just the 
`drm.edid_firmware` line).

closed 2023-10-20 by pelwell

michael1413, 2023-10-25:
> as I said before, remote those, they apply to deprecated firmware driver
>
>
> What display are you connected to? You are using a CM4? Is this though RPi I/O board, or your own board? Is it asserting [...]

I've now tried it with a CM3 and this works without an edid file.
cmdline.txt: video=HDMI-A-1:800x1280@60D 

kmsprint output CM3 dtoverlay=vc4-fkms-v3d
Connector 0 (32) HDMI-A-1 (connected)
  Encoder 0 (31) TMDS
    Crtc 3 (90) 800x1280@61.58 76.000 800/64/32/32/+ 1280/40/6/4/+ 62 (61.58) 0x5 0x48
      Plane 3 (80) fb-id: 295 (crtcs: 3) 0,0 800x1280 -> 0,0 800x1280 (XR24 AR24 AB2ente0121!!
dmes4 XB24 RG16 BG16 AR15 XR15 RG24 BG24 YU16 YV16 YU24 YV24 YU12 YV12 NV12 NV21 NV16 NV61 RGB8 BGR8 XR12 AR12 XB12 AB12 BX12 BA12 RX12 RA12)
        FB 295 800x1280

kmsprint output CM4 dtoverlay=vc4-kms-v3d-pi4:
terminate called after throwing an instance of 'std::runtime_error'
  what():  No modesetting DRM card found
Aborted

kmsprint output CM4 dtoverlay=vc4-fkms-v3d-pi4:
Connector 0 (89) HDMI-A-1 (connected)
  Encoder 0 (88) TMDS
    Crtc 0 (87) 800x1280@61.58 76.000 800/64/32/32/+ 1280/40/6/4/+ 62 (61.58) 0x5 0x48
      Plane 0 (31) fb-id: 99 (crtcs: 0) 0,0 800x1280 -> 0,0 800x1280 (XR24 AR24 XB24 AB24 RX24 BX24 RG16 RG24 BG24 YU16 YU12 YV12 NV12 NV21 P030)
        FB 99 800x1280

Tested with the same Kernel

michael1413, 2023-10-25:
On the same hardware with the same display, just with a CM4s

kmsprint output CM4s dtoverlay=vc4-kms-v3d-pi4:
terminate called after throwing an instance of 'std::runtime_error'
  what():  No modesetting DRM card found
Aborted

kmsprint output CM4s dtoverlay=vc4-fkms-v3d-pi4:
Connector 0 (89) HDMI-A-1 (connected)
  Encoder 0 (88) TMDS
    Crtc 0 (87) 800x1280@61.58 76.000 800/64/32/32/+ 1280/40/6/4/+ 62 (61.58) 0x5 0x48
      Plane 0 (31) fb-id: 99 (crtcs: 0) 0,0 800x1280 -> 0,0 800x1280 (XR24 AR24 XB24 AB24 RX24 BX24 RG16 RG24 BG24 YU16 YU12 YV12 NV12 NV21 P030)
        FB 99 800x1280

popcornmix, 2023-10-25:
> AP///////wAF1wAAAAAAAP8hAQOAMh94B+6Vo1RMmSYPUFQAAAABAQEBAQEBAQEBAQEBAQEBvBsA
oFAgFzAwIDYAQMgQAAAaAAAA/QAX8A//BwAKICAgICAgAAAA/AAxMjgweDgwMCBXWEdBAAAAAAAA
AAAAAAAAAAAAAAAAAXYCAxhAIwkGB2cDDAAAABAPZ9hdxAEPgAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA5Q==
I created this file with AW EDID Editor.

```
$ echo "AP///////wAF1wAAAAAAAP8hAQOAMh94B+6Vo1RMmSYPUFQAAAABAQEBAQEBAQEBAQEBAQEBvBsA
oFAgFzAwIDYAQMgQAAAaAAAA/QAX8A//BwAKICAgICAgAAAA/AAxMjgweDgwMCBXWEdBAAAAAAAA
AAAAAAAAAAAAAAAAAXYCAxhAIwkGB2cDDAAAABAPZ9hdxAEPgAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA5Q==" | base64 -d | edid-decode
edid-decode (hex):

00 ff ff ff ff ff ff 00 05 d7 00 00 00 00 00 00
ff 21 01 03 80 32 1f 78 07 ee 95 a3 54 4c 99 26
0f 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01
01 01 01 01 01 01 bc 1b 00 a0 50 20 17 30 30 20
36 00 40 c8 10 00 00 1a 00 00 00 fd 00 17 f0 0f
ff 07 00 0a 20 20 20 20 20 20 00 00 00 fc 00 31
32 38 30 78 38 30 30 20 57 58 47 41 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 76

02 03 18 40 23 09 06 07 67 03 0c 00 00 00 10 0f
67 d8 5d c4 01 0f 80 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 e5

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: ANW
    Model: 0
    Model year: 2023
  Basic Display Parameters & Features:
    Digital display
    Maximum image size: 50 cm x 31 cm
    Gamma: 2.20
    Monochrome or grayscale display
    Default (sRGB) color space is primary color space
    First detailed timing is the preferred timing
    Supports GTF timings within operating range
  Color Characteristics:
    Red  : 0.6396, 0.3300
    Green: 0.2998, 0.5996
    Blue : 0.1503, 0.0595
    White: 0.3134, 0.3291
  Established Timings I & II: none
  Standard Timings: none
  Detailed Timing Descriptors:
    DTD 1:  1280x800    59.909545 Hz   8:5     49.306 kHz     71.000000 MHz (320 mm x 200 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol P
                 Vfront    3 Vsync   6 Vback   14 Vpol N
    Display Range Limits:
      Monitor ranges (GTF): 23-240 Hz V, 15-255 kHz H, max dotclock 70 MHz
    Display Product Name: '1280x800 WXGA'
    Empty Descriptor
  Extension blocks: 1
Checksum: 0x76

----------------

Block 1, CTA-861 Extension Block:
  Revision: 3
  Basic audio support
  Native detailed modes: 0
  Audio Data Block:
    Linear PCM:
      Max channels: 2
      Supported sample rates (kHz): 48 44.1
      Supported sample sizes (bits): 24 20 16
  Vendor-Specific Data Block (HDMI), OUI 00-0C-03:
    Source physical address: 0.0.0.0
    DC_30bit
    Maximum TMDS clock: 75 MHz
  Vendor-Specific Data Block (HDMI Forum), OUI C4-5D-D8:
    Version: 1
    Maximum TMDS Character Rate: 75 MHz
    SCDC Present
Checksum: 0xe5
```
This edid is for a 1280x800@59.9 resolution.
But you also say `video=HDMI-A-1:800x1280@60D`

Are you wanting 1280x800 or 800x1280? They are different timings.

michael1413, 2023-10-25:
I created the edid using a windows tool and tried out some settings.
I tested it with 800x1280 and 1280x800 and the result was the same.

with the "video=HDMI-A-1:800x1280@60D" I don't need an edid file on the CM3.

I have testet it with a default 800x600 edid file and on the CM3 i get output on the CM4 don't.

michael1413, 2023-10-25:
> This edid is for a 1280x800@59.9 resolution. But you also say `video=HDMI-A-1:800x1280@60D`
> 
> Are you wanting 1280x800 or 800x1280? They are different timings.

I created the edid using a windows tool and tried out some settings.
I tested it with 800x1280 and 1280x800 and the result was the same.

with the "video=HDMI-A-1:800x1280@60D" I don't need an edid file on the CM3.

I have testet it with a default 800x600 edid file and on the CM3 i get output on the CM4 don't.

michael1413, 2024-02-15:
The error "Oops on vc4_platform_drm_probe" is back in kernel version  6.1.77-v7l

popcornmix, 2024-02-15:
Are you still using fkms? (which isn't a default, nor recommended).

michael1413, 2024-02-15:
Yes, I can't switch to the kms. The fkms works fine on kernel  6.1.58.

I can't use the kms driver with the cm4. The kms works with the cm3 but not with the cm4.
I have spend a lot of time to test it with the 58 kernel.

popcornmix, 2024-02-15:
It may be worth testing the 6.6 kernel. That will shortly become the stable kernel from apt.
See [here](https://forums.raspberrypi.com/viewtopic.php?t=361116).

6by9, 2024-02-15:
` Failed to allocate DLIST entry: -28` should have been resolved through #5935 (that is on 6.6).

Do we start being a bit more vocal on deprecating vc4-kms-v3d using overlay_map.dts? I'm not sure if we can as it also renames for bcm2711 and bcm2712.

pelwell, 2024-02-15:
> deprecating vc4-kms-v3d using overlay_map.dts

What did you have in mind?