raspberrypi/linux #5652: i2c1-pi5 dtbo load error on 6.1

digitalLumberjack, 2023-10-15:
### Describe the bug

On commit 392b9849c8d147c25557fea5d6dbcddbcb9e72d5

Trying to load the dtbo `i2c1-pi5` on pi5:
```
dtoverlay i2c1-pi5
```
will lead to the following error:
```
[  121.186991] i2c_designware 1f00074000.i2c: prop pinctrl-0 index 0 invalid phandle
```

### Steps to reproduce the behaviour

dtoverlay i2c1-pi5

### Device (s)

Other

### System

Recalbox 9.2 alpha

```
# vcgencmd version
2023/09/13 11:37:06 
Copyright (c) 2012 Broadcom
version 9d494316 (release) (embedded)
```

```
# uname -a
Linux RECALBOX 6.1.57-v8_16k #1 SMP PREEMPT Sun Oct 15 15:54:49 CEST 2023 aarch64 GNU/Linux
```

### Logs

[  121.186991] i2c_designware 1f00074000.i2c: prop pinctrl-0 index 0 invalid phandle

### Additional context

It seems to be related to the `target = <&frag0>;` in `i2c1-pi5-overlay.dts`.

Using the i2c1 target in fragments fix the issue, but may I suggest using the same pattern for i2c0 and i2c1 than the `i2c2-pi5-overlay.dts`, that also fix the issue.

```
/dts-v1/;
/plugin/;

/{
	compatible = "brcm,bcm2712";

	fragment@0 {
		target = <&i2c1>;
		frag0: __overlay__ {
			status = "okay";
			clock-frequency = <100000>;
			pinctrl-0 = <&rp1_i2c1_2_3>;
		};
	};

	__overrides__ {
		pins_2_3 = <&frag0>,"pinctrl-0:0=", <&rp1_i2c1_2_3>;
		pins_10_11 = <&frag0>,"pinctrl-0:0=", <&rp1_i2c1_10_11>;
		baudrate = <&frag0>, "clock-frequency:0";
	};
};
```

pelwell, 2023-10-16:
`dtoverlay=i2c1` (the `-pi5` is implied there, and will be in the Linux utlity once the change rolls out) works as expected. Applying overlays at runtime has never been a priority, but for the I2C subsystem it ought to work. Your rewrite is a good one, and if I was starting from scratch that's probably what I would have ended up with.

I'll take a look at the `dtoverlay` command and how it is failing, but until then I suggest you carry on with your modified overlay or letting the firmware do the application

pelwell, 2023-11-02:
A potential fix for these problems can be found here: https://github.com/raspberrypi/utils/pull/54