raspberrypi/linux #5656: Kodi 20 + Bookworm - no HW decoding on H.264 videos

mishka81, 2023-10-16:
### Describe the bug

When playing H.264 videos and pressing the "o" key to display video information, I see that the SW decoder is used.

It used to be "HW" on Bullseye ... 

### Steps to reproduce the behaviour

install kodi
play a h.264 video and press "o"
the decoder should be "HW"

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

Aug 10 2023 15:33:38 
Copyright (c) 2012 Broadcom
version 03dc77429335caee083e22ddc8eec09c07f12a7a (clean) (release) (start)

Linux rpi4 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux

### Logs

2023-10-16 19:11:22.532 T:11071    info <general>: VideoPlayer::OpenFile: /media/pi/NAS/tmp/XX.mp4
2023-10-16 19:11:22.538 T:21407    info <general>: Creating InputStream
2023-10-16 19:11:22.555 T:21407    info <general>: ScanPathsForAssociatedItems: found associated file /media/pi/NAS/tmp/XX.srt
2023-10-16 19:11:22.556 T:21407    info <general>: Creating Demuxer
2023-10-16 19:11:22.999 T:21407    info <general>: Opening stream: 0 source: 256
2023-10-16 19:11:23.000 T:21407    info <general>: Creating video codec with codec id: 27
2023-10-16 19:11:23.000 T:21407    info <general>: CDVDVideoCodecFFmpeg::Open() Using codec: H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
2023-10-16 19:11:23.001 T:21407    info <general>: Creating video thread
2023-10-16 19:11:23.001 T:21412    info <general>: running thread: video_thread
2023-10-16 19:11:23.001 T:21407    info <general>: Opening stream: 1 source: 256
2023-10-16 19:11:23.002 T:21407    info <general>: Finding audio codec for: 86018
2023-10-16 19:11:23.003 T:21407    info <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder aac
2023-10-16 19:11:23.003 T:21407    info <general>: OpenStream: Allowing max Out-Of-Sync Value of 10 ms
2023-10-16 19:11:23.003 T:21407    info <general>: Creating audio thread
2023-10-16 19:11:23.004 T:21413    info <general>: running thread: CVideoPlayerAudio::Process()
2023-10-16 19:11:23.004 T:21407    info <general>: Opening stream: 0 source: 1024
2023-10-16 19:11:23.004 T:21407    info <general>: CDVDSubtitlesLibass: Using libass version 1701000
2023-10-16 19:11:23.004 T:21407    info <general>: CDVDSubtitlesLibass: Creating ASS library structure
2023-10-16 19:11:23.004 T:21407    info <general>: CDVDSubtitlesLibass: Initializing ASS Renderer
2023-10-16 19:11:23.005 T:21407    info <general>: CDVDSubtitlesLibass: Initializing ASS library font settings
2023-10-16 19:11:23.034 T:21407    info <general>: CDVDSubtitlesLibass: Creating new ASS track
2023-10-16 19:11:23.045 T:21412    info <general>: CDVDVideoCodecFFmpeg::Open() Using codec: H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
2023-10-16 19:11:23.049 T:21413    info <general>: Creating audio stream (codec id: 86018, channels: 6, sample rate: 48000, no pass-through)
2023-10-16 19:11:23.050 T:11091    info <general>: CActiveAESink::OpenSink - initialize sink
2023-10-16 19:11:23.050 T:11091    info <general>: CAESinkALSA::Initialize - Attempting to open device "@"
2023-10-16 19:11:23.058 T:11091    info <general>: CAESinkALSA::Initialize - Opened device "sysdefault"
2023-10-16 19:11:23.059 T:11091    info <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2023-10-16 19:11:23.060 T:11091    info <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S24NE3
2023-10-16 19:11:23.104 T:11071    info <general>: GLES: Selecting single pass rendering
2023-10-16 19:11:23.104 T:11071    info <general>: GLES: Selecting YUV 2 RGB shader
2023-10-16 19:11:23.419 T:21412    info <general>: CDVDVideoCodecFFmpeg::CDropControl: calculated diff time: 41708

popcornmix, 2023-10-16:
Can you check in settings/videos/processing and make sure:

- Allow using DRM PRIME decoder is enabled
- Allow hardware acceleration with DRM PRIME is enabled
- PRIME render method is Direct to Plane

mishka81, 2023-10-16:
Thanks, that whas the issue.

I did a fresh install of kodi, don't know why these options were not the default.

closed 2023-10-16 by mishka81