raspberrypi/linux #5667: [REGRESSION/Bookworm] CM4/Pi 4 serial port enumeration unpredictable when enabling uart2-5

stefanb2, 2023-10-20:
### Describe the bug

On our `bullseye-arm64`-based baseline that we use to build lite images (`stage2`) for a CM4 project we enable UART0, 2 & 5 in `/boot/config.txt`:

~~~text
# Disable BT: this frees up UART0 (PL011 UART)
dtoverlay=disable-bt

# UART0 (PL011): /dev/serial0 -> /dev/ttyAMA0
# Linux kernel serial console
# GPIO 14: TXD0
# GPIO 15: RXD0

# UART1 (miniUART): /dev/serial1 -> /dev/ttyS0
# disabled
#enable_uart=1

# UART2 (PL011): /dev/ttyAMA1
# GPS UART
# GPIO 00: TXD2
# GPIO 01: RXD2
dtoverlay=uart2

# UART5 (PL011): /dev/ttyAMA2
# Tester UART
# GPIO 12: TXD5
# GPIO 13: RXD5
dtoverlay=uart5
~~~

After merging Bookworm changes from current `arm64` branch the device enumeration changes and there are UART warnings in the kernel log:

UART GPIO pins | kernel device | connected to
------------------------|---------------------|--------------------
UART0 14/15       | `ttyAMA1`       | tester console
UART2 00/01       | `ttyAMA2`       | GPS
UART5 12/13       | `ttyAMA0`       | Linux console

To make sure that the issue is not caused by our customization, I installed official Raspi OS Lite/Bookworm image on a Pi 4 Model B. After adding the lines to `/boot/firmware/config.txt` & rebooting, I was able to reproduce the issue.

### Steps to reproduce the behaviour

Add to `/boot/firmware/config.txt` & reboot:

~~~text
dtoverlay=disable-bt
dtoverlay=uart2
dtoverlay=uart5
~~~

The issue & kernel warnings appear as soon as you enable `uart2` or `uart5` in `config.txt`

### Device (s)

Raspberry Pi CM4

### System

**NOTE:** output from Pi 4 Model B with reference OS.

~~~sh
# cat /etc/rpi-issue 
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2

# vcgencmd version
Aug 10 2023 15:33:38 
Copyright (c) 2012 Broadcom
version 03dc77429335caee083e22ddc8eec09c07f12a7a (clean) (release) (start)

# uname -a
Linux <REDACTED> 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux
~~~

**NOTE:** output from CM4 with older Bullseye OS to compare with:

~~~sh
# cat /etc/rpi-issue 
Raspberry Pi reference 2023-10-17
Generated using pi-gen, git@<REDACTED>:<REDACTED>/pi-gen.git, 55e1f946f339b1a67756b7c42ed8adbba34ee635, stage2

# vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)

# uname -a
Linux <REDACTED> 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
~~~

### Logs

**NOTE:** output from Pi 4 Model B with reference OS.

~~~sh
# dmesg | grep -i -e " Linux " -e uart 
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.0-rpi4-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05)
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=<REDACTED> vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=12402fc5-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=FI
[    0.054776] Serial: AMBA PL011 UART driver
[    1.688024] usb usb1: Manufacturer: Linux 6.1.0-rpi4-rpi-v8 xhci-hcd
[    1.690254] usb usb2: Manufacturer: Linux 6.1.0-rpi4-rpi-v8 xhci-hcd
[    1.780273] uart-pl011 fe201a00.serial: aliased and non-aliased serial devices found in device tree. Serial port enumeration may be unpredictable.
[    2.296461] uart-pl011 fe201000.serial: there is not valid maps for state default
[    2.309221] uart-pl011 fe201000.serial: cts_event_workaround enabled
[    2.343964] uart-pl011 fe201000.serial: requested serial port 0  not available.
[    3.343999] uart-pl011 fe201000.serial: aliased and non-aliased serial devices found in device tree. Serial port enumeration may be unpredictable.
[    6.239361] uart-pl011 fe201a00.serial: no DMA platform data
[   10.426650] mc: Linux media interface: v0.10
[   10.579638] videodev: Linux video capture interface: v2.00
~~~

**NOTE:** output from CM4 with older Bullseye OS to compare with:

~~~sh
# dmesg | grep -i -e " Linux " -e uart
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=<REDACTED> vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=7bd92406-02 rootfstype=ext4 fsck.repair=yes rootwait
[    0.037513] Serial: AMBA PL011 UART driver
[    1.806461] usb usb1: Manufacturer: Linux 6.1.21-v8+ xhci-hcd
[    1.850992] uart-pl011 fe201000.serial: there is not valid maps for state default
[    1.851753] uart-pl011 fe201000.serial: cts_event_workaround enabled
[    4.141100] uart-pl011 fe201000.serial: no DMA platform data
[    8.687682] uart-pl011 fe201400.serial: no DMA platform data
[   45.796969] node: epoll_ctl support in io_uring is deprecated and will be removed in a future Linux kernel version.
~~~

As you can see the `alias` and `serial port 0` warnings are new.

pelwell, 2023-10-20:
This issue is caused by a bug in the firmware, a bug which was fixed on September 7th. However, the stable branch (from which the firmware apt package is built) was last updated on August 29th, which predates the fix.

@popcornmix Could we update stable firmware sometime soon?

In the meantime you can confirm that this was the cause of your problem by running "sudo SKIP_KERNEL=1 rpi-update" and agreeing to install the (essentially beta) software.

stefanb2, 2023-10-23:
I can confirm that on the Pi 4 Model B the problem is fixed after running `rpi-update`:

~~~sh
# SKIP_KERNEL=1 rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
FW_REV:5bcbf36d2502b1341ce280e1d4497847d439c726
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
WANT_32BIT:0 WANT_64BIT:1 WANT_PI4:1 WANT_PI5:1
#############################################################
...
##############################################################
Would you like to proceed? (y/N)
 *** Downloading specific firmware revision (this will take a few minutes)
 *** Updating firmware
 *** As requested, not updating kernel
 *** As requested, not updating kernel modules
 *** Updating VideoCore libraries
 *** Using SoftFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 5bcbf36d2502b1341ce280e1d4497847d439c726
 *** A reboot is needed to activate the new firmware
.... rebooted device .....

# dmesg | grep -i -e " Linux " -e uart
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.0-rpi4-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05)
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=E4:5F:01:D7:70:54 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=12402fc5-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=FI
[    0.049651] Serial: AMBA PL011 UART driver
[    1.982686] usb usb1: Manufacturer: Linux 6.1.0-rpi4-rpi-v8 xhci-hcd
[    1.984625] usb usb2: Manufacturer: Linux 6.1.0-rpi4-rpi-v8 xhci-hcd
[    2.102698] uart-pl011 fe201000.serial: there is not valid maps for state default
[    2.103395] uart-pl011 fe201000.serial: cts_event_workaround enabled
[    6.235232] uart-pl011 fe201000.serial: no DMA platform data
[   10.255433] mc: Linux media interface: v0.10
[   10.408154] videodev: Linux video capture interface: v2.00
~~~

With `HOME=/data/root SKIP_KERNEL=1 rpi-update` I was also able to apply the update to the CM4 with limited space on the root partition. I can confirm that the fix also works on that device.

**PS: To whom it may concern** I also noticed that Bookworm uses a new device naming scheme for `uart-pl011` , i.e. it now is `UARTx` -> `/dev/ttyAMAx`:

 UART | Bullseye | Bookworm
---------|--------------|-----------------
0 | ttyAMA0 | ttyAMA0
2 | ttyAMA1 | ttyAMA2
5 | ttyAMA2 | ttyAMA5

pelwell, 2023-10-23:
> PS: To whom it may concern I also noticed that Bookworm uses a new device naming scheme for uart-pl011 , i.e. it now is UARTx -> /dev/ttyAMAx

Yes, this is the result of a a once-in-a-lifetime realignment with the upstream kernel - something that we have put off for several years but which was made more necessary by Pi 5.

stefanb2, 2023-10-23:
IMHO it makes much more sense :smiley:

stefanb2, 2023-10-23:
To make sure I have understood this correctly, I have to wait for an update for the following package?

~~~text
ii  raspi-firmware  1:1.20230829+ds-1~rpt3  all  Raspberry Pi family GPU firmware and bootloaders
~~~

stefanb2, 2023-10-27:
Any idea about the timeframe when the new firmware package will hit stable?

pelwell, 2023-10-27:
The stable branch was updated on Tuesday: https://github.com/raspberrypi/rpi-firmware/commits/stable
A raspi-firmware bump should be imminent.

stefanb2, 2023-11-06:
I can confirm that `raspi-firmware-1:1.20231024+ds-1+rpt1`, that arrived on stable today, fixes the issue.

closed 2023-11-06 by stefanb2