raspberrypi/linux #5668: wiringPi broken since kernel 6.5.7

df8oe, 2023-10-20:
### Describe the bug

Starting with kernel 6.5.7 applications which uses gpios are broken.

### Steps to reproduce the behaviour

e.g. the start of an application which uses GPIO25 throws the message:
`gpio: Unable to open GPIO direction interface for pin 25: No such file or directory
wiringPiISR: unable to open /sys/class/gpio/gpio25/value: No such file or directory`

### Device (s)

Raspberry Pi 3 Mod. B+

### System

Archlinux

version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

Linux raspserver 6.5.8-1-rpi-ARCH #1 SMP PREEMPT

### Logs

ls /sys/class/gpio -l with working kernel 6.5.6:
`--w------- 1 root root 4096 20. Okt 09:35 export
lrwxrwxrwx 1 root root    0 20. Okt 09:38 gpio24 -> ../../devices/platform/soc/3f200000.gpio/gpiochip0/gpio/gpio24
lrwxrwxrwx 1 root root    0 20. Okt 09:38 gpio25 -> ../../devices/platform/soc/3f200000.gpio/gpiochip0/gpio/gpio25
lrwxrwxrwx 1 root root    0 20. Okt 09:38 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0
lrwxrwxrwx 1 root root    0 20. Okt 09:38 gpiochip100 -> ../../devices/gpiochip1/gpio/gpiochip100
lrwxrwxrwx 1 root root    0 20. Okt 09:38 gpiochip512 -> ../../devices/platform/soc/soc:firmware/soc:firmware:expgpio/gpio/gpiochip512
lrwxrwxrwx 1 root root    0 20. Okt 09:38 gpiochip520 -> ../../devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4:1.0/gpio/gpiochip520
--w------- 1 root root 4096 20. Okt 09:38 unexport`

ls /sys/class/gpio -l with non-working kernel 6.5.8 (started with kernel 6.5.7):
`--w------- 1 root root 4096 20. Okt 09:06 export
lrwxrwxrwx 1 root root    0 20. Okt 09:07 gpiochip100 -> ../../devices/gpiochip1/gpio/gpiochip100
lrwxrwxrwx 1 root root    0 20. Okt 09:07 gpiochip512 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip512
lrwxrwxrwx 1 root root    0 20. Okt 09:07 gpiochip566 -> ../../devices/platform/soc/soc:firmware/soc:firmware:expgpio/gpio/gpiochip566
--w------- 1 root root 4096 20. Okt 09:07 unexport`

### Additional context

wiringPi is latest version available.

pelwell, 2023-10-20:
This behaviour is by design.

1. By forcing GPIOs in /sys/class/gpio to be numbered from 0 we've been going against the wishes of the kernel developers for years. As of the 6.5 kernel this results in a warning message in the kernel log.
2. Pi 5 has several different GPIO controllers. One of the drivers - the one for the GPIOs on the SoC - relied on the GPIOs being numbered from 0, but users would clearly prefer the GPIOs on the header (those provided by RP1) to be zero-based.
3. wiringPi has been obsolescent since 2019 when the author (quite reasonably) decided he'd had enough.

As a result of which:
4. People shouldn't be using /sys/class/gpio - used libgpiod instead.
5. If they do insist on using /sys/class/gpio, they need to use it properly (searching for the correct GPIO controller and discovering its base number) and not just assume that they start at zero.
6. Giving up on wiringPi now will save you time overall.

df8oe, 2023-10-20:
Thank you for pointing me to that! Most programming examples on the web uses wiringPi so a rework is needed :) Hope libgpiod can handle interrupts.

6by9, 2023-10-20:
>     1. By forcing GPIOs in /sys/class/gpio to be numbered from 0 we've been going against the wishes of the kernel developers for years. As of the 6.5 kernel this results in a warning message in the kernel log.

Actually since 6.2, but it hadn't been noticed - https://github.com/raspberrypi/linux/commit/502df79b860563d79143be7a1453c2b3224cd836
 
> 2. Pi 5 has several different GPIO controllers. One of the drivers - the one for the GPIOs on the SoC - relied on the GPIOs being numbered from 0, but users would clearly prefer the GPIOs on the [...]

See https://waldorf.waveform.org.uk/2021/the-pins-they-are-a-changin.html for a summary by one of the Ubuntu devs (and author of gpiozero).

6by9, 2023-10-20:
WiringPi issue raised - https://github.com/WiringPi/WiringPi/issues/186

6by9, 2023-11-20:
Closing as WiringPi needs to change to match.

closed 2023-11-20 by 6by9

seamusdemora, 2023-11-21:
@6by9 
> > ```
> > 1. By forcing GPIOs in /sys/class/gpio to be numbered from 0 we've been going against the wishes of the kernel developers for years. As of the 6.5 kernel this results in a warning message in the kernel log.
> > ```
> 
> See https://waldorf.waveform.org.uk/2021/the-pins-they-are-a-changin.html for a summary by one of the Ubuntu devs (and author of gpiozero).

Very good article you referenced - thanks for that :)  

Beyond that: I guess that some of this pushback from `kernel developers` is political - rather than practical...  but that's just my opinion. I'll be sad to see WiringPi go, but it's not *life-changing* for me; if it were I guess I'd just hunker down with `buster` & forget updating. And finally, I don't think anyone here faults you for the decision.

df8oe, 2023-11-22:
Still looking for an example of using interrupts like this:

void myInterrupt (void)
{
  do_something_useful();
}

void main()
{
.....
  if (wiringPiSetup () < 0)
  {
    fprintf (stderr, "Unable to setup wiringPi: %s\n", strerror (errno)) ;
    return 1 ;
  }

  if (wiringPiISR (BUTTON_PIN, INT_EDGE_FALLING, &myInterrupt) < 0)
  {
    fprintf (stderr, "Unable to setup ISR: %s\n", strerror (errno)) ;
    return 1 ;
  }
.....
}

Any links to such an example?

pelwell, 2023-11-22:
If you look at the implementation of wiringPiISR (https://github.com/WiringPi/WiringPi/blob/91284c42759d3268b09c185ba1d176d54ff59eea/wiringPi/wiringPi.c#L1993) it creates a blocking thread that waits for events, calling your callback when one is received. The relevant libgpiod function is `gpiod_line_event_wait`, for which there is an example here: https://github.com/starnight/libgpiod-example/blob/master/libgpiod-event/main.c

6by9, 2023-11-22:
Both gadgetoid and I provided links for you on the wiringPi issue.
https://github.com/WiringPi/WiringPi/issues/186#issuecomment-1785595104 lgpio [gGpioSetSamplesFunc](https://abyz.me.uk/lg/lgpio.html#lgGpioSetSamplesFunc) although you probably want lgGpioSetAlertsFunc for specific GPIOs.
https://github.com/WiringPi/WiringPi/issues/186#issuecomment-1785665392 libgpiod [example code](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tree/examples/watch_multiple_line_values.c)

Yes it's more than 10 lines, but it's not exactly War and Peace.

df8oe, 2023-11-22:
The magic word I missed was "event" (instead of interrupt). I will dig into it and I think I can get rid of wiringpi now....

Many thanks!

6by9, 2023-11-22:
And I do note that gadgetoid and the others who were vaguely maintaining WiringPi have called it quits and archived the repo.
RIP WiringPi.

Gadgetoid, 2024-02-16:
> RIP WiringPi.

It's a shame, but for posterity we were never really maintaining it, had no intention to maintain it, and nobody was forthcoming with the inclination to maintain it. Formally declaring this and archiving the repositories was just a means to avoid having to repeat this ad nauseam to everyone who it breaks for.

The fork lives under an org for a reason, though, if anyone wants to revive it - be my guest! It's all yours 😆