raspberrypi/linux #5670: Driver drivetemp missing

spyesx, 2023-10-22:
### Describe the bug

The driver `hddtemp` has been replaced by `drivetemp`. Unfortunately, the kernel `6.1.0-13-arm64` is missing `drivetemp`.

Could anyone enable it please?

### Steps to reproduce the behaviour

In `6.1.0-rpi4-rpi-v8` we can see `drivetemp` is missing.

```bash
zgrep -i temp /boot/config-6.1.0-rpi4-rpi-v8
[...]
# CONFIG_SENSORS_DRIVETEMP is not set
[...]
```

In a regular `6.1.0-13` we can see `drivetemp` is compiled.

```bash
zgrep -i temp /boot/config-6.1.0-13-arm64
[...]
CONFIG_SENSORS_DRIVETEMP=m
[...]
```

### Device (s)

Raspberry Pi 3 Mod. B+, Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2

Kernel: `6.1.0-rpi4-rpi-v8`

### Logs

none

### Additional context

I found a way to downgrade to `6.1.0-13-arm64` from Debian but that requires to remove `raspi.list` and install `raspi-firmware` from Debian's repositories because the one delivered with Raspberry Pi OS is different (with hard coded valued).

spyesx, 2023-11-17:
I understand there is no activity on this issue for a month. Is it because my feedback is clear enough and will be part of a near upgrade or because it goes over heads?

pelwell, 2023-11-17:
> because it goes over heads?

Let's assume i'ts this, and nothing to do with activity levels after recent product launches, and you tell us exactly what needs to be done to fix it.

popcornmix, 2023-11-17:
I think the request is for:
[CONFIG_SENSORS_DRIVETEMP=m](https://cateee.net/lkddb/web-lkddb/SENSORS_DRIVETEMP.html)
to be added to kernel config.

spyesx, 2023-11-17:
@popcornmix yes, `hddtemp` was removed but it's replacement, `drivetemp`, is missing in `6.1.0-rpi4-rpi-v8`.

6by9, 2023-11-17:
> @popcornmix yes, `hddtemp` was removed but it's replacement, `drivetemp`, is missing in `6.1.0-rpi4-rpi-v8`.

I had a quick look for an hddtemp kernel driver, but couldn't find it.
The only reference to that string in 5.10 or 5.15 was indirectly at drivers/platform/x86/hp-wmi.c. Being x86, that would never have been built.

There was hddtemp as a Debian package under Buster and Bullseye - https://packages.debian.org/bullseye/hddtemp

pelwell, 2023-11-17:
I think @spyesx must be confusing the Debian package (dropped because it was unmaintained) with the driver, but never mind - there's a PR with the config changes here: #5717

spyesx, 2023-11-18:
Made a huge shortcut indeed from a misunderstanding indeed. Thx for the feedback and for the config changes.

B3rn475, 2024-10-26:
Out of curiosity, is there any update?

popcornmix, 2024-10-27:
The config option requested is enabled, so the issue should be closed
See, for example: https://github.com/raspberrypi/linux/blob/86099deff5abf5f63643eecaedb4c11ae77474ce/arch/arm64/configs/bcm2711_defconfig#L787

closed 2024-10-27 by popcornmix

B3rn475, 2024-10-27:
Thank you for confirming.

I thought a problem I'm having was related to this bug, but, in reality, it looks like it is another bug specific to my disks, that make them incompatible.