raspberrypi/linux #5671: Kenel build fail 6.1.y

sadhanpawar, 2023-10-23:
### Describe the bug

I'm trying to build kernel for raspiberrypi-4B with latest commit 5a52cae54a05499a8487f392cf5dfc3d8a837e6f and I'm facing build issue as following:
../arch/arm64/include/asm/atomic_ll_sc.h:318:38: error: ‘__uint128_t’ undeclared (first use in this function); did you mean ‘__int128__’?
  318 |  : "=&r" (tmp), "=&r" (ret), "+Q" (*(__uint128_t *)ptr) 

### Steps to reproduce the behaviour

git clone --depth=1 https://github.com/raspberrypi/linux
KERNEL=kernel8
IMAGE=Image.gz
make O=./build/ bcm2711_defconfig
make O=./build/ -j $IMAGE modules dtbs

### Device (s)

Raspberry Pi 4 Mod. B

### System

Raspberry Pi reference 2022-09-22
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 8a42abcd1dbd8c9c1fdfca4e0c3778255b2f9cc4, stage4

Oct 17 2023 15:39:16 
Copyright (c) 2012 Broadcom
version 30f0c5e4d076da3ab4f341d88e7d505760b93ad7 (clean) (release) (start)

Linux sadhanpi 6.1.58-v8+ #1692 SMP PREEMPT Thu Oct 19 17:06:12 BST 2023 aarch64 GNU/Linux

pelwell, 2023-10-23:
If you are trying to build a 64-bit kernel on a 32-bit OS (`file $(which ls)` will tell you) then you need to treat is as a cross-compile; see https://www.raspberrypi.com/documentation/computers/linux_kernel.html#cross-compiling-the-kernel for instructions.

pelwell, 2023-10-23:
I always include the `ARCH=arm` declaration - does that help?
```
KERNEL=kernel7l
make ARCH=arm bcm2711_defconfig
make ARCH=arm -j4 zImage modules dtbs
```
(The `KERNEL=kernel7l` line is only necessary if you use it later on during the installation, e.g.
`sudo cp arch/arm/boot/zImage /boot/firmware/$KERNEL.img`)

sadhanpawar, 2023-10-23:
it works!

closed 2023-10-23 by sadhanpawar