raspberrypi/linux #5676: Pi 5 kernel oops in v3d_mmu_insert_ptes

trejan, 2023-10-23:
### Describe the bug

Godot v4 editor with Vulkan renderer causes a kernel oops. Trying to move the broken Godot window or leaving the broken window for a few minutes causes the GUI to lock up but it remains accessible via SSH. Only way to recover is to reboot.

### Steps to reproduce the behaviour

[Godot v4 needs to be compiled](https://docs.godotengine.org/en/stable/contributing/development/compiling/compiling_for_linuxbsd.html). I can supply a self contained binary which doesn't need any other libs but it is very large.

Exact options used to compile were `scons platform=linuxbsd target=editor dev_build=yes`

Run with 'godot/bin/godot.linuxbsd.editor.dev.arm64 --rendering-driver vulkan' and download the "Godot 3D Flight Control Tutorial" from the Asset Library Projects tab. It will automatically open the project after downloading and hangs during the project loading process.

### Device (s)

Other

### System

```
System Information
------------------

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

Linux raspberrypi 6.1.0-rpi4-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux
Revision        : d04170
Serial          : e7aaa122ead0069e
Model           : Raspberry Pi 5 Model B Rev 1.0
Throttled flag  : throttled=0x0

Videocore information
---------------------

2023/10/18 18:30:17 
Copyright (c) 2012 Broadcom
version c2da2ae7 (release) (embedded)

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20231017
raspberrypi-sys-mods:
  Installed: 20231003
openbox:
  Installed: 3.6.1-10+rpt1
lxpanel:
  Installed: 0.10.1-2+rpt19
pcmanfm:
  Installed: 1.3.2-1+rpt16
rpd-plym-splash:
  Installed: 0.36

Display Information
-------------------

Running (F)KMS and X

Screen 0: minimum 0 x 0, current 2560 x 1440, maximum 32767 x 32767
default connected primary 2560x1440+0+0 (0x3f) normal (normal) 677mm x 381mm
        Identifier: 0x3e
        Timestamp:  55835109
        Subpixel:   unknown
        Gamma:      1.0:1.0:1.0
[... 518 lines cut ...]
[  200.560842] x26: ffffc0000afabd38 x25: ffff80016f471200 x24: 0000000000000042
[  200.568002] x23: ffff80016f471200 x22: ffff800100dc0000 x21: 0000000000100004
[  200.575163] x20: ffff8001dc0f3800 x19: 0000000000ffffff x18: 0000000000000000
[  200.582324] x17: 0000000000000000 x16: ffffd0000c64ab50 x15: 0000000000000100
[  200.589485] x14: ffffd0000d18c008 x13: 0000000000000000 x12: 0000000000000000
[  200.596646] x11: 0000000000000001 x10: 0000000000000068 x9 : ffffcfff9e1f8260
[  200.603807] x8 : ffff800100dc0700 x7 : 0000000000003fff x6 : ffff80019854c920
[  200.610968] x5 : 0000000000100000 x4 : ffffc00009738000 x3 : 0000000000100000
[  200.618129] x2 : 0000000030034bbf x1 : 0000000000100001 x0 : 0000000030034bc0
[  200.625290] Call trace:
[  200.627736]  v3d_mmu_insert_ptes+0xc0/0x154 [v3d]
[  200.632458]  v3d_bo_create_finish+0xd0/0xf0 [v3d]
[  200.637179]  v3d_create_bo_ioctl+0x50/0x150 [v3d]
[  200.641900]  drm_ioctl_kernel+0xd0/0x180 [drm]
[  200.646398]  drm_ioctl+0x214/0x430 [drm]
[  200.650363]  __arm64_sys_ioctl+0xb4/0xfc
[  200.654298]  invoke_syscall+0x50/0x120
[  200.658057]  el0_svc_common.constprop.0+0x68/0x124
[  200.662861]  do_el0_svc+0x34/0xd0
[  200.666181]  el0_svc+0x30/0x9c
[  200.669242]  el0t_64_sync_handler+0xf4/0x120
[  200.673523]  el0t_64_sync+0x18c/0x190
[  200.677193] Code: 2a0103e3 11000421 4b050020 0b020000 (b8237880) 
[  200.683306] ---[ end trace 0000000000000000 ]---
```

trejan, 2023-10-23:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.1.0-rpi4-rpi-2712 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05)
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000018000000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003f7fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 14 pages/cpu s177448 r8192 d43736 u229376
[    0.000000] pcpu-alloc: s177448 r8192 d43736 u229376 alloc=14*16384
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 521728
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:7C:DA:51 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=b08c7cc4-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=GB
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 9, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 8, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000fbffc000-0x00000000ffffc000] (64MB)
[    0.000000] Memory: 7893936K/8380416K available (12416K kernel code, 2184K rwdata, 4208K rodata, 4416K init, 1213K bss, 158800K reserved, 327680K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 40048 entries in 40 pages
[    0.000000] ftrace: allocated 40 pages with 2 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000153] Console: colour dummy device 80x25
[    0.000170] printk: console [tty1] enabled
[    0.000189] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000195] pid_max: default: 32768 minimum: 301
[    0.000219] LSM: Security Framework initializing
[    0.000290] Mount-cache hash table entries: 16384 (order: 3, 131072 bytes, linear)
[    0.000321] Mountpoint-cache hash table entries: 16384 (order: 3, 131072 bytes, linear)
[... 1119 lines cut ...]
[  200.560842] x26: ffffc0000afabd38 x25: ffff80016f471200 x24: 0000000000000042
[  200.568002] x23: ffff80016f471200 x22: ffff800100dc0000 x21: 0000000000100004
[  200.575163] x20: ffff8001dc0f3800 x19: 0000000000ffffff x18: 0000000000000000
[  200.582324] x17: 0000000000000000 x16: ffffd0000c64ab50 x15: 0000000000000100
[... cut ...]
[  200.641900]  drm_ioctl_kernel+0xd0/0x180 [drm]
[  200.646398]  drm_ioctl+0x214/0x430 [drm]
[  200.650363]  __arm64_sys_ioctl+0xb4/0xfc
[  200.654298]  invoke_syscall+0x50/0x120
[  200.658057]  el0_svc_common.constprop.0+0x68/0x124
[  200.662861]  do_el0_svc+0x34/0xd0
[  200.666181]  el0_svc+0x30/0x9c
[  200.669242]  el0t_64_sync_handler+0xf4/0x120
[  200.673523]  el0t_64_sync+0x18c/0x190
[  200.677193] Code: 2a0103e3 11000421 4b050020 0b020000 (b8237880) 
[  200.683306] ---[ end trace 0000000000000000 ]---
```

popcornmix, 2023-10-24:
The 3d driver isn't something we know a lot about here, so this may be better reported to mesa devs (although pi5 support is brand new, and I don't think merged yet).

But, a couple of questions come to mind. Does similar code work on Pi4?

As this code is related to kernel page size, does booting with 4K (rather than 16K) make a difference?
Use `kernel=kernel8.img` in config.txt to test this.

trejan, 2023-10-24:
> The 3d driver isn't something we know a lot about here, so this may be better reported to mesa devs (although pi5 support is brand new, and I don't think merged yet).

Okay. 

> But, a couple of questions come to mind. Does similar code work on Pi4?

I'll test it on a Pi 4 but I need to upgrade it to Bookworm first.

> As this code is related to kernel page size, does booting with 4K (rather than 16K) make a difference? Use `kernel=kernel8.img` in config.txt to test this.

Yes. There is a difference with 4k pages. It still doesn't load but it doesn't cause an oops or lock up any more. The Mesa Vulkan driver now repeatedly complains about out of memory.

> Failed to allocate device memory for BO
> ERROR: GENERAL - Message Id Number: 0 | Message Id Name: ../src/broadcom/vulkan/v3dv_device.c:2440
>         VK_ERROR_OUT_OF_DEVICE_MEMORY
>         Objects - 1
>                 Object[0] - VK_OBJECT_TYPE_DEVICE, Handle 367419007168
>    at: _debug_messenger_callback (drivers/vulkan/vulkan_context.cpp:267)

Godot does load the project if forced to use its OpenGL 3 renderer.

popcornmix, 2023-10-24:
> Yes. There is a difference with 4k pages. It still doesn't load but it doesn't cause an oops or lock up any more. The Mesa Vulkan driver now repeatedly complains about out of memory.

Running out of memory may be the correct behaviour (depending on how much memory is available and how much is requested). But, it could the result of another bug (some sort of leak).

I suspect there are ways of getting debug info for allocations (I think Mesa typically uses env variables for increasing debugging output), but I don't know much about this.

If this works on Pi4 (assuming it has the same RAM), then it's more likely it's a bug (it's not certain, as one platform may have a more optimised path available, due to hw differences, that may require more or less memory to use).

closed 2025-04-26 by trejan