raspberrypi/linux #5677: MCP23017 interrupt broke on 6.6.y overlays

hftsai256, 2023-10-24:
### Describe the bug

GPIO interrupt is not configured correctly when setting up mcp23017 with devicetree overlay in `config.txt` with kernel branch `rpi-6.6.y`.
```
dtoverlay=mcp23017,addr=0x23,gpiopin=23
```

If I roll back to the latest Raspberry OS Lite image (`2023-10-10-raspios-bookworm-arm64-lite`), the interrupt will then be configured as expected and can how up under `/proc/interrupts`:
```
 42:     418457          0          0          0  pinctrl-bcm2835  23 Level     1-0023
```

### Steps to reproduce the behaviour

1. Start with the latest Raspberry OS Lite release (2023-10-10).
2. Cross-compile the kernel from `rpi-6.6.y` branch, commit `f275f8f`: Typo in overlays README
```
KERNEL=kernel8
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- bcm2711_defconfig
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- menuconfig
make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- -j8 Image modules dtbs
```
3. Install kernel image to SD card/mounted eMMC:
```
cp arch/arm64/boot/Image $BOOTFS/$KERNEL.img
```
4. Install kernel modules:
```
sudo env PATH=$PATH make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu- INSTALL_MOD_PATH=$ROOTFS modules_install
```
5. Install Broadcom devicetree blobs.
```
cp arch/arm64/boot/dts/broadcom/*.dtb $BOOTFS/
```
6. Install overlays (breaking step)
```
cp arch/arm64/boot/dts/overlays/*.dtb* $BOOTFS/overlays/
```

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
~$ cat /etc/rpi-issue
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage2

~$ vcgencmd version
Aug 10 2023 15:33:38
Copyright (c) 2012 Broadcom
version 03dc77429335caee083e22ddc8eec09c07f12a7a (clean) (release) (start)

~$ uname -a
Linux pidev 6.6.0-rc6-v8+ #5 SMP PREEMPT Mon Oct 23 16:14:25 EDT 2023 aarch64 GNU/Linux

### Logs

`dmesg` at step 5 (working): http://0x0.st/HJDN.txt
`dmesg` at step 6 (broken): http://0x0.st/HJDK.txt

### Additional context

I dumped the devicetree base between step 5 and step 6. A diff quickly unveils the culprit:
```
--- step6_broken.dts
+++ step5_working.diff
@@ -248,12 +248,17 @@
 			phandle = <0x48>;
 
 			mcp@23 {
+				microchip,irq-mirror;
 				gpio-controller;
+				interrupts = <0x17 0x02>;
+				interrupt-parent = <0x07>;
 				compatible = "microchip,mcp23017";
+				#interrupt-cells = <0x02>;
 				status = "okay";
 				reg = <0x23>;
-				phandle = <0xf1>;
+				phandle = <0xf0>;
 				#gpio-cells = <0x02>;
+				interrupt-controller;
 			};
 		};
 
@@ -1170,7 +1175,7 @@
 
 			mcp23017_pins@23 {
 				brcm,function = <0x00>;
-				phandle = <0xf2>;
+				phandle = <0xf1>;
 				brcm,pins = <0x17>;
 			};
``` 

Not only the interrupt-related properties are missing from step 6, but the `phandle` addresses are also not consistent across the whole tree.

hftsai256, 2023-10-24:
I reverted `mcp23017-overlay.dts` back to [commit `06e36e8`](https://github.com/raspberrypi/linux/blob/06e36e8349831509b02fe489417cd45fd4ddc9f0/arch/arm/boot/dts/overlays/mcp23017-overlay.dts) seems resolve the problem. Please review the breaking change on [`27f0e0e`](https://github.com/raspberrypi/linux/commit/27f0e0e195568c06f21ce380f0736bdd219baf3c).

pelwell, 2023-10-24:
This is caused by the ease of getting the fragment order wrong, something I'd like to detect and automatically warn about, and actually affects rpi-6.1.y as well. If you have time it would be helpful it you could install a trial kernel - wait about an hour until the build checks complete then run `sudo rpi-update pulls/5678`.

pelwell, 2023-10-24:
>  the phandle is still inconsistent.

Inconsistent or different? phandles are allocated by the dtc compiler and firmware, and depend on the entire content of the dtc/dtb file. What matters is that if in one branch node A uses phandle X and node B has the property "phandle = <X>;",  that in other branches node A uses phandle Y and node B has the property "phandle = <X>;", where X and Y may be different values.

hftsai256, 2023-10-24:
Thanks for the clarification. "Different" may be the proper term in this case. Are there other places in your mind to investigate?

pelwell, 2023-10-24:
Are there any errors or warnings in the kernel log? I don't have an MCP23017 to test.

hftsai256, 2023-10-24:
The problem was due to a short on one of the i2c addressing pins. The interrupt is now adequately configured with your new patch. Thanks.

pelwell, 2023-10-24:
Thank goodness for that - you had me worried.
The changes are now in 6.5 and 6.6.

closed 2023-11-17 by pelwell