raspberrypi/linux #5683: __cfg80211_connect_result : Unable to handle kernel NULL pointer dereference

Zahrun, 2023-10-26:
### Describe the bug

`Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP`
Networking is lost. System cannot reconnect unless hard reboot.

### Steps to reproduce the behaviour

I’m using a Raspberry Pi 3 B+ with a Archer T2U+ USB Wi-Fi dongle as a Wi-Fi router.
The Wi-Fi dongle connects to my local university Wi-Fi.
I use the Raspberry’s internal Wi-Fi to create an access point.
The kernel error happens about once every 24 hours, the RPi becomes unresponsive, and I need to hard reboot it.

### Device (s)

Raspberry Pi 3 Mod. B+

### System

Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7c750947a959fb626a70c09fd17c65815df192ac, stage4

Sep 21 2023 18:44:30 
Copyright (c) 2012 Broadcom
version 3094eda5aadf2c719edd2c68d1900884abc3dbca (clean) (release) (start)

Linux love 6.1.57-v8+ #1688 SMP PREEMPT Thu Oct 12 15:14:26 BST 2023 aarch64 GNU/Linux

### Logs

Oct 20 15:41:47 love NetworkManager[589016]: <info>  [1697796707.0505] device (wlan1): supplicant interface state: associating -> disconnected
Oct 20 15:41:47 love NetworkManager[589016]: <info>  [1697796707.0507] device (p2p-dev-wlan1): supplicant management interface state: associating -> disconnected
Oct 20 15:41:57 love NetworkManager[589016]: <info>  [1697796717.0554] device (wlan1): supplicant interface state: disconnected -> scanning
Oct 20 15:41:57 love NetworkManager[589016]: <info>  [1697796717.0556] device (p2p-dev-wlan1): supplicant management interface state: disconnected -> scanning
Oct 20 15:42:15 love wpa_supplicant[415]: wlan1: CTRL-EVENT-SSID-REENABLED id=0 ssid="***"
Oct 20 15:42:15 love wpa_supplicant[415]: wlan1: Trying to associate with *** (SSID='***' freq=2412 MHz)
Oct 20 15:42:15 love wpa_supplicant[415]: nl80211: kernel reports: Authentication algorithm number required
Oct 20 15:42:15 love NetworkManager[589016]: <info>  [1697796735.6060] device (wlan1): supplicant interface state: scanning -> associating
Oct 20 15:42:15 love NetworkManager[589016]: <info>  [1697796735.6062] device (p2p-dev-wlan1): supplicant management interface state: scanning -> associating
Oct 20 15:42:19 love kernel: [269374.508611] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
Oct 20 15:42:19 love kernel: [269374.508649] Mem abort info:
Oct 20 15:42:19 love kernel: [269374.508659]   ESR = 0x0000000096000005
Oct 20 15:42:19 love kernel: [269374.508669]   EC = 0x25: DABT (current EL), IL = 32 bits
Oct 20 15:42:19 love kernel: [269374.508681]   SET = 0, FnV = 0
Oct 20 15:42:19 love kernel: [269374.508691]   EA = 0, S1PTW = 0
Oct 20 15:42:19 love kernel: [269374.508701]   FSC = 0x05: level 1 translation fault
Oct 20 15:42:19 love kernel: [269374.508712] Data abort info:
Oct 20 15:42:19 love kernel: [269374.508721]   ISV = 0, ISS = 0x00000005
Oct 20 15:42:19 love kernel: [269374.508731]   CM = 0, WnR = 0
Oct 20 15:42:19 love kernel: [269374.508741] user pgtable: 4k pages, 39-bit VAs, pgdp=00000000047ca000
Oct 20 15:42:19 love kernel: [269374.508753] [0000000000000000] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
Oct 20 15:42:19 love kernel: [269374.508787] Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
Oct 20 15:42:19 love kernel: [269374.508797] Modules linked in: tun rfcomm nft_chain_nat xt_MASQUERADE xt_state xt_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp nft_compat nf_tables nfnetlink nf_nat_h323 nf_conntrack_h323 nf_nat_pptp nf_conntrack_pptp nf_nat_tftp nf_conntrack_tftp nf_nat_sip nf_conntrack_sip nf_nat_irc nf_conntrack_irc nf_nat_ftp nf_conntrack_ftp iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep 88XXau(O) vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper drm_kms_helper brcmfmac sg brcmutil snd_soc_core hci_uart btbcm cfg80211 snd_compress bluetooth snd_pcm_dmaengine syscopyarea sysfillrect sysimgblt fb_sys_fops raspberrypi_hwmon bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) videobuf2_vmalloc bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig ecdh_generic videobuf2_memops snd_bcm2835(C) videobuf2_v4l2 ecc videobuf2_common snd_pcm i2c_bcm2835 rfkill videodev libaes snd_timer mc snd vc_sm_cma(C)
Oct 20 15:42:19 love kernel: [269374.509056]  raspberrypi_gpiomem uio_pdrv_genirq uio drm i2c_dev fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
Oct 20 15:42:19 love kernel: [269374.509100] CPU: 1 PID: 1470221 Comm: kworker/u8:0 Tainted: G        WC O       6.1.57-v8+ #1688
Oct 20 15:42:19 love kernel: [269374.509110] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.4 (DT)
Oct 20 15:42:19 love kernel: [269374.509118] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
Oct 20 15:42:19 love kernel: [269374.509365] pstate: 40000005 (nZcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Oct 20 15:42:19 love kernel: [269374.509375] pc : __cfg80211_connect_result+0x22c/0x690 [cfg80211]
Oct 20 15:42:19 love kernel: [269374.509519] lr : __cfg80211_connect_result+0x1c8/0x690 [cfg80211]
Oct 20 15:42:19 love kernel: [269374.509662] sp : ffffffc00fcebcd0
Oct 20 15:42:19 love kernel: [269374.509668] x29: ffffffc00fcebcd0 x28: 0000000000000000 x27: 0000000000000000
Oct 20 15:42:19 love kernel: [269374.509682] x26: 0000000000000000 x25: dead000000000100 x24: 0000000000000001
Oct 20 15:42:19 love kernel: [269374.509696] x23: 0000000000000000 x22: ffffff80045ea000 x21: ffffffd2325ab818
Oct 20 15:42:19 love kernel: [269374.509710] x20: ffffff8009bdfc18 x19: ffffff8009bdfc18 x18: 0000000000000000
Oct 20 15:42:19 love kernel: [269374.509724] x17: 0000000000000000 x16: ffffffd2321a8940 x15: 0000000000000000
Oct 20 15:42:19 love kernel: [269374.509738] x14: 0000000000000040 x13: 0000000000000000 x12: 0000000000000001
Oct 20 15:42:19 love kernel: [269374.509752] x11: 0000000000000040 x10: ffffffd232aa6150 x9 : 0000000000000000
Oct 20 15:42:19 love kernel: [269374.509766] x8 : ffffff80045ea710 x7 : 0000000000000000 x6 : 000000000000003f
Oct 20 15:42:19 love kernel: [269374.509780] x5 : 0000000000000040 x4 : 0000000000000000 x3 : 0000000000000010
Oct 20 15:42:19 love kernel: [269374.509793] x2 : 0000000000000000 x1 : 0000000000000001 x0 : 0000000000000001
Oct 20 15:42:19 love kernel: [269374.509807] Call trace:
Oct 20 15:42:19 love kernel: [269374.509813]  __cfg80211_connect_result+0x22c/0x690 [cfg80211]
Oct 20 15:42:19 love kernel: [269374.509956]  cfg80211_process_wdev_events+0x124/0x188 [cfg80211]
Oct 20 15:42:19 love kernel: [269374.510100]  cfg80211_process_rdev_events+0x38/0x58 [cfg80211]
Oct 20 15:42:19 love kernel: [269374.510244]  cfg80211_event_work+0x30/0x48 [cfg80211]
Oct 20 15:42:19 love kernel: [269374.510387]  process_one_work+0x208/0x480
Oct 20 15:42:19 love kernel: [269374.510404]  worker_thread+0x50/0x428
Oct 20 15:42:19 love kernel: [269374.510414]  kthread+0xfc/0x110
Oct 20 15:42:19 love kernel: [269374.510425]  ret_from_fork+0x10/0x20
Oct 20 15:42:19 love kernel: [269374.510439] Code: b4fffec0 394332c0 32000000 390332c0 (b94002e0) 
Oct 20 15:42:19 love kernel: [269374.510448] ---[ end trace 0000000000000000 ]---
Oct 20 15:42:58 love NetworkManager[589016]: <warn>  [1697796778.1861] device (wlan1): Activation: (wifi) association took too long
Oct 20 15:42:58 love NetworkManager[589016]: <info>  [1697796778.1863] device (wlan1): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Oct 20 15:42:58 love NetworkManager[589016]: <warn>  [1697796778.2032] device (wlan1): Activation: (wifi) asking for new secrets
Oct 20 15:42:58 love NetworkManager[589016]: <info>  [1697796778.2485] device (wlan1): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Oct 20 15:42:58 love NetworkManager[589016]: <info>  [1697796778.2566] device (wlan1): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')

### Additional context

See a similar issue https://github.com/raspberrypi/linux/issues/5491
That one was fixed, so I updated the kernel to one that has the fix included but that didn’t work.

kralo, 2023-10-26:
My few tips for debugging this: 

first, you should reduce your setup to the absolute minimum to reproduce. Your setup looks quite complicated.
You seem to use the Archer T2U with an out-of-tree module, 88XXau. This is probably not from raspberrypi kernel nor linux-mainline, so there will probably not be much motivation in this repo.

This leads to the question, does it ever happen without this usb dongle?
It suggests it happens when the dongle, (wlan1)  tries to reconnect after CTRL-EVENT-SSID-REENABLED. When you look further up your logs, why is it even (temp) disabled?

Then you could turn to increasing wpa_supplicant verbosity, compile kernel with debug info (line numbers), ...

Zahrun, 2023-11-07:
Thank you for the helpful answer.
I am using the driver from https://github.com/aircrack-ng/rtl8812au. I understand that the issue should probably be raised there.

I am pretty sure that the issue would not happen without the USB dongle, but I might confirm this later.

When looking further up in the logs, the temp disabled event happens due to authentification failure. I am getting a lot of these and I think sometimes it is due to my local university Wi-Fi just becoming impossible to connect to. Maybe other times it is something related to my setup. It so happenned that after a reboot, the pi could connect again. But sometimes also no device can connect, whatever be the OS.

<details>
<summary>Fold/Unfold</summary>
<br>

```
Oct 30 14:53:57 love NetworkManager[355]: <warn>  [1698657837.4710] device (wlan2): Activation: (wifi) association took too long
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.4711] device (wlan2): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Oct 30 14:53:57 love NetworkManager[355]: <warn>  [1698657837.4844] device (wlan2): Activation: (wifi) asking for new secrets
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5304] device (wlan2): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5323] device (wlan2): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5341] device (wlan2): Activation: (wifi) connection '***' has security, and secrets exist.  No new secrets needed.
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5342] Config: added 'ssid' value '***'
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5343] Config: added 'scan_ssid' value '1'
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5343] Config: added 'bssid' value '**:**:**:**:**:**'
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5344] Config: added 'key_mgmt' value 'WPA-EAP WPA-EAP-SHA256 FT-EAP FT-EAP-SHA384'
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5345] Config: added 'password' value '<hidden>'
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5345] Config: added 'eap' value 'PEAP'
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5346] Config: added 'fragment_size' value '1266'
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5347] Config: added 'phase2' value 'auth=MSCHAPV2'
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5348] Config: added 'identity' value '**@**.**'
Oct 30 14:53:57 love NetworkManager[355]: <info>  [1698657837.5348] Config: added 'proactive_key_caching' value '1'
Oct 30 14:54:01 love wpa_supplicant[392]: wlan2: Trying to associate with **:**:**:**:** (SSID='***' freq=5240 MHz)
Oct 30 14:54:01 love wpa_supplicant[392]: nl80211: kernel reports: Authentication algorithm number required
Oct 30 14:54:01 love NetworkManager[355]: <info>  [1698657841.0378] device (wlan2): supplicant interface state: scanning -> associating
Oct 30 14:54:01 love NetworkManager[355]: <info>  [1698657841.0402] device (p2p-dev-wlan2): supplicant management interface state: scanning -> associating
Oct 30 14:54:01 love wpa_supplicant[392]: wlan2: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5240 ht_enabled=1 ch_offset=-1 ch_width=80 MHz cf1=5210 cf2=0
Oct 30 14:54:01 love wpa_supplicant[392]: wlan2: CTRL-EVENT-ASSOC-REJECT status_code=40
Oct 30 14:54:01 love wpa_supplicant[392]: wlan2: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="***" auth_failures=1 duration=10 reason=CONN_FAILED
Oct 30 14:54:01 love wpa_supplicant[392]: wlan2: CTRL-EVENT-ASSOC-REJECT status_code=1
Oct 30 14:54:01 love NetworkManager[355]: <info>  [1698657841.1082] device (wlan2): supplicant interface state: associating -> disconnected
Oct 30 14:54:01 love NetworkManager[355]: <info>  [1698657841.1093] device (p2p-dev-wlan2): supplicant management interface state: associating -> disconnected
Oct 30 14:54:01 love systemd-resolved[294]: Using degraded feature set TCP instead of UDP for DNS server **.**.**.**.
Oct 30 14:54:11 love NetworkManager[355]: <info>  [1698657851.1203] device (wlan2): supplicant interface state: disconnected -> scanning
Oct 30 14:54:11 love NetworkManager[355]: <info>  [1698657851.1205] device (p2p-dev-wlan2): supplicant management interface state: disconnected -> scanning
Oct 30 14:54:12 love wpa_supplicant[392]: wlan2: CTRL-EVENT-SSID-REENABLED id=0 ssid="***"
Oct 30 14:54:12 love wpa_supplicant[392]: wlan2: Trying to associate with **:**:**:**:** (SSID='***' freq=5240 MHz)
Oct 30 14:54:12 love wpa_supplicant[392]: nl80211: kernel reports: Authentication algorithm number required
Oct 30 14:54:12 love NetworkManager[355]: <info>  [1698657852.1366] device (wlan2): supplicant interface state: scanning -> associating
Oct 30 14:54:12 love NetworkManager[355]: <info>  [1698657852.1368] device (p2p-dev-wlan2): supplicant management interface state: scanning -> associating
Oct 30 14:54:16 love wpa_supplicant[392]: wlan2: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5240 ht_enabled=1 ch_offset=-1 ch_width=80 MHz cf1=5210 cf2=0
Oct 30 14:54:16 love wpa_supplicant[392]: wlan2: CTRL-EVENT-ASSOC-REJECT status_code=40
Oct 30 14:54:16 love wpa_supplicant[392]: wlan2: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="***" auth_failures=2 duration=33 reason=CONN_FAILED
[... cut ...]
My take away from now is to avoid devices which don’t have support in the linux kernel. I found this interesting ressource about that: https://github.com/morrownr/USB-WiFi/tree/main. 
I didn’t get the issue in 5 days now. Maybe that is since I changed the USB extension cable to a much shorter one.
I will keep monitoring the situation and act accordingly. I may want to try the alternative driver at https://github.com/morrownr/8821au-20210708/tree/main.

I think this issue can be closed. Thank you for the support.

closed 2023-11-07 by Zahrun

khuongtrinh2, 2024-12-27:
Hi @Zahrun 
I have same issue as you. How can you fix it? 
Thank you so much

Zahrun, 2024-12-27:
The main issue in my case was the USB extension cable that I was using at the time, which caused the RPi to crash for some reason

khuongtrinh2, 2024-12-27:
Thank you, 
I enable MAC randomization but got kernel panic when changing from **randomized MAC** to **device MAC** 

USB Device: TP-LINK AC1200 T4U
Kernel: 6.1

Try with Kernel 5.10, it's work well so I think kernel 6.1 is missing some implement