raspberrypi/linux #5686: irs1125 can't get pwdn gpio

Coimbra1984, 2023-10-27:
### Describe the bug

Hi,

I installed a fresh Raspberry Pi Os (2023-10-10) (64 bit) without desktop with kernel 6.1. I'm unable to get out irs1125 driver to work. It fails on this line: https://github.com/raspberrypi/linux/blob/cb013b6602de32c647ed08faf899596664a18635/drivers/media/i2c/irs1125.c#L1111
However when I decompile the device tree, I can't see any error:
```
i2c0mux {
    pinctrl-names = "i2c0\0i2c_csi_dsi";
    #address-cells = <0x01>;
    pinctrl-0 = <0x33>;
    #size-cells = <0x00>;
    compatible = "i2c-mux-pinctrl";
    pinctrl-1 = <0x34>;
    status = "okay";
    i2c-parent = <0x32>;
    phandle = <0x47>;

    i2c@0 {
        #address-cells = <0x01>;
        #size-cells = <0x00>;
        reg = <0x00>;
        phandle = <0xd8>;
    };

    i2c@1 {
        #address-cells = <0x01>;
        #size-cells = <0x00>;
        status = "okay";
        reg = <0x01>;
        phandle = <0xd9>;

        irs1125@3d {
            pwdn-gpios = <0x0b 0x05 0x00>;
            clocks = <0xe6>;
            compatible = "infineon,irs1125";
            status = "okay";
            reg = <0x3d>;
            phandle = <0xf2>;

            port {

                endpoint {
                    data-lanes = <0x01 0x02>;
                    clock-noncontinuous;
                    clock-lanes = <0x00>;
                    remote-endpoint = <0xf0>;
                    phandle = <0xf1>;
                    link-frequencies = <0x00 0x11b3dc40>;
                };
            };
        };
    };
};

```

### Steps to reproduce the behaviour

- Install Raspberry Pi os 64 bit - without desktop
- enable irs1125 overlay
- call dmesg and see "could not get reset"

### Device (s)

Raspberry Pi 4 Mod. B

### System

pi@raspicam:~ $ uname -a
Linux raspicam 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux

pelwell, 2023-10-27:
Please post the non-standard content of your config.txt. Does `dmesg -l err,warn` show anything relevant?

6by9, 2023-10-27:
```dtparam=cam1_reg=off```

All the other drivers have been using regulators for quite a while. Regulators allow multiple users, whilst GPIO doesn't. CM4IO shares one GPIO between both camera ports.

Regulators (and clocks) behave very badly if you try and disable them dynamically, as they'll complete their remove before all clients have gone. When the client then tries to deregister as a user, "amusing" things happen. The camera regulators and clocks are therefore enabled by default.

Coimbra1984, 2023-10-27:
Okay, thanks. In that case I would change that in our driver as well. Is there any sample implementation I can follow?

6by9, 2023-10-27:
> Okay, thanks. In that case I would change that in our driver as well. Is there any sample implementation I can follow?

https://github.com/raspberrypi/linux/commit/189bda685567ad364f621b86803e92c6f76d06a8 did the same for ov5647

Looking at the irs1125 driver quickly, please note that the s_power hook is being dropped by upstream, preferring the use of pm_runtime instead. https://github.com/raspberrypi/linux/commit/089b7c70f0d80f5169cc87f65b065fc80668a0eb was that conversion for ov5647 although it's not the clearest to follow.
If you look at [imx290 in mainline](https://github.com/torvalds/linux/blob/master/drivers/media/i2c/imx290.c), that's probably one of the better examples of how to do it.

pelwell, 2023-11-13:
I think we've answered this one.

closed 2023-11-13 by pelwell