raspberrypi/linux #5687: error: variable ‘mip_msi_domain_info’ has initializer but incomplete type

piterbuchcic1990, 2023-10-30:
### Describe the bug

Hi 

I am trying to compile Linux kernel 6.1.58 based on config arch/arm64/configs/bcm2712_defconfig.

I am observing several errors:
drivers/irqchip/irq-bcm2712-mip.c:70:15: error: variable ‘mip_msi_domain_info’ has initializer but incomplete type
 static struct msi_domain_info mip_msi_domain_info = {
               ^~~~~~~~~~~~~~~
drivers/irqchip/irq-bcm2712-mip.c:71:3: error: ‘struct msi_domain_info’ has no member named ‘flags’
  .flags = (MSI_FLAG_USE_DEF_DOM_OPS | MSI_FLAG_USE_DEF_CHIP_OPS |
   ^~~~~
drivers/irqchip/irq-bcm2712-mip.c:71:12: error: ‘MSI_FLAG_USE_DEF_DOM_OPS’ undeclared here (not in a function); did you mean ‘SIMPLE_DEV_PM_OPS’?
  .flags = (MSI_FLAG_USE_DEF_DOM_OPS | MSI_FLAG_USE_DEF_CHIP_OPS |
            ^~~~~~~~~~~~~~~~~~~~~~~~
            SIMPLE_DEV_PM_OPS
  CC      io_uring/fs.o
drivers/irqchip/irq-bcm2712-mip.c:71:39: error: ‘MSI_FLAG_USE_DEF_CHIP_OPS’ undeclared here (not in a function)
  .flags = (MSI_FLAG_USE_DEF_DOM_OPS | MSI_FLAG_USE_DEF_CHIP_OPS |
                                       ^~~~~~~~~~~~~~~~~~~~~~~~~
  CC      arch/arm/kernel/reboot.o
  CC      security/keys/process_keys.o
drivers/irqchip/irq-bcm2712-mip.c:72:6: error: ‘MSI_FLAG_PCI_MSIX’ undeclared here (not in a function); did you mean ‘SS_FLAG_BITS’?
      MSI_FLAG_PCI_MSIX),
      ^~~~~~~~~~~~~~~~~
      SS_FLAG_BITS
drivers/irqchip/irq-bcm2712-mip.c:71:11: warning: excess elements in struct initializer
  .flags = (MSI_FLAG_USE_DEF_DOM_OPS | MSI_FLAG_USE_DEF_CHIP_OPS |
           ^
drivers/irqchip/irq-bcm2712-mip.c:71:11: note: (near initialization for ‘mip_msi_domain_info’)
drivers/irqchip/irq-bcm2712-mip.c:73:3: error: ‘struct msi_domain_info’ has no member named ‘chip’
  .chip = &mip_msi_irq_chip,
   ^~~~
drivers/irqchip/irq-bcm2712-mip.c:73:10: warning: excess elements in struct initializer
  .chip = &mip_msi_irq_chip,
          ^
drivers/irqchip/irq-bcm2712-mip.c:73:10: note: (near initialization for ‘mip_msi_domain_info’)
  CC      lib/crypto/des.o
  CC      init/do_mounts_initrd.o
drivers/irqchip/irq-bcm2712-mip.c: In function ‘mip_init_domains’:
drivers/irqchip/irq-bcm2712-mip.c:222:15: error: implicit declaration of function ‘pci_msi_create_irq_domain’; did you mean ‘pci_msi_get_device_domain’? [-Werror=implicit-function-declaration]
  msi_domain = pci_msi_create_irq_domain(of_node_to_fwnode(node),

I see that file drivers/irqchip/irq-bcm2712-mip.c has included 'include/linux/msi.h' file however somehow struct declaration from that header is not visible in compilation process...

### Steps to reproduce the behaviour

For compiling kernel I am using buildroot2020.02 with enabled glibc and gcc 8.3.0:

cd buildroot2020.02 ; make linux-reconfigure

### Device (s)

Other

### System

piotr@piotr-AORUS-15P-KD:~/recovery/buildroot-2020.02$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.5 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.5 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

pelwell, 2023-10-30:
Something you have changed has broken the compilation. That shouldn't happen - the configuration dependencies should prevent it - but we need to start by finding out what you've changed.

What do the following commands report?
```
$ make ARCH=arm64 savedefconfig
$ diff arch/arm64/configs/bcm2712_defconfig defconfig
```

piterbuchcic1990, 2023-10-30:
Here it is:

piotr@piotr-AORUS-15P-KD:~/recovery/buildroot-2020.02/output/build/linux-rpi-6.1.y$ diff arch/arm64/configs/bcm2712_defconfig defconfig
1c1
< CONFIG_LOCALVERSION="-v8_16k"
---
> CONFIG_LOCALVERSION="-rx500"
26d25
< CONFIG_CPUSETS=y
41,50c40
< # CONFIG_CAVIUM_ERRATUM_22375 is not set
< # CONFIG_CAVIUM_ERRATUM_23154 is not set
< # CONFIG_CAVIUM_ERRATUM_27456 is not set
< CONFIG_ARM64_16K_PAGES=y
< CONFIG_COMPAT=y
< CONFIG_ARMV8_DEPRECATED=y
< CONFIG_SWP_EMULATION=y
< CONFIG_CP15_BARRIER_EMULATION=y
< CONFIG_SETEND_EMULATION=y
< CONFIG_RANDOMIZE_BASE=y
---
> CONFIG_HZ_100=y
51a42
> # CONFIG_EFI is not set
63d53
< CONFIG_CPU_FREQ_GOV_SCHEDUTIL=y
66,67d55
< CONFIG_VIRTUALIZATION=y
< CONFIG_KVM=y
69,70c57
< CONFIG_ARCH_MMAP_RND_BITS=18
< CONFIG_ARCH_MMAP_RND_COMPAT_BITS=11
---
> CONFIG_COMPAT_32BIT_TIME=y
464d450
< # CONFIG_EFI_VARS_PSTORE is not set
768,769d753
< # CONFIG_POWER_RESET_BRCMSTB is not set
< CONFIG_POWER_RESET_GPIO=y
1463d1446
< CONFIG_BCM2712_IOMMU=y
1645,1655d1627
< CONFIG_CRYPTO_GHASH_ARM64_CE=m
< CONFIG_CRYPTO_SHA1_ARM64_CE=m
< CONFIG_CRYPTO_SHA2_ARM64_CE=m
< CONFIG_CRYPTO_SHA512_ARM64_CE=m
< CONFIG_CRYPTO_SHA3_ARM64=m
< CONFIG_CRYPTO_SM3_ARM64_CE=m
< CONFIG_CRYPTO_AES_ARM64=m
< CONFIG_CRYPTO_AES_ARM64_CE_BLK=m
< CONFIG_CRYPTO_AES_ARM64_BS=m
< CONFIG_CRYPTO_SM4_ARM64_CE=m
< CONFIG_CRYPTO_AES_ARM64_CE_CCM=m
1662a1635
> CONFIG_FRAME_WARN=1024

piterbuchcic1990, 2023-10-30:
Not sure how to test it, when I starting:
make ARCH=arm64 bcm2712_defconfig ; make 
or
cp arch/arm64/configs/bcm2712_defconfig; make

the 'make' command asking be about selecting/not selecting hundreds of options which are not selected in defconfig...

pelwell, 2023-10-30:
That's probably because you need to specify the ARCH for each make:
```
$ make ARCH=arm64 bcm2712_defconfig; make ARCH=arm64
```

pelwell, 2023-10-30:
As well as specifying what you want to build (see above) you'll probably want to increase the number of simultaneous jobs to speed up the build. On a Pi 4 that number should be at least 4, perhaps 6:
```
$ make ARCH=arm64 bcm2712_defconfig; make -j6 ARCH=arm64 Image.gz modules dtbs
```

piterbuchcic1990, 2023-10-30:
I need build it by buildroot crosscompilation process,  I am not trying do that on PI4 board,

with pure git repo (without buildroot) I have something like that:
piotr@piotr-AORUS-15P-KD:~/recovery/buildroot-2020.02/dl/linux/git$ make ARCH=arm64 bcm2712_defconfig; make ARCH=arm64 Image.gz modules dtbs
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
arch/arm64/Makefile:36: Detected assembler with broken .inst; disassembly will be unreliable
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/dtc/checks.o
  HOSTCC  scripts/dtc/util.o
  LEX     scripts/dtc/dtc-lexer.lex.c
  YACC    scripts/dtc/dtc-parser.tab.[ch]
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  HOSTLD  scripts/dtc/dtc
  HOSTCC  scripts/dtc/libfdt/fdt.o
  HOSTCC  scripts/dtc/libfdt/fdt_ro.o
  HOSTCC  scripts/dtc/libfdt/fdt_wip.o
  HOSTCC  scripts/dtc/libfdt/fdt_sw.o
  HOSTCC  scripts/dtc/libfdt/fdt_rw.o
  HOSTCC  scripts/dtc/libfdt/fdt_strerror.o
  HOSTCC  scripts/dtc/libfdt/fdt_empty_tree.o
  HOSTCC  scripts/dtc/libfdt/fdt_addresses.o
  HOSTCC  scripts/dtc/libfdt/fdt_overlay.o
  HOSTCC  scripts/dtc/fdtoverlay.o
  HOSTLD  scripts/dtc/fdtoverlay
  HOSTCC  scripts/genksyms/genksyms.o
  YACC    scripts/genksyms/parse.tab.[ch]
  HOSTCC  scripts/genksyms/parse.tab.o
  LEX     scripts/genksyms/lex.lex.c
  HOSTCC  scripts/genksyms/lex.lex.o
  HOSTLD  scripts/genksyms/genksyms
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/sorttable
  HOSTCC  scripts/asn1_compiler
  WRAP    arch/arm64/include/generated/uapi/asm/kvm_para.h
  WRAP    arch/arm64/include/generated/uapi/asm/errno.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctl.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctls.h
  WRAP    arch/arm64/include/generated/uapi/asm/ipcbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/msgbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/poll.h
  WRAP    arch/arm64/include/generated/uapi/asm/resource.h
  WRAP    arch/arm64/include/generated/uapi/asm/sembuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/shmbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/siginfo.h
  WRAP    arch/arm64/include/generated/uapi/asm/socket.h
  WRAP    arch/arm64/include/generated/uapi/asm/sockios.h
  WRAP    arch/arm64/include/generated/uapi/asm/stat.h
  WRAP    arch/arm64/include/generated/uapi/asm/swab.h
  WRAP    arch/arm64/include/generated/uapi/asm/termbits.h
  WRAP    arch/arm64/include/generated/uapi/asm/termios.h
  WRAP    arch/arm64/include/generated/uapi/asm/types.h
  WRAP    arch/arm64/include/generated/asm/early_ioremap.h
  WRAP    arch/arm64/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm64/include/generated/asm/qrwlock.h
  WRAP    arch/arm64/include/generated/asm/qspinlock.h
  WRAP    arch/arm64/include/generated/asm/parport.h
  WRAP    arch/arm64/include/generated/asm/user.h
  WRAP    arch/arm64/include/generated/asm/bugs.h
  WRAP    arch/arm64/include/generated/asm/delay.h
  WRAP    arch/arm64/include/generated/asm/div64.h
  WRAP    arch/arm64/include/generated/asm/dma-mapping.h
  WRAP    arch/arm64/include/generated/asm/dma.h
  WRAP    arch/arm64/include/generated/asm/emergency-restart.h
  WRAP    arch/arm64/include/generated/asm/hw_irq.h
  WRAP    arch/arm64/include/generated/asm/irq_regs.h
  WRAP    arch/arm64/include/generated/asm/kdebug.h
  WRAP    arch/arm64/include/generated/asm/kmap_size.h
  WRAP    arch/arm64/include/generated/asm/local.h
  WRAP    arch/arm64/include/generated/asm/local64.h
  WRAP    arch/arm64/include/generated/asm/mmiowb.h
  WRAP    arch/arm64/include/generated/asm/msi.h
  WRAP    arch/arm64/include/generated/asm/serial.h
  WRAP    arch/arm64/include/generated/asm/softirq_stack.h
  WRAP    arch/arm64/include/generated/asm/switch_to.h
  WRAP    arch/arm64/include/generated/asm/trace_clock.h
  WRAP    arch/arm64/include/generated/asm/unaligned.h
  WRAP    arch/arm64/include/generated/asm/vga.h
  UPD     include/config/kernel.release
  UPD     include/generated/uapi/linux/version.h
  UPD     include/generated/utsrelease.h
  UPD     include/generated/compile.h
  GEN     arch/arm64/include/generated/asm/cpucaps.h
  GEN     arch/arm64/include/generated/asm/sysreg-defs.h
warning: ld does not support --fix-cortex-a53-843419; kernel may be susceptible to erratum
warning: LSE atomics not supported by binutils
  CC      scripts/mod/empty.o
gcc: error: unrecognized command line option ‘-mlittle-endian’
make[1]: *** [scripts/Makefile.build:250: scripts/mod/empty.o] Error 1
make: *** [Makefile:1288: prepare0] Error 2
piotr@piotr-AORUS-15P-KD:~/recovery/buildroot-2020.02/dl/linux/git$ git status
On branch rpi-6.1.y
nothing to commit, working tree clean
piotr@piotr-AORUS-15P-KD:~/recovery/buildroot-2020.02/dl/linux/git$ gcc --version
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Is this build compilable on Ubuntu OS without using PI4 board?

pelwell, 2023-10-30:
I should be fine - I cross-compile the kernel daily. You'll need to `sudo apt install gcc-aarch64-linux-gnu`, and run `export CROSS_COMPILE=aarch64-linux-gnu` before your make commands.

closed 2023-10-30 by pelwell