raspberrypi/linux #5689: Unable to boot/login Raspberry Pi 4 Bookworm headless

alphamike-1612, 2023-10-31:
### Describe the bug

This issue appeared only with Pi Bookworm OS.
With Bullseye, no changes were required.

A Raspberry Pi 4 running the latest full version of Pi OS Bookworm is unable to boot/ssh or access webservers headless (no monitor).

The Pi refuses to boot into the OS(for lack of a better term) without detecting a monitor. 

When switching it on, the Red LED lights up and the Green LED blinks in a random fashion(ie.) Shows activity going on, after a few seconds, the Green LED goes dull and only the Red LED along with the Ethernet port light up. At this juncture, attaching a physical monitor does nothing, I am required to power cycle the Pi in order for the boot process to complete.

When a monitor is connected, the Pi boots and is accessible via SSH as expected, without any issue.

This issue can be circumvented by adding "force_hdmi_hotplug=1" in /boot/config.txt

However, I had asked on the Raspberry Pi forums and was told the updated method (to let the kernel know) is to add "video=HDMI-A-1:1920×1080@60D" in /boot/cmdline.txt.

The second line of code (the "video=HDMI") when used alone, has no effect on my Pi. I am still forced to use a physical monitor to access the Pi.

I was told it might be an issue with my Pi and was asked to raise an issue here.

Adding relevant links to forum posts below:

Initial post asking for help - 

https://forums.raspberrypi.com/viewtopic.php?t=357538&sid=b2ead654cea66533b423aace70947f1a&start=608

User Dom's reply about cmdline.txt - 

https://forums.raspberrypi.com/viewtopic.php?t=357538&sid=b2ead654cea66533b423aace70947f1a&start=617

My response about cmdline.txt not working - 

https://forums.raspberrypi.com/viewtopic.php?t=357538&sid=b2ead654cea66533b423aace70947f1a&start=620

Further conversation- 

https://forums.raspberrypi.com/viewtopic.php?t=357538&sid=b2ead654cea66533b423aace70947f1a&start=621

Detailed explanation about my issue - 

https://forums.raspberrypi.com/viewtopic.php?t=357538&sid=b2ead654cea66533b423aace70947f1a&start=622

Dom asking me to raise an issue here - 

https://forums.raspberrypi.com/viewtopic.php?t=357538&sid=b2ead654cea66533b423aace70947f1a&start=636

### Steps to reproduce the behaviour

Install Pi OS Bookworm GUI on Pi 4 8GB

Update to latest version including kernel.

Power off Pi.

Disconnect monitor.

Start the Pi.

Issue produced.

Connect monitor. Nothing will happen.

Power cycle (switch off and on) the Pi with the monitor connected, everything will work as normal.

### Device (s)

Raspberry Pi 4 Mod. B

### System

Pi OS Bookworm.

cat /etc/rpi-issue - Raspberry Pi reference 2023-10-10
962bf483c8f326405794827cce8c0313fd5880a8, stage4

vcgencmd version - September 21 2023 18:42:00
version 3094eda5aadf2c719edd2c68d1900884abc3dbca (clean) (release) (start)

unname -a --- Linux pi-home 6.1.57-v8+ #1688 SMP PREEMPT Thu Oct 12 15:14:26 BST 2023 aarch64 GNU/Linux

### Additional context

Not found in Pi OS Bullseye.

popcornmix, 2023-10-31:
I don't support you have a uart (3.3V compatible) available?
That may help identify how far through the boot we get when there is no monitor attached.

Can you confirm nothing is connected to Pi other that ethernet cable and power? (no case, no HATs, no USB, etc)?

Do you have a different power supply you can try? (the hdmi lead does connect to GND and 5V lines, and possibly is making the Pi happier with the output it gets).

alphamike-1612, 2023-10-31:
> I don't support you have a uart (3.3V compatible) available? That may help identify how far through the boot we get when there is no monitor attached.

I do have a UART, though I'm not sure if it's 3.3 or 5V, would an ESP 32/8266 help in any way?

> Can you confirm nothing is connected to Pi other that ethernet cable and power? (no case, no HATs, no USB, etc)?

Apologies, 1 USB HDD and 1 USB SSD are connected, however, they are powered via a separate 5v power adapter.

However, disconnecting everything makes no difference.

> Do you have a different power supply you can try? (the hdmi lead does connect to GND and 5V lines, and possibly is making the Pi happier with the output it gets).

Have tried it with a different supply, but no change.

Also(potentially dumb, but still) I wonder how the power supply could be a cause especially when there is no power to supply to the HDMI port (since there's no monitor connected). Conversely it works as intended when a monitor is attached.

Thank you.

popcornmix, 2023-10-31:
Do you have a link to your uart (or make/model)?
An ESP 32/8266 (or another Pi or pico) could act as a uart if you know how to configure them, and can attach wires to gpio connector. Pins 6 (gnd), 8 (GPIO14/TXD), 10 (GPIO15/RXD).

alphamike-1612, 2023-10-31:
I have a no name Chinese branded FT232RL that has a jumper for changing between 3.3 and 5v. Could you guide me on what you'd like me to do next?

Thank you.

popcornmix, 2023-10-31:
Okay, I assume this is a USB device? Connect USB end to a PC that can run a uart terminal (putty is a common one).
Configure for 115200 8N1.
Add to config.txt on boot partition:
```
dtoverlay=disable-bt
enable_uart=1
uart_2ndstage=1
```
Add to cmdline.txt (start of existing line)
```
console=serial0,115200
```
From FT232RL:
Connect GND to pin 6 of GPIO
Connect RXD to pin 8 of GPIO
Connect TXD to pin 10 of GPIO

Connect with putty, boot pi in working configuration (either with hdmi attached, or force_turbo=1) and you should see lots of uart output. Then try again in non working configuration and see where the output stops.

[This](https://www.jeffgeerling.com/blog/2021/attaching-raspberry-pis-serial-console-uart-debugging) may be useful as reference.

alphamike-1612, 2023-10-31:
Thank you. This might take me a day or 2 to complete,  I will post the results here once I've done it.

pelwell, 2023-10-31:
Another option is to use another Pi, if you have one - connect the pin 6s together, and cross link pins 8 and 10 between them.

alphamike-1612, 2023-11-05:
I did run the program, this is the output on the console in non-working configuration:
```
0x00d03115 0x00000000 0x00001fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 18
+

MESS:00:00:03.962111:0: arasan: arasan_emmc_open
MESS:00:00:03.963768:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.083989:0: arasan: arasan_emmc_set_clock C0: 0x00800000 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 400000 max: 400000 delay: 5
MESS:00:00:04.096858:0: arasan: arasan_emmc_set_clock C0: 0x00800f00 C1: 0x000e0047 emmc: 200000000 actual: 390625 div: 0x00000100 target: 400000 min: 390000 max: 400000 delay: 5
MESS:00:00:04.130575:0: arasan: arasan_emmc_set_clock C0: 0x00800f06 C1: 0x000e0207 emmc: 200000000 actual: 50000000 div: 0x00000002 target: 50000000 min: 0 max: 50000000 delay: 1
MESS:00:00:04.215425:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.218704:0: brfs: File read: 1450 bytes
MESS:00:00:04.244219:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.248727:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:04.265618:0: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.270120:0: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:04.276846:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:05.083304:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not defined
MESS:00:00:05.090609:0: *** Restart logging
MESS:00:00:05.092019:0: brfs: File read: 1450 bytes
MESS:00:00:05.102125:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.107148:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:05.117771:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.122800:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:05.128399:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:05.142193:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.147222:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.157845:0: hdmi: HDMI1:EDID error reading EDID block 0 attempt 0
MESS:00:00:05.162867:0: hdmi: HDMI1:EDID giving up on reading EDID block 0
MESS:00:00:05.168464:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:05.177229:0: HDMI0: hdmi_pixel_encoding: 300000000
MESS:00:00:05.182696:0: HDMI1: hdmi_pixel_encoding: 300000000
MESS:00:00:06.610689:0: brfs: File read: /mfs/sd/initramfs8
MESS:00:00:06.613147:0: Loaded 'initramfs8' to 0x0 size 0x1053115
MESS:00:00:06.632249:0: initramfs loaded to 0x2dfac000 (size 0x1053115)
MESS:00:00:06.640040:0: dtb_file 'bcm2711-rpi-4-b.dtb'
MESS:00:00:06.642072:0: brfs: File read: 17117461 bytes
MESS:00:00:06.652969:0: brfs: File read: /mfs/sd/bcm2711-rpi-4-b.dtb
MESS:00:00:06.656211:0: Loaded 'bcm2711-rpi-4-b.dtb' to 0x100 size 0xd503
MESS:00:00:06.676196:0: brfs: File read: 54531 bytes
MESS:00:00:06.687020:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:06.715843:0: brfs: File read: 4743 bytes
MESS:00:00:06.719376:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:06.723007:0: dtparam: audio=on
MESS:00:00:06.731833:0: brfs: File read: 1450 bytes
MESS:00:00:06.745844:0: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d-pi4.dtbo
MESS:00:00:06.812676:0: Loaded overlay 'vc4-kms-v3d'
MESS:00:00:06.967088:0: brfs: File read: 3913 bytes
MESS:00:00:06.972080:0: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
MESS:00:00:06.990814:0: Loaded overlay 'disable-bt'
MESS:00:00:07.035676:0: brfs: File read: 1073 bytes
MESS:00:00:07.038833:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:07.042856:0: Read command line from file 'cmdline.txt':
MESS:00:00:07.048728:0: 'console=serial0,115200 console=tty1 root=PARTUUID=a7d8344a-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=US'
MESS:00:00:07.183347:0: brfs: File read: 131 bytes
MESS:00:00:07.907945:0: brfs: File read: /mfs/sd/kernel8.img
MESS:00:00:07.910504:0: Loaded 'kernel8.img' to 0x80000 size 0x8522e9
MESS:00:00:09.172134:0: Kernel relocated to 0x200000
MESS:00:00:09.173985:0: Device tree loaded to 0x2df9e600 (size 0xd94f)
```

In working configuration, (with force_hdmi_hotplug=1) the same HDMI error EDID appears, but it continues and the entire system loads.

Sorr for the delay.

Barry-IA, 2023-12-04:
Found this thread as I also have problem trying to run an RPi4 with Bookworm headless -- right now using the work-around of a dummy monitor.  
Real reason for my post: is it "force_hdmi_hotplug=1"?  I've usually seen and used "hdmi_force_hotplug=1".

pelwell, 2023-12-04:
It's `hdmi_force_hotplug=1` : https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#hdmi_force_hotplug

Barry-IA, 2023-12-05:
Thanks pelwell - both for the verification and the link.  ...Wonder if that word-switch was why alphamike-1612 did not get quite the results expected.

alphamike-1612, 2023-12-05:
Apologies, this was a typo. My config file has hdmi_force_hotplug=1.
My issue is that it works with this command, but not with video=HDMI-A-1:1920×1080@60D and that this issue is specific to Bookworm.

Cheers.

mdPlusPlus, 2024-06-12:
For some reason my Pi Zero 2 W does not boot bookworm 64-bit headlessly even with `hdmi_force_hotplug=1` in `config.txt`.  
Switching to bullseye 64-bit works flawlessly however (even without `hdmi_force_hotplug=1`.

At the head of https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html it says:

>The config.txt options described here are considered legacy settings, are not used by Raspberry Pi OS Bookworm, and are no longer officially supported.

So I wonder if that is why the option is ignored. Is there an equivalent for bookworm? I can't find anything in the newer https://www.raspberrypi.com/documentation/computers/config_txt.html.

Barry-IA, 2024-06-13:
Hi mdPlusPlus!

On 6/12/24 13:51, mdPlusPlus wrote:
>
> For some reason my Pi Zero 2 W does not boot bookworm 64-bit headlessly even with hdmi_force_hotplug=1 in config.txt.
> Switching to bullseye 64-bit works flawlessly however (even without hdmi_force_hotplug=1.

Disclaimer before I get started: my comments are based on the semi-original Bookworm; things may have been updated in the interim.  I'm also working with a RPi 4.

It appears the issue is Wayland (which replaced X11) so far does not support being headless: currently it requires a monitor.  The sneaky way around that is to use a Dummy Monitor.  This is a little thumbdrive-sized device which emulates a monitor.  I haven't found any with the micro-HDMI connection so you'll need to get a micro-HDMI to HDMI adapter.  Personally I use an adapter with an approximately 6-inch cable so if I accidentally smack the dummy monitor I don't damage the Pi's video port.

S,N,: I have been using Remmina to remotely connect to the Desktop.  Yup: requires X11 and so far doesn't play nice with Wayland.  TigerVNC does connect.  I'll admit to being used to using Remmina so have clicked too many times and not have Remmina work with the Bookworm Pi's.  Sloppy solve: in Remmina edit your option to your Bookworm Pi with the Remmina Plugin; Behaviour Tab > Execute a Command > Before Connecting (enter in the field)  xtigervncviewer <IP_address>  (whatever is the dotted quad of the remote).

> At the head of https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html it says:
>
>     The config.txt options described here are considered legacy settings, are not used by [...]

On the RPi unit I checked (running Bookworm)    /boot/config.txt    says something like 'do not edit' and look to /boot/firmware/config.txt.  I also didn't see anything to fix the headless issue (at the time I originally checked -- may be updates since).       

There is a way to have Bookworm use X11, which would solve your problem now but from what I have read X11 is being discontinued eventually, plus the utilities we use will be written for Wayland.  In my opinion switching back to X11 now will cause future problems,so to me seems better to solve -- and in some instances work-around -- minor issues.

Barry

popcornmix, 2024-06-13:
There should be no need to use a "Dummy Monitor" (a fake edid).
Try adding `video=HDMI-A-1:1920x1080@60D` to cmdline.txt (on same line) which should have a similar effect.

Barry-IA, 2024-06-13:
Hi popcornmix!

> Try adding video=HDMI-A-1:1920x1080@60D to cmdline.txt (on same line) which should have a similar effect.

Thanks for the suggestion.  I tried on a newly formatted RPi card (fresh install, update/upgrade).  raspi-config to turn on ssh.  Tested TigerVNC but also needed to turn on the VNC option (not unexpected, just thought I’d test).  

Unplugged the HDMI cable, rebooted.  Still could access with TigerVNC.  Powered off, unplugged the power, waited a while.  Connect the power; TigerVNC still connected!!  So for this little experiment I didn‘t need to add anything to /boot/firmware/cmdline.txt (it does not contain any resolution commands).

I will definitely keep your suggestion.  

Barry