raspberrypi/linux #5692: Preempt-rt : rcd_node context switch during boot Rpi3B+ 64 bit

yipholamjrcs, 2023-11-02:
### Describe the bug

Hello everyone!
I am not sure if it is the right place to raise this issue so don't hesitate to tell me if it needs to be posted somewhere else.
I am currently trying to build and run a Preempt-rt patched linux distribution using Yocto on a Raspberry Pi 3B+.
But I get a rcd_node context error which prevents the filesystem to be booted.
I am filling the other fields for this issue.
So far I managed to make it boot on a 32 bit architecture, but on a 64 bit architecture I get a kernel error which halts the boot.
I can provide other info if needed.
Thanks !

### Steps to reproduce the behaviour

I append the usual recipe linux-raspberrypi for linux version 6.1.54
I clone from the raspberry pi linux repository from the branch rpi-6.1.y (96db45c50c0faf6d92c7a608ca25968aaaa70bd3) [https://github.com/raspberrypi/linux/tr ... 8aaaa70bd3](https://github.com/raspberrypi/linux/tree/96db45c50c0faf6d92c7a608ca25968aaaa70bd3)
I apply this patch [https://cdn.kernel.org/pub/linux/kernel ... t15.tar.gz](https://cdn.kernel.org/pub/linux/kernel/projects/rt/6.1/older/patches-6.1.54-rt15.tar.gz)
For the yocto kernel cache I clone it from the branch yocto-6.1 (943c6fe5dac329c7e1c30d602788f409e692b3a8) [https://git.yoctoproject.org/yocto-kern ... 09e692b3a8](https://git.yoctoproject.org/yocto-kernel-cache/commit/?h=yocto-6.1&id=943c6fe5dac329c7e1c30d602788f409e692b3a8)
In the def config file I add these parameters to bcmrpi3_defconfig(64 bit) or bcm2709_defconfig(32bit):
CONFIG_PREEMPT_RT=y
CONFIG_VIRTUALIZATION=n
CONFIG_RCU_EXPERT=n
CONFIG_RCU_BOOST=y
CONFIG_SMP=y
CONFIG_BROKEN_ON_SMP=n
CONFIG_RCU_BOOST_DELAY=500
CONFIG_KVM=n

### Device (s)

Raspberry Pi 3 Mod. B+

### System

OS and version :
Poky 4.0.13 (kirkstone)
Kernel version:
Linux version 6.1.54-rt15-v8 (oe-user@oe-host) (aarch64-poky-linux-gcc (GCC) 11.4.0, GNU ld (GNU Binutils) 2.38.20220708) #1 SMP PREEMPT_RT Wed Oct  4 11:00:59 UTC 2023

### Logs

from the serial console :
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd��]
[    0.000000] Linux version 6.1.54-rt15-v8 (oe-user@oe-host) (aarch64-poky-linux-gcc (GCC) 11.4.0, GNU ld (GNU Binutils) 2.38.20220708) #1 SMP PREEMPT_RT Wed Oct  4 11:00:59 UTC 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000037ffffff]
[    0.000000] percpu: Embedded 28 pages/cpu s77056 r8192 d29440 u114688
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 843419
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 225792
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 clk_ignore_unused video=HDMI-A-1:1920x1080M@60 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 614428K/917504K available (11648K kernel code, 2068K rwdata, 3792K rodata, 3904K init, 1036K bss, 40932K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[... 210 lines cut ...]
[    3.604216]  dwc_otg_hcd_handle_intr+0x270/0x38c
[    3.604234]  dwc_otg_hcd_irq+0x1c/0x30
[    3.604249]  usb_hcd_irq+0x38/0x54
[    3.604264]  irq_forced_thread_fn+0x44/0xd0
[    3.604277]  irq_thread+0x184/0x250
[    3.604290]  kthread+0x120/0x124
[    3.604307]  ret_from_fork+0x10/0x20
[    3.604323] ---[ end trace 0000000000000000 ]---
[    3.606747] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[    3.624130] of_cfs_init
[    3.624752] of_cfs_init: OK
[    3.626952] clk: Not disabling unused clocks
[    3.626975] ALSA device list:
[    3.626985]   No soundcards found.
[    3.677051] mmc0: host does not support reading read-only switch, assuming write-enable
[    3.681442] mmc0: new high speed SDHC card at address aaaa
[    3.683541] mmcblk0: mmc0:aaaa SC32G 29.7 GiB 
[    3.693548]  mmcblk0: p1 p2
[    3.695380] mmcblk0: mmc0:aaaa SC32G 29.7 GiB (quirks 0x00004000)
[    3.717119] EXT4-fs (mmcblk0p2): mounted fV�����ѕ� with ordered data mode. Quota mode: none.
[    3.717268] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
[    3.718711] devtmpfs: mounted
[    3.730600] Freeing unused kernel memory: 3904K
[   13.799062] mmc1: Timeout waiting for hardware interrupt.
[   13.799207] mmc1: error -110 whilst initialising SDIO card

pelwell, 2023-11-02:
The -rt patches aren't supported by us, but the FIQ usage of the dwc_otg driver is known to be problematic - try with `dtoverlay=dwc2` in config.txt.

yipholamjrcs, 2023-11-06:
Thank you, it indeed worked! I will close this issue and reopen another issue about the FIQ usage of the dwc_otg driver at the appropriate project.

closed 2023-11-06 by yipholamjrcs