raspberrypi/linux #5696: SPI: Errno 110 (Connection timed out) with transfer > 64 bytes on Pi 5

Gadgetoid, 2023-11-03:
### Describe the bug

Attempting to initiate an `spi_ioc_transfer` with a `len` greater than 64 bytes will cause a "Connection Timed Out" error.

In the spidev library we get the max SPI buffer size from `/sys/module/spidev/parameters/bufsiz`, which reports `4096`:

```
$ cat /sys/module/spidev/parameters/bufsiz
4096
```

However it is not possible to initiate an SPI transfer with a 4096 byte buffer, or indeed anything greater than 64 bytes.

### Steps to reproduce the behaviour

Enable SPI:

```
dtparam=spi=on
```

Save the following to a new file `spitest.c`:

```c
#include <fcntl.h>
#include <stdio.h>
#include <stdint.h>
#include <sys/ioctl.h>
#include <linux/spi/spidev.h>
#include <string.h>
#include <errno.h>

#define LEN 64

int main()
{
    int spi_fd;
    int status;
    uint8_t tx_buf[LEN];
    uint8_t rx_buf[LEN];
    struct spi_ioc_transfer xfer;

    if ((spi_fd = open("/dev/spidev0.1", O_RDWR)) < 0) {return -1;}

    memset(&xfer, 0, sizeof(xfer));
    xfer.tx_buf = (unsigned long)tx_buf;
    xfer.rx_buf = (unsigned long)rx_buf;
    xfer.len = LEN;
    xfer.delay_usecs = 0;
    xfer.speed_hz = 4000000;
    xfer.bits_per_word = 8;
    status = ioctl(spi_fd, SPI_IOC_MESSAGE(1), &xfer);

    if (status < 0) {
        printf("Err: %d\n", errno);
        return status;
    }
}
```

Build and run the above snippet on a Pi 5. It should run fine.

```
gcc -o spitest spitest.c
```

Change the `LEN` to anything greater than 64 and it will exit with: "Err: 110"

### Device (s)

Other

### System

```
Raspberry Pi reference 2023-09-25
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7197d4cf73c15a697279a731246ed28e76da3523, stage4
```

`vcgencmd verison`:

```
2023/10/18 18:30:17
```

`uname -a`

```
Linux raspberrypi 6.1.0-rpi4-rpi-2712 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux
```

pelwell, 2023-11-06:
Same here - we're not getting an interrupt at the end of the RX DMA. This is a regression, probably with the DMA code, probably with one of the kernel updates.

pelwell, 2023-11-07:
The transfer was timing out because the DMA stalls when the FIFO level falls below the programmed threshold. A hacky patch forcing the threshold to 1 got dropped somewhere along the way. #5699 is a slightly less hacky patch that defers the lowering of the threshold until the TX transfer completes (if there is one).

If you wait about an hour for the auto-builds/checks to complete, `sudo rpi-update pulls/5699` will install a trial kernel with the patch.

pelwell, 2023-11-07:
The GitHub elves must have had more coffee today - the builds are complete.

Gadgetoid, 2023-11-07:
Thank you. Updated and confirmed fixed.

Interestingly spidev has a very light touch on gpio, not marking them as claimed so I can still grab one of the ostensibly SPI pins to use as data/command. I wonder how long that'll last :grimacing:

pelwell, 2023-11-07:
Thanks for the confirmation - the patch is now merged.

closed 2023-11-13 by pelwell