raspberrypi/linux #5697: audioinjector octo no longer loads

flatmax, 2023-11-07:
### Describe the bug

At some point the pi kernel sound soc was updated but the audioinjector octo sound card was not updated and doesn't probe any more.

### Steps to reproduce the behaviour

set the octo as the sound card in the config file and it doesn't probe

### Device (s)

Raspberry Pi 4 Mod. B

### System

latest kernels

pelwell, 2023-11-07:
Is that all you've got for me, Matt? No kernel error messages? It's not a lot to go on.

flatmax, 2023-11-07:
I'm working on it ... stay tuned

flatmax, 2023-11-07:
Here is the stack trace from init : 

[   10.145239] CPU: 0 PID: 297 Comm: (udev-worker) Tainted: G         C         6.1.0-rpi4-rpi-v8 #1  Debian 1:6.1.54-1+rpt2
[   10.145254] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[   10.145262] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   10.145276] pc : snd_card_init+0x3cc/0x4b0 [snd]
[   10.145342] lr : snd_card_init+0xc4/0x4b0 [snd]
[   10.145404] sp : ffffffc008ed36e0
[   10.145410] x29: ffffffc008ed36e0 x28: 0000000000000000 x27: 0000000000000000
[   10.145433] x26: ffffffe75262d000 x25: ffffffe75259a000 x24: ffffffe7525996c8
[   10.145454] x23: ffffffe75259a408 x22: ffffff804020d410 x21: 0000000000000000
[   10.145474] x20: 0000000000000003 x19: ffffff8043597000 x18: 0000000000000000
[   10.145494] x17: ffffff98ff40c000 x16: ffffffe780203484 x15: 0000000000000000
[   10.145513] x14: 000477b604b69082 x13: 006f74634f20726f x12: 7463656a6e496f69
[   10.145533] x11: 00000000000001d6 x10: 0000006711365d28 x9 : ffffffe75258be74
[   10.145553] x8 : ffffff8043598000 x7 : 0000000000000007 x6 : 0000000000000018
[   10.145572] x5 : 0000000000000000 x4 : 0000000000000000 x3 : ffffffe7525996c8
[   10.145591] x2 : 0000000000000000 x1 : ffffff80447b8000 x0 : ffffff80447b8000
[   10.145611] Call trace:
[   10.145618]  snd_card_init+0x3cc/0x4b0 [snd]
[   10.145684]  snd_card_new+0x78/0xb0 [snd]
[   10.145748]  snd_soc_bind_card+0x324/0xb80 [snd_soc_core]
[   10.145870]  snd_soc_register_card+0xf8/0x110 [snd_soc_core]
[   10.145978]  devm_snd_soc_register_card+0x54/0xb0 [snd_soc_core]
[   10.146088]  audioinjector_octo_probe+0x1d0/0x274 [snd_soc_audioinjector_octo_soundcard]
[   10.146118]  platform_probe+0x70/0xcc
[   10.146144]  really_probe+0xc4/0x2e0
[   10.146157]  __driver_probe_device+0x80/0x120
[   10.146170]  driver_probe_device+0xe0/0x164
[   10.146182]  __driver_attach+0x9c/0x1a4
[   10.146193]  bus_for_each_dev+0x7c/0xe0
[   10.146205]  driver_attach+0x2c/0x40
[   10.146215]  bus_add_driver+0x15c/0x210
[   10.146226]  driver_register+0x80/0x13c
[   10.146238]  __platform_driver_register+0x30/0x40
[   10.146253]  audioinjector_octo_driver_init+0x28/0x1000 [snd_soc_audioinjector_octo_soundcard]
[   10.146281]  do_one_initcall+0x60/0x2c0
[   10.146296]  do_init_module+0x50/0x200
[   10.146313]  load_module+0x1b04/0x1f60
[   10.146327]  __do_sys_init_module+0x19c/0x1d4
[   10.146341]  __arm64_sys_init_module+0x24/0x30
[   10.146356]  invoke_syscall+0x50/0x120
[   10.146374]  el0_svc_common.constprop.0+0x104/0x124
[   10.146391]  do_el0_svc+0x34/0xd0
[   10.146408]  el0_svc+0x30/0x9c
[   10.146425]  el0t_64_sync_handler+0xf4/0x120
[   10.146438]  el0t_64_sync+0x18c/0x190
[   10.146450] ---[ end trace 0000000000000000 ]---

pelwell, 2023-11-07:
The first few lines of the trace are missing - what are they?

It would be interesting to see if the latest rpi-update kernel has the same issue.

flatmax, 2023-11-08:
Here it is again. Word has it that people are resorting to roll back a few years to get this to work.

[   10.144799] ------------[ cut here ]------------
[   10.144825] WARNING: CPU: 0 PID: 297 at sound/core/init.c:309 snd_card_init+0x3cc/0x4b0 [snd]
[   10.144918] Modules linked in: binfmt_misc brcmfmac brcmutil cfg80211 hci_uart snd_soc_cs42xx8_i2c snd_soc_cs42xx8 regmap_i2c raspberrypi_hwmon btbcm snd_soc_audioinjector_octo_soundcard(+) bluetooth rpivid_hevc(C) bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common ecdh_generic ecc rfkill videodev libaes snd_bcm2835(C) snd_soc_bcm2835_i2s vc_sm_cma(C) raspberrypi_gpiomem mc nvmem_rmem uio_pdrv_genirq uio i2c_dev fuse dm_mod ip_tables x_tables ipv6 vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper v3d drm_kms_helper syscopyarea gpu_sched sysfillrect sysimgblt fb_sys_fops drm_shmem_helper drm drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine i2c_brcmstb i2c_bcm2835 snd_pcm snd_timer snd backlight
[   10.145239] CPU: 0 PID: 297 Comm: (udev-worker) Tainted: G         C         6.1.0-rpi4-rpi-v8 #1  Debian 1:6.1.54-1+rpt2
[   10.145254] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[   10.145262] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   10.145276] pc : snd_card_init+0x3cc/0x4b0 [snd]
[   10.145342] lr : snd_card_init+0xc4/0x4b0 [snd]
[   10.145404] sp : ffffffc008ed36e0
[   10.145410] x29: ffffffc008ed36e0 x28: 0000000000000000 x27: 0000000000000000
[   10.145433] x26: ffffffe75262d000 x25: ffffffe75259a000 x24: ffffffe7525996c8
[   10.145454] x23: ffffffe75259a408 x22: ffffff804020d410 x21: 0000000000000000
[   10.145474] x20: 0000000000000003 x19: ffffff8043597000 x18: 0000000000000000
[   10.145494] x17: ffffff98ff40c000 x16: ffffffe780203484 x15: 0000000000000000
[   10.145513] x14: 000477b604b69082 x13: 006f74634f20726f x12: 7463656a6e496f69
[   10.145533] x11: 00000000000001d6 x10: 0000006711365d28 x9 : ffffffe75258be74
[   10.145553] x8 : ffffff8043598000 x7 : 0000000000000007 x6 : 0000000000000018
[   10.145572] x5 : 0000000000000000 x4 : 0000000000000000 x3 : ffffffe7525996c8
[   10.145591] x2 : 0000000000000000 x1 : ffffff80447b8000 x0 : ffffff80447b8000
[   10.145611] Call trace:
[   10.145618]  snd_card_init+0x3cc/0x4b0 [snd]
[   10.145684]  snd_card_new+0x78/0xb0 [snd]
[   10.145748]  snd_soc_bind_card+0x324/0xb80 [snd_soc_core]
[   10.145870]  snd_soc_register_card+0xf8/0x110 [snd_soc_core]
[   10.145978]  devm_snd_soc_register_card+0x54/0xb0 [snd_soc_core]
[   10.146088]  audioinjector_octo_probe+0x1d0/0x274 [snd_soc_audioinjector_octo_soundcard]
[   10.146118]  platform_probe+0x70/0xcc
[   10.146144]  really_probe+0xc4/0x2e0
[   10.146157]  __driver_probe_device+0x80/0x120
[   10.146170]  driver_probe_device+0xe0/0x164
[   10.146182]  __driver_attach+0x9c/0x1a4
[   10.146193]  bus_for_each_dev+0x7c/0xe0
[   10.146205]  driver_attach+0x2c/0x40
[   10.146215]  bus_add_driver+0x15c/0x210
[   10.146226]  driver_register+0x80/0x13c
[   10.146238]  __platform_driver_register+0x30/0x40
[   10.146253]  audioinjector_octo_driver_init+0x28/0x1000 [snd_soc_audioinjector_octo_soundcard]
[   10.146281]  do_one_initcall+0x60/0x2c0
[   10.146296]  do_init_module+0x50/0x200
[   10.146313]  load_module+0x1b04/0x1f60
[   10.146327]  __do_sys_init_module+0x19c/0x1d4
[   10.146341]  __arm64_sys_init_module+0x24/0x30
[   10.146356]  invoke_syscall+0x50/0x120
[   10.146374]  el0_svc_common.constprop.0+0x104/0x124
[   10.146391]  do_el0_svc+0x34/0xd0
[   10.146408]  el0_svc+0x30/0x9c
[   10.146425]  el0t_64_sync_handler+0xf4/0x120
[   10.146438]  el0t_64_sync+0x18c/0x190
[   10.146450] ---[ end trace 0000000000000000 ]---

pelwell, 2023-11-08:
PR https://github.com/raspberrypi/linux/pull/5705 should remove the kernel backtrace, but I'm not convinced it will get things working again as this is only a warning.

As usual these days, you can either apply the patch to your own tree and build it yourself or wait about an hour and run `sudo rpi-update pulls/5705`.

flatmax, 2023-11-08:
Thanks - I'll give it a check and continue from there.

flatmax, 2023-11-09:
I can confirm that the driver now loads - thanks for doing that.

It seems that the audio is corrupt however. Are there any major changes to the i2s audio driver over the last while ?

pelwell, 2023-11-09:
There don't seem to be any significant changes to the I2C driver since back in 2017, except for one. I changed how DMA addresses are calculated, but the end results should be the same. If that was broken then no sound cards would work at all.

In what way is the audio corrupt?

flatmax, 2023-11-09:
It doesn't resemble tones on the output when I put that in over DMA.

When I get some time I'll have a poke around with an oscilloscope on the 
I2S bus and try to work out what is cooking.

Matt

LaiArturs, 2024-04-18:
Any resolution on this? Having similar issue.
[stackTrace2.log](https://github.com/raspberrypi/linux/files/15028679/stackTrace2.log)

```
Linux raspberrypi 6.6.26+ #1754 Thu Apr 11 14:43:19 BST 2024 armv6l GNU/Linux
```

LaiArturs, 2024-04-19:
Thank you @pelwell, the kernel warning indeed is gone now for wm8731.

pelwell, 2024-04-20:
I'm going to close this. If it turns out that there are audio problems, please open another issue with a more specific name.

closed 2024-04-20 by pelwell

F-Haferkorn, 2025-03-14:
Hi folks,
I am strongly interested  to get the audioInjector up and running again.
In 2025-03-13 I do not get it up to work neither with a PI-4 nor with a PI5 (on a CM5IO-board) 
@flatmax  What is the eStatus of this AudionInjector Octo Problem.

flatmax, 2025-03-14:
Sounds great.

I have a conjecture on this.

I think it is likely the Octo stopped working when the i2s driver became a tdm driver.
If you were to commit the original non tdm driver as a new i2s driver, it is possible it would work again.

If the old i2s non tdm driver were tested and worked, that would suggest we need a second i2s driver for more interesting tinkering.

flatmax, 2025-03-15:
The other conjecture is that the dma buffer isn't moduli 8 any more.