raspberrypi/linux #5698: VC04: Address space mismatch - bulk_userdata

uajain, 2023-11-07:
### Describe the bug

Hello,

I am wondering if anyone here as visibility on [1]

It would help squashing the VC04 Sparse warnings for upsteram VC04 de-stage.

[1] : https://github.com/raspberrypi/linux/blob/rpi-6.6.y/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_dev.c#L543

### Steps to reproduce the behaviour

Running Sparse on the mainline linux-staging branch, I get:

```
/home/uajain/src/kernel/mainline/media_tree/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_dev.c:335:26: warning: incorrect type in assignment (different address spaces)
/home/uajain/src/kernel/mainline/media_tree/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_dev.c:335:26:    expected void *[assigned] userdata
/home/uajain/src/kernel/mainline/media_tree/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_dev.c:335:26:    got void [noderef] __user *userdata
/home/uajain/src/kernel/mainline/media_tree/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_dev.c:553:47: warning: incorrect type in assignment (different address spaces)
/home/uajain/src/kernel/mainline/media_tree/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_dev.c:553:47:    expected void [noderef] __user *[addressable] [assigned] bulk_userdata
/home/uajain/src/kernel/mainline/media_tree/drivers/staging/vc04_services/interface/vchiq_arm/vchiq_dev.c:553:47:    got void *bulk_userdata
```

On one of the location, there is clear FIXME about figuring out the correct address spaces. 

### Device (s)

Raspberry Pi 3 Mod. B+

### System

Compile / static analysis on upstream branch:
https://git.kernel.org/pub/scm/linux/kernel/git/gregkh/staging.git/log/?h=staging-testing

### Additional context

VC04 mainline de-staging

pelwell, 2023-11-07:
`bulk_userdata` is an opaque pointer that gets returned to the client when the bulk transfer completes. It may be a user-space pointer if initiated via the ioctl interface, or it may be a kernel pointer if initiated via the in-kernel API.

lategoodbye, 2023-11-08:
@uajain Please also look at the comments below this patch:
https://patchwork.kernel.org/project/linux-arm-kernel/patch/20210104120929.294063-2-phil@raspberrypi.com/#23871217