raspberrypi/linux #5700: Something up with python 3.11 / rust / arm64 / grpc / glib

samskiter, 2023-11-07:
### Describe the bug

Apologies in advance - this is going to be a poor bug report. I'm hoping someone more knowledgeable looks at all this and sees something obvious that needs to be fixed!

Recently, I've updated python (to 3.11 and 3.12 - I've tried both) on my RPi 4 to get some of the new language features. Unfortunately I've since had some issues after `pip install`ing some common modules:

- `grpcio` (via `google-cloud-pubsub`)
- `cryptography` (via `certbot`)

In the case of `google-cloud-pubsub` I get a runtime error, as outlined here: https://github.com/grpc/grpc/issues/34888
This seems to be related to GLIBC versions on Raspbian. I tried downgrading grpcio but the installation hangs when building the wheel - why wouldn't it find an pre-compiled wheel?

In the case of `cryptography` I get an install time error - the issue is that rust can't be found - but again, why wouldn't pip pull a prebuilt wheel? Relevant bug reports are:
- https://github.com/rust-lang/rustup/issues/3342
- https://github.com/rust-lang/rustup/issues/3307

*Here's my question/potential bug report*
Are these two issues related? Why am I getting pip 'misses' and why are the installs/runtimes failing with very 'low level' issues? (like compilers missing). I may be way of the mark, but if I am maybe this serves as a bit of a more general bug report along the lines of 'python 3.11 has a lot of incompatibilities with Raspbian'

### Steps to reproduce the behaviour

- upgrade python to 3.11
- pip install google-cloud-pubsub
- python
- `>>> from google.cloud.pubsub_v1 import PublisherClient`
- BUG OCCURS
- pip install certbot
- BUG OCCURS

### Device (s)

Raspberry Pi 4 Mod. B

### System

```
$ cat /etc/rpi-issue
Raspberry Pi reference 2023-05-03
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 47eee1f0ddcf8811559d51eea1c1bb48335e3e88, stage4
```
```
$ vcgencmd version
Mar 17 2023 10:50:39 
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
```
```
$ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

### Additional context

Additional context? Just that I'm sorry for such a pants bug report - but I felt I needed to file something on Raspbian rather than the individual libraries...

popcornmix, 2023-11-07:
Sorry, this is the wrong place. This repo is for bugs identified in the linux kernel.

I think you would be better off bringing this up on the [forum](https://forums.raspberrypi.com)

closed 2023-11-07 by popcornmix

samskiter, 2023-11-07:
> forum

No problem, thanks :) 

Just FYI I followed instructions for reporting raspbian bugs here: https://www.raspbian.org/RaspbianBugs#:~:text=If%20you've%20noticed%20something,re%20able%20to%20fix%20it. Which led to this project's issues page. Maybe worth updating those instructions?

popcornmix, 2023-11-07:
That's not a raspberry pi site, so we don't have control of it.
It does say
> In particular kernel bugs should be filed under [raspberrypi/linux](https://github.com/raspberrypi/linux/issues)

(which links to this githb repo).